ngardiner/TWCManager #3: Documentation Improvements

ngardiner, 2019-10-20:
Documentation, especially for modules are lacking. Target a few doco improvements per release for the next few releases leading up to v1.2.0:

v1.1.4:
- Powerwall2 EMS module
- TED EMS module

Future

- Finish installation documentation
- HASS EMS Module
- Fronius EMS Module
- HASS Status Module
- MQTT Status Module
- HTTP Control Module
- MQTT Control Module

one4spl, 2020-03-05:
Hi, on the subject of documentation the README.md stops at "To be completed shortly" 

Any chance you could spin through documenting a clean installation onto a new Rpi running a current OS?

ngardiner, 2020-03-08:
Absolutely, will do shortly

ngardiner, 2020-03-09:
I've made some progress on the documentation, with a more logical step through from installation of the OS to installation of the script. Would be great to get some feedback on whether it is helpful in installing TWCManager and if anything is missing that I could add.

KCmangelma, 2020-04-17:
It sure is helpful to have all the steps laid out, but.. what if the red light is blinking, but the script output isn't really telling me what's wrong? Linkready messages are being sent at start, but no regular status updates are shown. Debug level turned up to 11, some messages and bytes are being ignored. All those fancy modules are no use it the basic serial communication isn't working :) Thanks.

MikeBishop, 2020-04-17:
First, check in the manual that the blink code is that it's a slave without a master.  There are other reasons for error states, so make sure you're debugging the right one.  If there's more than 1-2 initial messages being ignored, my bet would be that it's not reading or sending the RS485 correctly.  (One or two ignored / invalid messages at start-up is normal, since it could have started listening mid-message.) I'd make sure that connection is using the right wires, etc.

ngardiner, 2020-10-09:
Closing this, a lot of the documentation has been improved.

closed 2020-10-09 by ngardiner