ngardiner/TWCManager #4: Alternate to serial interface

ngardiner, 2019-10-20:
Provide the ability to use a different interface to serial (recommend a network interface such as a TCP socket) so that:

- Installations where the RS-485 interface is remote to the TWCManager machine could be supported
- Testing will be much easier

AndySchroder, 2020-02-08:
How will you physically convert RS-484 to TCP?

ngardiner, 2020-02-09:
Hi @AndySchroder,

Devices such as this would allow RS-485 over TCP/IP:

https://www.ebay.com.au/i/274123299148

MikeBishop, 2020-03-01:
Also, if someone reverse-engineers the Gen3 load sharing protocol, TWCManager likely just need to be running somewhere on the network.  (Unless Tesla has built more security into the Gen3, which isn't unlikely.)

ngardiner, 2020-03-10:
As a first step toward this goal, 388ed5d introduces a dedicated module for RS485 communication which currently only provides the ability to send messages over the serial interface. All functions now use this module to send messages to slave TWCs. In future, this will be extended to recieve and handle serial setup via this module, which will then allow substitution for other interface modules in future.

ngardiner, 2020-03-21:
Serial communications is now entirely contained within the RS485 interface module as of c070c46. It is still statically referenced in the code, the complexity here is that with other modules we can have multiple types active simultaneously (eg multiple EMS, Control or Status modules won't conflict) but for Interface we should only use one. The idea will be that in an upcoming change, the config file will elect 1 module to use (RS485 by default) and we'll have a pointer to the module selected.

Next up will be a TCP interface module, which will do exactly what the current module does but over a TCP socket.

AndySchroder, 2020-05-29:
How is your RS-485 over TCP/IP authenticated and secured?

ngardiner, 2020-05-29:
We don't have great latitude to dictate how the protocol is implemented, as we're only one half of the equation. This implementation assumes direct raw TCP to serial conversion. If we needed to be running on both sides of the link to support a more custom implementation, it wouldn't make a lot of sense (as you'd just hard wire it instead of splitting it into a client/server role across the network).

So the idea is to interoperate. It would be somewhat doubtful that any of the embedded options out there offer TLS, but if they did, it would be quite trivial to offer TLS on our side. The problem is that if no solution or device implements it, it's rather pointless for us to, as we'd only be able to talk to ourselves.

AndySchroder, 2020-05-30:
That makes sense, I guess the best application of this is a network that is not internet connected so you don't have to worry about the insecurity. Many other controls related communication over TCP usually has this same insecurity problem, so having a separate network (or VLAN) dedicated to that kind of stuff may be something people would normally do. This way you can still harness the advantages of ethernet (shared, reduced total facility wiring) over RS-485 (which has much lower bandwidth and is more of a broadcast based messaging).

tjikkun, 2021-01-20:
Just for info: pyserial can handle serial over tcp as well: https://pyserial.readthedocs.io/en/latest/url_handlers.html

ngardiner, 2021-05-03:
Going to close this off now as we have plenty of alternatives to the physical serial port (Dummy, TCP, Serial URL). Thanks all!

closed 2021-05-03 by ngardiner

bikeymouse, 2021-05-10:
Using the search-function I ended here trying to find an answer how to connect my RS-485 to TCP/IP convertor (I'm using a [WaveShare RS485 to ETH convertor](https://www.waveshare.com/wiki/RS485_TO_ETH]), that also works fine reading Modbus Utility meters)

I did find the "tcp" option in the config.json file 

```
      "TCP": {
        "enabled": false

        # The TCP module allows communications over a TCP listener or client
        # socket. This can be used to integrate with network-based RS485
        # interfaces.
      }
```
...but how can I use this to set the IP-address, port and possibly thee protocol?

Thanks a lot!

tjikkun, 2021-05-11:
You can use https://github.com/ngardiner/TWCManager/pull/214#issue-564399801
So this means something like:
```
"RS485": {
  "enabled": true,
  "port": "socket://localhost:7777"
},
```
For port, you can use anything from https://pyserial.readthedocs.io/en/latest/url_handlers.html

bikeymouse, 2021-05-11:
Hi thanks for the answer! 

I have tried this, but then I get this error:

```

TWCManager 20 Unhandled Exception:Traceback (most recent call last):

  File "TWCManager.py", line 568, in <module>

    master.send_master_linkready1()

  File "/usr/src/TWCManager/lib/TWCManager/TWCMaster.py", line 1055, in send_master_linkready1

    self.getInterfaceModule().send(

  File "/usr/src/TWCManager/lib/TWCManager/TWCMaster.py", line 270, in getInterfaceModule

    return self.getModulesByType("Interface")[0]["ref"]

IndexError: list index out of range

```
This is my interfaces section in the config.json:

```
    "interface": {
      "Dummy": {
        "enabled": false,

        # The dummy module is used for testing. It will simulate an actual
        # slave TWC for the purpose of offline testing of TWCManager.
        # Most people would not enable this module.
        # This should be a two-byte ID
        "twcID": "AB"
      },
      "RS485": {
        "enabled": false,

        # TWC's rs485 port runs at 9600 baud which has been verified with an
        # oscilloscope. Don't change this unless something changes in future hardware.
        "baud": 9600,

        # Most users will have only one ttyUSB adapter plugged in and the default value
        # of '/dev/ttyUSB0' below will work. If not, run 'dmesg |grep ttyUSB' on the
        # command line to find your rs485 adapter and put its ttyUSB# value in the
        # parameter below.
        # If you're using a non-USB adapter like an RS485 shield, the value may need to
        # be something like '/dev/serial0'.
        "port": "/dev/ttyUSB0"

      },
      "TCP": {
        "enabled": true,
        "port": "socket://192.168.1.160:520"

        # The TCP module allows communications over a TCP listener or client
        # socket. This can be used to integrate with network-based RS485
        # interfaces.
      }
    }

```
Any idea, what could be wrong?

bikeymouse, 2021-05-11:
To add to the above: that error is using the Docker image (version v.1.2.2)

If I use the local Python installed version the error is the same, only the line numbers are different:

```
23:43:21 TWCManager 20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 547, in <module>
    master.send_master_linkready1()
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 1003, in send_master_linkready1
    self.getInterfaceModule().send(
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 245, in getInterfaceModule
    return self.getModulesByType("Interface")[0]["ref"]
IndexError: list index out of range
```

tjikkun, 2021-05-12:
Please try wirh
```
    "interface": {
      "Dummy": {
        "enabled": false,

        # The dummy module is used for testing. It will simulate an actual
        # slave TWC for the purpose of offline testing of TWCManager.
        # Most people would not enable this module.
        # This should be a two-byte ID
        "twcID": "AB"
      },
      "RS485": {
        "enabled": true,

        # TWC's rs485 port runs at 9600 baud which has been verified with an
        # oscilloscope. Don't change this unless something changes in future hardware.
        "baud": 9600,

        # Most users will have only one ttyUSB adapter plugged in and the default value
        # of '/dev/ttyUSB0' below will work. If not, run 'dmesg |grep ttyUSB' on the
        # command line to find your rs485 adapter and put its ttyUSB# value in the
        # parameter below.
        # If you're using a non-USB adapter like an RS485 shield, the value may need to
        # be something like '/dev/serial0'.
        "port": "socket://192.168.1.160:520"

      },
      "TCP": {
        "enabled": false,
        "port": "not_used"

        # The TCP module allows communications over a TCP listener or client
        # socket. This can be used to integrate with network-based RS485
        # interfaces.
      }
    }
```
So use the RS485 module, not the TCP one. Because basically it still is RS485, just over socket connection, not a local device.
I can understand the confusion though.

@ngardiner Do you know the status of the TCP module, is it supposed to be working? If not, it might be better to disable it, or have it spit out a message telling so?

ngardiner, 2021-05-12:
The TCP module is still under development - it is to allow us to interface with other protocols and transports such as TLS encrypted transports or those with more complex encodings. I'll update the documentation to reflect but the path chosen depends on the remote side. If it's capable of raw serial or rfc2217 telnet interface provided by pyserial it can use the RS485 module, if anything more complicated than that, it will be the TCP module.

bikeymouse, 2021-05-12:
> So use the RS485 module, not the TCP one. Because basically it still is RS485, just over socket connection, not a local device.
> I can understand the confusion though.

Ah, Ok of course. I have now changed it and it works perfect. Thanks for the great support!

So good to know for who is interested, this € 25,- [WaveShare RS485 to ETH](https://www.waveshare.com/wiki/RS485_TO_ETH) adapter works great: it just transparently sends the data across TCP/IP. I have exactly the same device to read my Modbus consumption meters to have real-time measurements of my Solar panels and Charger and also that works, with the same settings.

So thanks, way better then again another USB-device in my home-computer.

bikeymouse, 2021-05-13:
P.s. I just noticed the updated[ documentation on the RS-485 interface](https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Interface_RS485.md). 

For network communications it mentions these options:
```
"device": "rfc2217://192.168.1.2:4000/"
"device": "socket://192.168.1.2:4000/"
```
but not the configuration I now have working:
` 
"port": "socket://192.168.1.160:520"
`

Perhaps good to add that too?