ngardiner/TWCManager #14: Feature request: Charge limit per mode

MikeBishop, 2019-11-16:
Since you can set the charge limit via the Tesla API, would be nice to have the desired charge as a parameter for each charge mode (e.g. only charge to 50% during scheduled charging, but to 90% when on green power).

ngardiner, 2019-11-24:
Hi @MikeBishop (apologies for the lack of updates for a while, it's been a busy time) - unfortunately I don't believe there is any such control over charging rates via API, such a feature would provide an alternative to modding the TWC which has until now been the only method to achieve this.

In terms of implementing the logic you've described from the TWC side, there is also a secondary limitation in that whilst we can read the SoC from the API (but not from the TWC), we can't be sure which car is charging. We can do a best effort guess which is to query the state of each car and find one that says it is charging whilst TWC is in use and then start/stop charging from there, but it could very easily stop the wrong car from charging. I'll probably not do this myself as it doesn't really align with my use case, but I'd have no problem extending the policy approach to detect SOC value of a charging vehicle and use that as a condition for charging/not charging.

MikeBishop, 2019-12-03:
The API doesn't permit controlling the charge rate, but it permits controlling the charge limit, which was the request here.  If the TWC offers current, but the car is at/above its configured charge limit, then the car won't start charging.  This is purely an API call that would get made before entering a particular configuration.

The use case for this is essentially to opportunistically charge from solar during the day, but ensure the car always starts the day with at least 50% charge from off-peak power.  So the sequence would look like:

- Midnight, start of scheduled charging:  Set car's charge limit to 50% via API, tell slave TWC to offer up to 48A.
- 6 AM, start of green power charging:  Set car's charge limit to 90% via API, begin allowing TWC current based on the available green power

I admit I hadn't thought much about which car it would apply to, and I thought the charger did know which VIN it was charging.  But if not, it might be possible to use the car's location as a trigger -- only apply the charge limit to cars which are "at home," and re-do that check whenever charging actually begins to account for cars that have arrived since the start of the period.

As I said in the other issue, I don't have the hardware installed yet, so I'm theorizing and designing at this point, but if the feature doesn't exist by the time I do, I'll take a crack at it.

ngardiner, 2020-02-02:
When combined with geo-location, this makes plenty of sense. One thing I would like to test before the implementation is the behaviour when adjusting the charge level - ie will it let us lower the level and then increase it without refusing further charging. From memory there's no such limitation, you could set the limit to the current SOC for a while and then increase it later. If so, we could offer this as an additional control to manage the car's charging behavior. I'll tag this as needs testing, do some testing on my car, and once I have my head around how it behaves we can look to add it in.

MikeBishop, 2020-02-03:
Yes, I think geolocation becomes more important with this.  One important property of the current setup that I'm having trouble designing a good way to maintain is that a long-term idle TWC makes no (vehicle) API requests.  You can check for cars that have arrived whenever power starts to flow on a slave, because it's set up to allow them to charge for ~1 minute when you first plug in.  You can apply charge limits to all cars you already know are at home whenever the limit changes (and if there are none, take no action).

It's departure that's tough.  My first thought was just to run a periodic task and check if the cars at home have left every 30 minutes, but that would run in perpetuity even if no cars are ever present.  My current thought is to check up to three times at intervals after a slave stops delivering measurable power.  That should work *if* a car that wasn't otherwise charging draws enough shore power to make a slave indicate there's usage, even if briefly.  You then drive away and one of the polls will discover that you've left.

So the other thing I'd like to request you test:  When a car is charged and asleep, how common is it that the car can wake and be unplugged before the slave sees any current draw (>2A, since that seems to be the threshold the software currently uses for "non-zero."  Worst case, it'll get picked up by the next change in policy, but that could be a long time away.

MikeBishop, 2020-02-04:
Have a look at https://github.com/ngardiner/TWCManager/compare/v1.1.7...MikeBishop:limit_by_mode -- I haven't checked that it works, or even that everything is properly formatted, but if you have feedback on the general approach, I'd welcome it.

ngardiner, 2020-03-21:
Closed thanks to @MikeBishop's great work

closed 2020-03-21 by ngardiner