ngardiner/TWCManager #21: MQTT charge_now command sets charging to only 12A

MikeBishop, 2020-02-03:
Why not set it to the maximum instead?  (Filing an issue instead of a PR because I'm curious what the rationale is.)

ngardiner, 2020-02-04:
The MQTT charge now logic is in a bit of an incomplete state at this point. The intention was to provide customizable MQTT charge control, however only a binary on/off is implemented as yet. In order to test MQTT-triggered charging, a number needs to be supplied which would both fit the criteria of being a valid amperage and also proving that MQTT triggered the manual charging (hence the use of 12 amps).

This issue is certainly the right vehicle for this, as the testing can be done and then the necessary parameter added to allow the amperage to be specified.

ngardiner, 2020-02-23:
Commit f7de3a0 adds support for a configurable set of parameters via the MQTT payload to the chargeNow topic. The command now accepts 2 parameters separated by comma, the first parameter is the number of amps to charge at, and the second parameter is the number of seconds to charge for.

An example is below:

```
mosquitto_pub -t 'TWC/control/chargeNow' -m 16,1200
```

```
MQTTControl: (3) MQTT Message called chargeNow with payload 16,1200
TWCMaster: (8) All policy conditions have matched. Policy chosen is Charge Now
00:31:26: Green energy generates 0W, Consumption 0W, Charger Load 0W
          Limiting car charging to 0.00A - 0.00A = 16.00A.
          Charge when above 6A (minAmpsPerTWC).
```

closed 2020-02-23 by ngardiner