ngardiner/TWCManager #22: Charge policy should be in config, not code

MikeBishop, 2020-02-04:
It doesn't make sense for someone setting this up to be modifying change-tracked code to configure their local policies.  We should import the policy from the config file at startup; code can contain a simple default policy

ngardiner, 2020-02-04:
Totally agree, @MikeBishop. I am not sure it should be in _the_ config file, though. Perhaps in a separate (JSON? YAML?) file - the benefit of which would be that the web interfaces could be used to modify policy and write it out without clobbering the rest of the config.

Some things to consider:

- Base policy is probably a good idea implemented inside the code, as you say, as we may want to maintain/customise the default behaviour of the policy in later releases.
- Once the rest of the policy gets broken out into another file, it's going to be very hard to synchronise any changes to it (unless they are done programmatically in the code), so we would want to be sure of two things:
   - That the current spec won't change too much (or can be fixed up programmatically)
   - That we won't need to add any rules for it after the separation - which I believe is known already, as the current code works fine for green energy charging

So I guess the key questions are, should we just deprioritise the code-based policy we have now and allow overrides on top, or should we break out some of that policy into an external file and keep some of it in the code? Would be good to get your thoughts as well as your policy is clearly going to be different to mine (which is effectively just charge while on green energy) so your use case is very much the key to customizable policy.

I am leaning towards the following whilst writing this out:

- Current default policy stays in code, no file = use default
- Custom policy overrides default in-code policy entirely, allowing entirely custom policy if the file exists and has custom policy within it

MikeBishop, 2020-02-04:
Here's the policy I'd ultimately like to be able to do:

- If Powerwall is off-grid, do not charge at all
- If Charge Now command received (probably MQTT via Hubitat eventually; web interface for now), charge at full speed to 100%
- If Powerwall's Storm Watch mode is active, charge at full speed to 90%
    - The API for detecting Storm Watch has not yet been identified that I can find, so this is impossible for now
- Between midnight and 5 AM, charge at full speed to 50%
- During the day, follow green power to 90%
- Otherwise, do not charge

I agree that a simple default policy of:
- Charge Now (at speed set by web interface)
- Track green energy during the day
- Do not charge (or charge at non-scheduled speed set by web interface)

should be in code, and the config file overrides the whole set.  One possible usability improvement might be to write out a file containing the default policy if the file is not found, so people can use that as a starting point to add/tweak conditions.

ngardiner, 2020-02-23:
I've made some progress towards this with commit 6b47238, which has added the following:

- A section of the configuration which allows overriding of the policy entirely, although I don't necessarily recommend it. It allows full customization of the charging policy and contains the current policy commented out which may be edited directly in the config file.

- Two hooks into the default policy, one section (before) is after the Charge Now rule, but before all other rules, and allows one or more rules to be inserted into the policy.

- The other (after) is inserted before the unscheduled charging rule, and would be more suited to small tweaks to the current policy 

I will work on an example for each, I may even be able to give a simple approximation of one of your examples here in the configuration file. Let me know what you think about this approach. It hasn't been tested at all yet, so there's still an update to come here after testing, but it is not far off completion.

MikeBishop, 2020-02-24:
This mechanism for supplying the states seems totally reasonable.  The broader framework of Charge Now / On-Schedule / Green / Off-Schedule isn't something I'd expect most people want to mess with; having the ability to insert overrides is probably sufficient.  (For me, Powerwall offline and Storm Watch).

However, in the branch where I started to implement supplying a charge limit for each mode, it's looking for that as a property of each policy.  The easiest way around that is probably just using variables for the limit on each of the built-in policies, but I haven't added variable evaluation to that yet.  Sounds like I need to.  :-)

MikeBishop, 2020-02-24:
Actually, one adjustment -- I would put the hooks before and after Charge Now.  In my mind, the two categories would be things that should override even explicit commands (grid outage) and things that should override the normal schedule (Storm Watch).

ngardiner, 2020-04-26:
Just doing some pruning of the issues and I think we've well and truly made inroads on this one with your help, @MikeBishop. While I think there's scope for future improvements such as potentially providing variable override points in the policy, I also agree that it's very unlikely it will be needed given the three overrides currently, so I'm going to tentatively close this off unless we find we need extra flexibility, at which time we can add it.

closed 2020-04-26 by ngardiner