ngardiner/TWCManager #27: Can policy check time be reduced less than 60 seconds?

AndySchroder, 2020-02-09:
Policy check is done once every 60 seconds, per the following 

`if ((self.lastPolicyCheck + 60) > time.time()):`

https://github.com/ngardiner/TWCManager/blob/v1.1.7/lib/TWCManager/TWCMaster.py#L625

Are there any problems with reducing the policy check time to something much lower, like 5 seconds?

ngardiner, 2020-02-09:
No issue there, however green energy polling occurs once per policy cycle, so reducing the policy check to 5 seconds would result in any configured green energy system being polled that often. 

60 seconds is a conservative setting, it's unlikely to cause an issue when polling a solar inverter for example at that rate. I couldn't confidently state that 5 seconds would be a safe default, but I definitely do think it should have a configurable override, I think that's a good suggestion and could be implemented easily.

AndySchroder, 2020-02-09:
I've been running at 10 seconds and it seems okay so far. Probably 5 seconds is too low though because of the spike current that occurs every time you increase the current. This is only an issue if your policy changes every time it cycles. If your policy doesn't change twice within a 5 second period, it should probably be okay.

MikeBishop, 2020-02-10:
I've also been wondering about a way to respond to changes more frequently.   My concern is with the Powerwall when it switches to backup mode -- max output is 5kW sustained / 7kW peak x number of units.

Case in point, I have two, so 10kW sustained, 14kW peak.  If I'm charging at full power, that's ~11kW right there.  The Powerwalls can only sustain their peak output for a few seconds, so ideally TWCManager would respond to that event quasi-instantly.

In reality, it only finds out about grid status by polling.  It runs the policy check every 60 seconds, then imposes a 60 second constraint on changing the offered amps.  Since the offered amps primarily changes based on policy / green energy, those *should* always line up perfectly, but hypothetically a slight variation in run time could cause the periods to have minimal overlap, meaning a ~2 minute worst case response time.  In terms of draining the Powerwall, that's not a big deal.  In terms of overloading it, it might be.

I might back my overnight charging down to 32A or 40A, just to avoid tripping the Powerwalls if there's an outage during that time period.

ngardiner, 2020-02-23:
That's a good point @MikeBishop. For drawing from green energy sources we have assumed a grid backup, and have therefore not been too sensitive about the polling interval. Whilst it may have a slight cost implication by polling at a longer rate, it never posed a problem from an excessive draw perspective.

What I think we could start off with is an approach where we globally reduce the polling time as 60 seconds is, I think we all agree, a conservative value. I'm planning to commit a 30 second interval as a starting point (2 polls per minute instead of 1), with a configurable override in the config file should you want to poll more (or less) often.

This should help us come to a consensus on a value that works for everyone as a starting default, and give everyone the option of overriding the value if their own scenario calls for it. I'll update shortly when the relevant changes are committed.

ngardiner, 2020-02-23:
6b47238 adds the following:

- Default 30 second (replaces 60 second) policy check interval, tunable using the policy->engine->policyCheckInterval parameter of the configuration file
- Default 10 second (replaces 60 second) EMS module caching time. The cache was always intended to avoid a situation where multiple references to the EMS module functions cause excessive polling to EMS modules. This is still useful, as we could tune the cache times on different platforms to guard against overloading an inverter/meter/battery system regardless of the policy check time.

This may not be the last tuning of these values that we do, but it does provide a framework to do so. I'll extend the same concept to each of the EMS modules shortly to allow tuning of the default cache time in the module configuration, which will ultimately allow full tuning of the polling intervals and allow these numbers to be refined with feedback from users of the various EMS systems.

ngardiner, 2020-04-26:
Tentatively closing this issue off, perhaps there will be use cases going forward for even shorter intervals. This can be configured in the config file, so it would only be if we wanted the default reduced or if we felt there were limitations to the tuning that we should need to make further changes. If so, feel free to re-open and we can discuss.

closed 2020-04-26 by ngardiner