ngardiner/TWCManager #34: Setup process needs improvement

ngardiner, 2020-02-28:
Currently, we ship a Makefile that does a half-install (it copies the main TWCManager.pl script to /usr/bin but does no more than this). This adds no value as the libraries on which the script is dependent are neither installed, nor located in an inbuilt search path.

This is a good opportunity to investigate a proper setup mechanism using setuptools, for example. Ultimately we could build a package ready distribution rather than having to install from source.

ngardiner, 2020-03-09:
This is now complete. The Makefile will now call setup.py which will use setuptools to install our entire distribution in the site-packages location for the python installation, as any other installed package would.

I have also added some conditionals to the Makefile to detect particular versions of Ubuntu, Debian or Raspbian which used the older php7.0 package vs the php7.3 package used in newer distributions. This allows both cases to work with the same Makefile.

I will close the issue now, however there is a related set of changes required per #3 which will document this process better for new users.

Anyone who used the old Makefile and still has the following file installed:

```
ls -l /usr/bin/TWCManager.py
```

This file is now redundant and can be safely removed:

```
rm /usr/bin/TWCManager.py
```

The new way to start TWCManager when installed with setuptools is:

```
python3 -m TWCManager
```

This will be reflected in the documentation shortly.

closed 2020-03-09 by ngardiner

MikeBishop, 2020-03-10:
In this new install paradigm, what's the best "quick deploy" option?  This is dramatically better for install-and-go users, but makes it a little more cumbersome to push a commit to my Pi and try it.

ngardiner, 2020-03-10:
Hi @MikeBishop,

The good news is that you can continue to run it directly from the source directory without needing to repackage each time:

```./TWCManager.py```

I'll put together a quick developer guide which will cover how to run it directly from a clone of the repo, and how to add additional dependencies which are automatically fetched by setuptools when setup.py is run.

MikeBishop, 2020-03-10:
Even better -- that means you can leave a stable version installed and revert to it without messing up the local repo.

MikeBishop, 2020-03-10:
Question for the developer guide....  If you re-run `make install` without changing the version number, does it actually deploy the new code, or does it decide the installed version is up-to-date?  If the latter, we might want to put a timestamp as a subversion to force it to reinstall every time.

ngardiner, 2020-03-10:
Good one, I'll add that to the developer guide. The good news is that it will update each time, the version is basically there as a manifest for the created package (it is used if it is uploaded to pypi.org to be installed with pip).