ngardiner/TWCManager #36: Charging rate too high when on 3 phase

notreallybob, 2020-03-02:
Hi,
Just installed and running TWCManager on rpi zero W.
My setup is 3 phase, 40A breaker, with solar generation monitoring by one device (Sunnyboy inverter) and consumption monitoring by another device (smappee). Since there were no modules fitting my use-case, I have frankensteined together a new MQTT EMS module to integrate with TWCManager. It seems to be working correctly, I think.

One of my problems is that when TWCManager calls setMaxAmpsToDivideAmongSlaves(), the actual slave usage is always about 3 times what was allowed. ie. If master calculated max as 5A, the slave's reportedAmpsActual would be about 15A. 
As a test, I modified TWCMaster.py to account for 3 phase electricity by simply dividing the amps by 3 in the call to setMaxAmpsToDivideAmongSlaves(). The result _appears_ to be correct, give or take a small fluctuations as generation increases or decreases. 
I have set wiringMaxAmpsAllTWCs=16, wiringMaxAmpsPerTWC=16, minAmpsPerTWC=3.
Please let me know if this was a reasonable thing to do, or if I am totally wrong and that the code already caters for 3 phase.

The second problem I'm seeing is that the charging is flapping on/off a lot instead of ramping the amps up/down as appropriate in response to the fluctuating solar generation. ie. scattered clouds causing frequent high and low generation.
I do see a few rare times the charging amps have adapted from, say 16A to 12A to 6A, but most often the TeslaAPI is called every minute or so to start and then stop the charging. I have set subtractChargerLoad to true, but the issue remains.

Third question is; Is there a minimum charge rate at which the slave will begin charging? I seem to only see charging begin when the minimum is 5A (or is it 6A?). This is independent of the minAmpsPerTWC setting. My slave heartbeat shows:
     SHB 1820: 09 00.00/06.00A 0000 0000  M: 09 00.00/02.85A 0000 0000
Is that telling me the min charge rate is 6A?

ngardiner, 2020-03-04:
Hi @notreallybob,

Great questions and hopefully we can get it sorted out for you quickly. 

> One of my problems is that when TWCManager calls setMaxAmpsToDivideAmongSlaves(), the actual slave usage is always about 3 times what was allowed. ie. If master calculated max as 5A, the slave's reportedAmpsActual would be about 15A.

That's strange, I'll need to have a look at the code and get back to you. The reportedAmpsActual during charging for a single slave setup should be the same as the ampsToDivideAmongSlaves, even in a 3 phase deployment. This is because for 3 phase, we're still limiting the number of amps that the vehicle charges at, however it is the number of amps **per phase**, so the total power is higher than a single phase TWC but still limited in the same way. It's not much different to how we limit both North America (110v) and Europe (230v) using the same code - the voltage and power differ but the amperage is controlled via the same mechanism.

> As a test, I modified TWCMaster.py to account for 3 phase electricity by simply dividing the amps by 3 in the call to setMaxAmpsToDivideAmongSlaves(). The result appears to be correct, give or take a small fluctuations as generation increases or decreases.

Interesting. I'll need to do more digging as above but it *shouldn't* be the case as the number of amps should not differ whether it is single or 3 phase, only the power output.

> I have set wiringMaxAmpsAllTWCs=16, wiringMaxAmpsPerTWC=16, minAmpsPerTWC=3.
Please let me know if this was a reasonable thing to do, or if I am totally wrong and that the code already caters for 3 phase.

This shouldn't be necessary, and in fact the TWC technically wouldn't be able to offer a minimum amps value of 3 (see below) *unless* there is some strange multiplication happening somewhere. I might leave this one as a pending question based on the findings of the above.

> The second problem I'm seeing is that the charging is flapping on/off a lot instead of ramping the amps up/down as appropriate in response to the fluctuating solar generation. ie. scattered clouds causing frequent high and low generation.

The behaviour you're seeing with charge on/off fluctuation shouldn't happen, so it does definitely need more investigation. Even with generation changes due to weather conditions, it would generally be a more smooth transition (ie, off when generation drops below minimum, on again afterward) than you're seeing. I wonder if there's a situation where the readings might come in at 0 from time to time causing the on/off behaviour? If you would like, I could take a look at your EMS module to see if there's any obvious situation where it might drop down suddenly?

> I do see a few rare times the charging amps have adapted from, say 16A to 12A to 6A, but most often the TeslaAPI is called every minute or so to start and then stop the charging. I have set subtractChargerLoad to true, but the issue remains.

subtractChargerLoad would be the first recommendation I would make in this instance, you have enabled that which is great. Having a look at the code, subtractChargerLoad in particular is not 3-phase aware, it only multiplies the current load by 240 whereas in your installation it would likely be 400v (depending on how your 3-phase TWC is wired). This may be contributing to the above as some of your charger load is almost certainly not being subtracted from the power consumption value. 

If you could please do me a favour and take a look at the following document:

https://www.tesla.com/sites/default/files/pdfs/wall-connector-eu/tesla-32a-wall-connector-installation-manual-en-EU-v1.pdf

Under Voltage and Wiring on page 5, would you be able to let me know which of the 3 wiring options your installation is using?

> Third question is; Is there a minimum charge rate at which the slave will begin charging? 

Yes, the minimum charge rate is limited by the TWC/Vehicle Onboard Charger to 5A in North America or 6A in Europe (ie, your TWC/Vehicle has an inbuilt minimum charge value of this value). The default TWCManager minAmpsPerTWC configuration of 6A to cover both these cases. Below minAmpsPerTWC, the TWC will still offer the physical min amps whilst the master will believe it is offering less, so minAmpsPerTWC serves as a cut-off at this point to instruct the vehicle to stop charging below this value.

> My slave heartbeat shows:
```
SHB 1820: 09 00.00/06.00A 0000 0000 M: 09 00.00/02.85A 0000 0000
```
Is that telling me the min charge rate is 6A?

Yes, the slave in this case is offering a minimum of 6A, whilst the Master is offering 2.85A per TWC. This is a side-effect of the minimum physical limitation of the TWC/onboard charger being 6A in your case.

MikeBishop, 2020-03-05:
> It's not much different to how we limit both North America (110v) and Europe (230v) using the same code - the voltage and power differ but the amperage is controlled via the same mechanism.

Side-note:  Most houses in North America have split-phase 240V service, which is then broken out into an assortment of 120V and 240V circuits.  North American wall connectors typically deliver 240V, modulo voltage drop along the lines.  It's possible to install them on a 120V circuit (or 100V, or 208V with a center-tapped three-phase setup) but it's very unusual.

ngardiner, 2020-04-26:
Hi @notreallybob,

There's a commit today that may help us further with diagnosing the issue you're seeing.

If you're still looking to solve it, could you please try the latest v1.1.8 branch, and look for the following output:

```
14:43:38 TWCManager 1 Slave TWC XXXX: Delivered xxx kWh, voltage per phase: (247, 0, 0).
```

If you could share that with us, it would be very helpful.

notreallybob, 2020-04-29:
Sorry, I gave up on the project. I debugged it plenty and also cdragon's version too. In the end, I just couldn't get past the charger's minimum turn-on of 5 amps. So whenever I had enough solar to trigger the charger, the end result was a power draw of 15 amps, then the code would try to readjust itself every minute down to 5-ish amps, or cloud cover would drop it below the minimum 5 amps. I could never get smooth charging, always on-off within minutes of each other. And then I realized I'm on net metering and don't need to charge only when the sun is shining.
In a nutshell, it would have been nice to control charging if I could convince the TWC to turn on at minimum of 3A total, but it hardware prevents it.

Keep up the good work!

MikeBishop, 2020-04-29:
Yeah, this only makes pure financial sense if you don't have full net metering.  If you do, it's still beneficial to the power grid because you avoid transmission losses in exporting and re-importing power, but there's no direct benefit to the customer.  It's a convenience + geek project at that point.

If you ever decide to give it another shot, I've added several features to help mitigate the on-off behavior; flex is probably the one most applicable to your situation.  Flex declares that charging may exceed green energy by `greenEnergyFlexAmps` in order to avoid a stop.  You could also use the ability to extend conditions to make Track Green Energy only apply when your solar system is generating "enough" -- I put a 2.5kW floor on mine with a 15 minute latch, so passing clouds don't interrupt charging but an extended overcast will.

Ultimately, though, it sounds like there are some baked-in assumptions that don't play well with three-phase TWCs, and we'd need someone with that setup to help us find and fix them.  There might also be a difference between single-phase and three-phase inverter setups alongside a three-phase TWC.

MikeBishop, 2020-04-29:
Actually, I'd bet the heart of the bug is in `getMaxAmpsToDivideGreenEnergy()`, where it's explicitly assumed that the relationship between Watts and Amps is division by 240.  That's going to vary based on regional electric properties.  Now that we're collecting the phase voltage, we could probably generalize that.

notreallybob, 2020-04-29:
@MikeBishop I played with that too. And I also made the voltage an updatable variable, fed from my inverter or electricity box monitor. It just didn't make an overall difference. The maths in calculating all the numbers is easy. It's the 5A trigger for 3-phase that's stopping me.

MikeBishop, 2020-04-29:
Well, for future reference when someone wants to try improving this area of the code....

The EU and APAC versions of the TWC (32A max) support the following installations:

- 230V single phase (either phase-to-neutral or phase-to-phase; TWC doesn't care)
- 400V three phase (400V between phases, 230V between each phase and neutral)
- 230V three phase ("neutral" is third phase, so 230V between phases; probably looks like two-phase to the API)

The NA version of the TWC (80A max) supports 208-240V from L1 to neutral, and can achieve that in several different ways, including two legs of three-phase supply.

`getMaxAmpsToDivideGreenEnergy` should perform all calculations in Watts, then check the TWC's reported voltage(s) to convert that to an amperage before generating a return value.  This would also improve accuracy somewhat for most non-US users, since they're calculating with 240V and actually supplying 220-230V.

ngardiner, 2020-04-29:
Good info @MikeBishop. It would be nice to substitute the 240v constant for live per-phase readings where we have them, although not all TWCs will support this, it is dependent on firmware version.

PR #102 (2020-05-27): Slave voltage
Mostly fixes #36.

This should properly handle single-phase and three-phase, using the actual detected voltage to do the Watts/Amps conversion.  If only some TWCs support voltage reporting, their voltages are used; if none do, the default can be set in the config file.  (Defaults are still 240V / 1 phase, of course.)

I'm unable to test this with a three-phase or multi-TWC installation, of course, but I think those scenarios are covered.

Of particular note, I have not attempted to handle the "two-phase" case, because I'm unclear what the math should be for that and I suspect it' [... cut]
closed 2020-05-27 by ngardiner