ngardiner/TWCManager #40: Chatty policy output at level 8, silence at level 7

MikeBishop, 2020-03-04:
It appears that all the output from the policy selection is level 8, everything from the chosen policy to the individual parameter comparisons.  It might be more useful to segment these across multiple levels, e.g.:

| New policy selected | Level 1 |
| Policy selection | Level 7 |
| Parameter comparison | Level 8 |
| Policy not selected | Level 8 |

ngardiner, 2020-03-09:
Good suggestion - this would make the debug output much easier to read

MikeBishop, 2026-10-26:
The policy logging now uses the levels proposed here: "New policy selected" at INFO, "All policy conditions have matched" at INFO7, and per-condition comparisons and non-matches at INFO8 (`lib/TWCManager/Policy/Policy.py`). Closing as done.

closed 2026-10-26 by MikeBishop