ngardiner/TWCManager #45: Policy Ordering

MikeBishop, 2020-03-09:
This is a follow-up to a comment on an already-merged PR that I'm not sure if you saw.  The current policy ordering is this:

- Charge Now
- User-defined `before` policy
- Green Energy
- Scheduled
- User-defined `after` policy
- Non-scheduled

I have two issues with this order, and I'd like to suggest a different one.  As a premise to both, the "normal schedule" can be thought of as a block -- Green/Scheduled/Other.  First, I'd like to suggest that this block is in the wrong order -- if Scheduled times overlap with the (necessarily broad) hours that Green Energy is checked, it's ignored.  I think Scheduled/Green/Other makes more sense for this "normal scheduled" block.

(Perhaps even better would be to say that if multiple policies match, take the highest amps allowed by any of them, but that gets really tricky working with background tasks.)

If we do that, then conceptually the units are Charge Now (i.e. an explicit command) and Normal Schedule.  Then the simple extensions to that that make sense are emergencies that justify overriding user commands, and exceptions to your normal schedule.  So to me, the more sensible policy order would be:

- User-defined `before` policy
- Charge Now
- User-defined `after` policy
- Scheduled
- Green Energy
- Non-scheduled

In order to avoid breaking anyone using the current order, a simpler change might be to add one more extension point, and make it:

- User-defined `emergency` policy
- Charge Now
- User-defined `before` policy
- Scheduled
- Green Energy
- User-defined `after` policy
- Non-scheduled

ngardiner, 2020-03-09:
Yes, I concur with the additional user defined policy block and I'll take that approach. I held off on moving the user defined policy before Charge Now, as Charge Now is a baked in feature of the web interface (and arguably only fires with specific user interaction), if it is broken by the first user-defined policy block it has a high likelihood of causing confusion for those who use the web interface.

That said, I wouldn't want to impede customization - I think that not only is the emergency user-defined policy block the way to go, I'm thinking doing away with the labels altogether (keeping them for reference but not using them as an insertion point) and allowing specification of the rule # to insert at is probably the best solution all around.

ngardiner, 2020-03-09:
The ordering of Scheduled Charging vs Track Green Energy definitely requires adjustment to ensure the behaviour we would expect. A scheduled charge should override tracking green energy.

MikeBishop, 2020-03-09:
If you add a user-defined slot between Scheduled and Green as well, then it's possible to insert anywhere in the list.  So that might be another route besides trying to provide number-based insertion.

PR #59 (2020-03-20): Rearrange policies, add emergency
Fixes #45, but keeps the extension point model for now.
closed 2020-03-20 by ngardiner