ngardiner/TWCManager #46: update_statuses() output inappropriate when not on green

MikeBishop, 2020-03-10:
Similar to #40, the output of update_statuses() is really only correct / relevant when tracking green energy.  Perhaps a different message should be shown less frequently when other policies apply?

(Note, several of these minor issues are as much notes to myself of things to get to later as requests for you.  Please don't mind picky items.  🙂)

ngardiner, 2020-03-21:
This should be better as of c070c46.

We now check the policy in use and if it is Track Green Energy, the output shows the energy generation and consumption values:

```
08:57:13 TWCManager 1 Green energy generates 0W, Consumption 0W, Charger Load 0W
08:57:13 TWCManager 1 Limiting charging to 0.00A - 0.00A = 0.00A.
08:57:13 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
```

If it's any other policy, it will show the current amperage set, as well as the minimum to charge. I've left the minimum to charge value in there, because for example you could have non-scheduled charging set to 2A and minamps set to 6A, in which case it would instruct the vehicle not to charge via the API:

```
08:59:13 TWCManager 1 Limiting charging to 32.00A.
08:59:13 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
```

MikeBishop, 2020-03-22:
The latter line is really only relevant if the selected value is below the limit.  Maybe something like

    1, "Limiting charging to {limit}A." if limit > minAmpsPerTWC else "Charging not allowed."
    7, "Limit of {limit}A is below {minAmpsPerTWC}A (minAmpsPerTWC)"

closed 2021-06-09 by ngardiner