ngardiner/TWCManager #86: Unable to login to Tesla API through Web Front end

neilrees, 2020-04-25:
On providing a username and password the following error is logged when the login button is clicked:

```
----------------------------------------
Exception happened during processing of request from ('x.x.x.x', yyyyy)
Traceback (most recent call last):
  File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 302, in do_POST
    self.process_teslalogin()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 343, in process_teslalogin
    carapi = self.master.getModuleByName("TeslaAPI")
AttributeError: 'HTTPControlHandler' object has no attribute 'master'
----------------------------------------
```

The browser displays: 
[image]

Running v1.1.8

ngardiner, 2020-04-25:
Thanks for the heads up, @neilrees. I'll take a look at this now.

ngardiner, 2020-04-25:
This should be fixed now in commit 715b259. Could you please try the following steps:

1. git pull to sync the update
2. Run setup.py again
3. Try running TWCManager again and try logging in with your Tesla credentials

neilrees, 2020-04-25:
Success!! 👍 Thank you!

ngardiner, 2020-04-25:
Great stuff, thanks for letting us know!

closed 2020-04-25 by ngardiner