ngardiner/TWCManager #98: Overriding current on the car's touch screen

AndySchroder, 2020-05-11:
I've noticed that if I override the actual current on the car's touch screen, it seems as though the max current seems to jump around, going to 21 and to 19, and then back up to the original setting. I'm guessing that this is TWCManager thinking that it needs to spike the amps in order to get the car to pickup the actual amps it is trying to tell it to use? When this happens, the car seems to then jump the actual current up and down from the setting I told it to use on the touch screen.

Is this a known issue?

ngardiner, 2020-05-11:
> I'm guessing that this is TWCManager thinking that it needs to spike the amps in order to get the car to pickup the actual amps it is trying to tell it to use?

Yes, TWCManager uses this mechanism to override circumstances where the vehicle is stuck at a lower charge rate, such as with a firmware update that was sent out in Oct 2017 which limited the vehicle to 6A, or in some cases where a vehicle limits itself under the belief that a charger is unreliable (or geolocation has the charge rate saved at a lower rate). It has been determined that 21A for US TWCs or 16A for EU TWCs spikes will reset these limits in vehicles.

Some examples:

   - https://forums.tesla.com/forum/forums/ummmm-anyone-else-suddenly-limited-32-amps
   - https://teslamotorsclub.com/tmc/threads/default-charging-reverts-to-5a-at-48a-home-charger.178093/

AndySchroder, 2020-05-12:
Has anyone been re-testing this phenomenon with recent software releases on the car, or just assuming it's still going on because you don't want to waste time re-investigating an old problem? Just wondering if the issue is isolated to a few software releases only.

Do TWC in networked in a standard configuration without TWCManager have the same issue? Seems weird that tesla would keep something broken with their own products. I don't see how people would tolerate this kind of inability to manually lower the max charge current in the car when networking TWC's. Also, I haven't noticed this kind of current spiking with a stand alone TWC with just the rotary switch set to a fixed current.

ngardiner, 2020-05-12:
TWCManager is managing the vehicle's charge rate from the TWC. If the question is does the software expect to co-exist with charge controls on the vehicle side the answer is absolutely not, we cannot directly manipulate those controls from TWCManager (or we would not need to wire anything at all) and as a result you'd see a loss of functionality by implementing something on the vehicle side which is a core feature of TWCManager.

The successful override of the vehicle's hard set limit is a feature, not a bug. It did what it intended to, it got the vehicle to consume the current offered. The fact that it worked leads to this issue existing at all. I'm not at all in favour of removing this feature as it helps those of us who for one reason or another have vehicles locked to a lower charge current and want them to consume the current offered, hence running this software in the first place. This is an ongoing problem, I've had it happen to me a number of times and my vehicle is a 2019 vehicle so yes, the condition can still occur and you can find threads discussing it on the forums.

I will happily implement an on/off switch for it but it will default to on, which is the current behaviour.

If the use case specifically was charge limit per vehicle rather than per TWCManager, it is pretty easily implemented. The problem is that in dissecting the method itself rather than the use case, it's very hard to suggest an approach that both meets your requirement and doesn't remove functionality from other people's deployments.

AndySchroder, 2020-05-12:
Okay, so maybe I misunderstood the purpose of the spike amps feature. If you don't have spike amps, is the car still offered the peak amps you tell the TWC, the car just may not use the peak amps unless you increase the setpoint in the car on the touch screen?

It sounds like your goal of TWCManager is to force the car to charge at a specific current. I was interpreting it to tell the car the maximum current it is allowed to draw. I think it is definitely value to have both options.

ngardiner, 2020-05-12:
>If you don't have spike amps, is the car still offered the peak amps you tell the TWC, the car just may not use the peak amps unless you increase the setpoint in the car on the touch screen?

Yes, but for a variety of reasons. One of those is a firmware version with a bug from some time back, another is if the car detects an issue with the charger (which can very easily happen with TWCManager, if you are starting and stopping the master handshakes repeatedly for example), and another is geolocation "memory" for chargers that the car tends to hang on to. All of these are considered sub-optimal as we want to avoid having to walk out to the vehicle to adjust the charging rate when we are offering a higher rate via the Slave TWC.

>It sounds like your goal of TWCManager is to force the car to charge at a specific current. I was interpreting it to tell the car the maximum current it is allowed to draw. I think it is definitely value to have both options.

In general terms yes - we can't force a car to do something it shouldn't/can't/won't, but what we would do is optimize the charging by overcoming any vehicle behaviour that runs counter to consuming available power. If you think of this in the context of green power, every missed opportunity to charge off of green power is a potential cost - drawing from the grid to increase SoC is in general more expensive than Green Energy charging, and that is the general motivation for people using TWCManager.

I'm 100% sure there are other use cases and we can accommodate them, however I am squarely in the category above (it effectively costs me 12c to charge off solar, that is my feed-in tariff, it effectively costs me 20-22c to draw from the grid, so I want to charge off of green energy at almost half the rate to halve my costs). If I were to turn this functionality off, and if my vehicle were to limit me to 6A when there were 12A available during daylight hours, I would add only half of the potential capacity during the cheaper energy production and instead export it to the grid for only 12c, yet if I needed the higher SoC and continued charging I would be drawing it from the grid for 20-22c and effectively paying an additional 8-10c per kWh.

AndySchroder, 2020-05-27:
Here is an example of a use case where the car owner may want to override the current that the TWCManager offers: http://andyschroder.com/DistributedCharge/ . In my initial prototype, the energy sale rate is independent of the power, but in future releases, the seller could offer a more complex rate structure to the buyer and it would be up to the buyer to determine how quickly they would like energy delivered. If the buyer could adjust the current in their car on the screen, my devices will automatically adjust the frequency that payment requests and payments are made.

MikeBishop, 2020-05-27:
That's a really neat application of this, @AndySchroder.  Love the work on the CAN bus!  (I will note, though, that I don't see any acknowledgement of this project on either your web page or the GitHub page with your code.)

AndySchroder, 2020-05-27:
I have listed on http://andyschroder.com/DistributedCharge/ that a fork of TWCManager is one of the "Special Software Dependencies". It is under "python 3.8". As mentioned in https://github.com/ngardiner/TWCManager/issues/90#issuecomment-634388624 , I'd like it if my required changes could be merged into this upstream branch. I'd rather reference your original branch as there is some excellent work being done here every week.

MikeBishop, 2020-05-27:
Sorry if I missed it before!

AndySchroder, 2020-05-27:
No problem, I just didn't want you to think you were being intentionally hidden. There is some great work going on with TWCManager and I'd love to see it continue. There are so many ways these projects can work together, especially helping those generating electrical power at specific times of the day.

ngardiner, 2020-05-29:
That is neat, @AndySchroder. I can also see how it makes sense in the context where the ownership of the charging service and the vehicle differ. In cases like this, it would certainly make sense to have a switch to actively avoid overriding vehicle limits. I appreciate the use case example for this one, it illustrates it very clearly

MikeBishop, 2026-10-26:
The promised switches exist: `spikeAmpsProactively` and `spikeAmpsReactively` can each be turned off in the debug settings (eaae7ae1 / 4e1f0108, v1.2.2). Closing as done.

closed 2026-10-26 by MikeBishop