ngardiner/TWCManager #100: API Wish:  Charging history

MikeBishop, 2020-05-18:
I'm building something separate that will display a graph much like what the Powerwall app displays -- where energy came from, where it went -- but I'd like to support pointing to a TWCManager instance and showing car charging as a separate line on the graph from house consumption.  In order to do that, I'd need to be able to grab the charge current over the last ~24 hours of history.  (I think the Tesla API reports history with averages over 5 minute intervals, so I'd probably do the same here.)

I'll likely build this myself, but if you have suggestions about where to hook in, feel free to suggest!

ngardiner, 2020-05-18:
Good one, @MikeBishop. I did have a idea for the data storage part of the equation, not so much the api side though. What I have been considering is a new module class, perhaps called Logging which would then allow one or more modules to be loaded simultaneously which would all get passed the statistics (charge rate, cars connected etc) that are currently printed to screen on a regular interval.

Each of the modules could individually control their interval, so it may even be that the information is passed more regularly than that (eg every few iterations of the loop) and it is outsourced to the Module to decide when to write the values out?

One of those Logging modules would be an (optional) console output module which would provide the current functionality around printing status updates to the screen, but for those who would rather run it as a daemon there could be CSV, sqlite, mysql or postgres modules which would store the values in a flat file or database for later graphing or interrogation, such as through an api interface as you've suggested.

It's a pretty easy thing to get started on as it really only requires moving the regular output function to its own module to start with, and then calling each module of type eg. Logging in a loop to pass the values from the main loop, and each could then store the data in their own defined back-end.

If there's a uniform way of doing it, the web interface could call the API and display some stats on the front page if a supported Logging mechanism exists.

What do you think? Would it provide a useful backend for storing the data that you want? You could of course store the data in memory, but given the platform of choice is the rpi it could end up being somewhat of a constraint, with 17,280 samples being stored at 5 minute intervals for a 24 hour period. There's no reason an in-memory cache could not be a Logger module of its own, it could define 5 minutes as the interval it records at, store it in a list or hash in a circular buffer style and then implement a standardised get/set interface so it could be swapped out for an on-disk Logger interface if someone preferred longer retention periods.

MikeBishop, 2020-05-18:
My original thought had been to restore the accumulating list of samples in each TWCSlave instance, then every five minutes capture the list and average it for retention.  Capturing that data with incoming update calls instead would be fine, but I wouldn't want most of the things currently written to the screen.

Nothing says I couldn't handle the call and ignore anything that's not relevant; getting called frequently and doing nothing most of the time is a pretty entrenched part of the current code flow already.  😉

ngardiner, 2020-05-19:
Okay that's clear, you'll poll a value and do the data storage externally. I don't think any of the above will be of use to you then, you can safely ignore it :) what you're proposing makes sense to me and should be a straightforward implementation. In fact, once it's part of the slave TWC module, it could be added to the current ```getSlaveTWCs``` API call very easily, just by adding a line around line 283 of ```lib/TWCManager/Control/HTTPControl.py``` to add a parameter to the ```getSlaveTWCs``` call.

PR #109 (2020-06-23): History
Had a chance to look at this over the weekend.  Fixes #100, plus some extraneous Black impacts.

This solution does increase writes of settings.json, but I think that's unavoidable if we want any persistence of data.  It minimizes the writes by:

- Only storing non-zero values; missing values are interpolated as zero by the response code
- Sampling every five minutes (for alignment with the Tesla API)

So at most, it's one write every five minutes while charging, and an idle system doesn't do any extra writes.
closed 2020-06-23 by ngardiner