ngardiner/TWCManager #101: Stale information on web interface

neilrees, 2020-05-22:
I keep having to restart TWCManager in order to have live information on the Web interface, although I suspect that the stale web interface is a side effect of something core stopping running or getting blocked.

I currently have TWCManager in a failed state so if there are any fault finding steps I can carry out happy to investigate.

From the logs I can see that logs from category `TWCManager` have stopped, as have logs from `TeslaVehic`, but other logs are continuing to be generated.  These logs were generated every 60 seconds and then stop, at around the time they stop this error is generated:
```
May 20 22:01:01 twc python[19637]: 22:01:01 HASSStatus 4 Error connecting to HomeAssistant to publish sensor values
```

However that was not the first time that error is logged, so I'm not convinced it's related.  Full section of log from when the TWCManager log stops:
```
May 20 22:00:23 twc python[19637]: 22:00:23 TWCManager 1 Limiting charging to 0.00A.
May 20 22:00:23 twc python[19637]: 22:00:23 TWCManager 1 Charge when above 5A (minAmpsPerTWC).
May 20 22:00:23 twc python[19637]: 22:00:23 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_config_min_amps_per_twc (value 5).
May 20 22:00:24 twc python[19637]: 22:00:24 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
May 20 22:00:24 twc python[19637]: 22:00:24 TeslaAPI   8 car_api_available returning True
May 20 22:00:49 twc python[19637]: 22:00:49 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_amps_max (value 0.0).
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Evaluating Policy match (settings.chargeNowAmps), condition (gt), value (0), iteration (<built-in function iter>)
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Policy conditions were not matched.
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Evaluating Policy match (checkScheduledCharging()), condition (eq), value (1), iteration (<built-in function iter>)
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Policy conditions were not matched.
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Evaluating Policy match (tm_hour), condition (gte), value (6), iteration (<built-in function iter>)
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Evaluating Policy match (tm_hour), condition (lt), value (20), iteration (<built-in function iter>)
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Policy conditions were not matched.
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     8 Evaluating Policy match (none), condition (none), value (0), iteration (<built-in function iter>)
May 20 22:00:53 twc python[19637]: 22:00:53 Policy     7 All policy conditions have matched. Policy chosen is Non Scheduled Charging
May 20 22:00:53 twc python[19637]: 22:00:53 TWCManager 1 Limiting charging to 0.00A.
May 20 22:00:53 twc python[19637]: 22:00:53 TWCManager 1 Charge when above 5A (minAmpsPerTWC).
May 20 22:00:53 twc python[19637]: 22:00:53 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_all_max_amps_for_slaves (value 0).
May 20 22:00:53 twc python[19637]: 22:00:53 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
May 20 22:00:53 twc python[19637]: 22:00:53 TeslaAPI   8 car_api_available returning True
May 20 22:00:59 twc python[19637]: 22:00:59 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_state (value 0).
May 20 22:01:01 twc python[19637]: 22:01:01 HASSStatus 4 Error connecting to HomeAssistant to publish sensor values
May 20 22:01:08 twc python[19637]: 22:01:08 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_cars_charging (value 0).
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Evaluating Policy match (settings.chargeNowAmps), condition (gt), value (0), iteration (<built-in function iter>)
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Policy conditions were not matched.
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Evaluating Policy match (checkScheduledCharging()), condition (eq), value (1), iteration (<built-in function iter>)
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Policy conditions were not matched.
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Evaluating Policy match (tm_hour), condition (gte), value (6), iteration (<built-in function iter>)
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Evaluating Policy match (tm_hour), condition (lt), value (20), iteration (<built-in function iter>)
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Policy conditions were not matched.
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     8 Evaluating Policy match (none), condition (none), value (0), iteration (<built-in function iter>)
May 20 22:01:24 twc python[19637]: 22:01:24 Policy     7 All policy conditions have matched. Policy chosen is Non Scheduled Charging
May 20 22:01:51 twc python[19637]: 22:01:51 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_amps_max (value 0.0).
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Evaluating Policy match (settings.chargeNowAmps), condition (gt), value (0), iteration (<built-in function iter>)
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Policy conditions were not matched.
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Evaluating Policy match (checkScheduledCharging()), condition (eq), value (1), iteration (<built-in function iter>)
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Policy conditions were not matched.
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Evaluating Policy match (tm_hour), condition (gte), value (6), iteration (<built-in function iter>)
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Evaluating Policy match (tm_hour), condition (lt), value (20), iteration (<built-in function iter>)
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Policy conditions were not matched.
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     8 Evaluating Policy match (none), condition (none), value (0), iteration (<built-in function iter>)
May 20 22:01:55 twc python[19637]: 22:01:55 Policy     7 All policy conditions have matched. Policy chosen is Non Scheduled Charging
May 20 22:01:59 twc python[19637]: 22:01:59 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_state (value 0).
May 20 22:02:09 twc python[19637]: 22:02:09 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_cars_charging (value 0).
May 20 22:02:15 twc python[19637]: 22:02:15 TWCSlave   1 SHB 0301: 00 00.00/00.00A 0000  M: 05 00.00/00.00A 0000
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Evaluating Policy match (settings.chargeNowAmps), condition (gt), value (0), iteration (<built-in function iter>)
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Policy conditions were not matched.
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Evaluating Policy match (checkScheduledCharging()), condition (eq), value (1), iteration (<built-in function iter>)
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Policy conditions were not matched.
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Evaluating Policy match (tm_hour), condition (gte), value (6), iteration (<built-in function iter>)
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Evaluating Policy match (tm_hour), condition (lt), value (20), iteration (<built-in function iter>)
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Policy conditions were not matched.
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     8 Evaluating Policy match (none), condition (none), value (0), iteration (<built-in function iter>)
May 20 22:02:26 twc python[19637]: 22:02:26 Policy     7 All policy conditions have matched. Policy chosen is Non Scheduled Charging
May 20 22:02:52 twc python[19637]: 22:02:52 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_amps_max (value 0.0).
May 20 22:02:56 twc python[19637]: 22:02:56 Policy     8 Evaluating Policy match (settings.chargeNowAmps), condition (gt), value (0), iteration (<built-in function iter>)
May 20 22:02:56 twc python[19637]: 22:02:56 Policy     8 Policy conditions were not matched.
[... cut ...]
May 20 22:03:58 twc python[19637]: 22:03:58 Policy     7 All policy conditions have matched. Policy chosen is Non Scheduled Charging
May 20 22:04:02 twc python[19637]: 22:04:02 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_state (value 0).
May 20 22:04:12 twc python[19637]: 22:04:12 HASSStatus 8 Sending POST request to HomeAssistant for sensor sensor.twcmanager_0301_cars_charging (value 0).

```

No more logs from TWCManager after 22:00:53.  Any guidance in diagnosing would be appreciated. 👍

MikeBishop, 2020-05-22:
Sounds like the main thread is still happily processing away, but the background task thread hit an exception and is stuck.  Normally when that happens, it's because something on the background task threw an exception, but I don't see error output in that log.

Can you try running it foreground with `screen -L -S TWCManager python -m TWCManager`?  That should give you a screenlog.0 that contains all the output as well.  If you can find the exception, that's probably the shortest path to knowing what's crashing in your setup.

neilrees, 2020-05-23:
Thanks @MikeBishop, I'll try that and report back

neilrees, 2020-05-26:
The user it was running as did not have write access to `settings.json`.  But if any exception is raised in that loop the background thread exits.

[image]

MikeBishop, 2020-05-27:
Yes, I'd done some more aggressive error handling around the HTTP requests, especially cloud-facing ones.  I didn't do a wholesale try/catch block mostly because having an exception get that far was almost always a bug I needed to fix before submitting the PR, so I wanted it to break things.  😉

Still, #103 means it's still there in the log output, so easy enough to check.

closed 2020-05-27 by neilrees