ngardiner/TWCManager #104: VIN Whitelist

MikeBishop, 2020-05-26:
This would only be applicable to a subset of TWCs, since not all of them support reporting the VIN.  There was a Reddit post where someone was concerned about having the TWC outside, since anyone could pull up during the day and charge.  So here's a random idea:

Now that we support getting the VIN from the TWC, it would be fairly straightforward to check that the VIN matches one of the VINs on the Tesla account and/or an explicit list in the config file, and stop responding to that TWC if an unauthorized car starts charging.

The harder piece would be recovery; if we do that, we won't know when the offending vehicle is gone.  Maybe just clear it at the next policy change, which means that the Charge Now button on the web interface unlocks it?

ngardiner, 2020-05-29:
This would be a great feature

mattiasclaesson, 2021-01-04:
[(https://www.ncsc.gov.uk/blog-post/terminology-its-not-black-and-white)]

MikeBishop, 2021-01-06:
Fair enough.  Now, if we could just get Master and Slave out of the codebase....

mattiasclaesson, 2021-01-06:
> Fair enough. Now, if we could just get Master and Slave out of the codebase....

yeah, we do what we can!

ngardiner, 2021-05-03:
This has been implemented in #265

closed 2021-05-03 by ngardiner