ngardiner/TWCManager #106: VIN Query Missed

ngardiner, 2020-05-31:
While doing some testing around basic recording of charging sessions for vehicles, I've found that the current VIN detection may not work in all cases. I connected a vehicle to a Slave TWC and no VIN response was received from the Slave TWC when queried. I will need to do more testing to reproduce the actual behavior. Reconnecting the vehicle to the TWC allowed the VIN to be detected successfully. 

We may need to consider if a more robust mechanism is required, such as delayed subsequent queries of VIN if the first query is not responded to.

ngardiner, 2020-05-31:
An observation is that this appears reproducible when starting TWCManager where a vehicle is already charging. The catalyst for the VIN query is the transition of ```isCharging = 0``` to ```isCharging = 1``` for a Slave TWC. 

Slave TWCs start with ```isCharging = 0``` and would quickly make the transition to 1 on the first reading from the Slave TWC for which the amperage is >= 1. There is nothing wrong with this logic.

The most likely cause is that this transition occurs during initial heartbeat and policy evaluation, and is likely clobbering other messages on the wire due to RS485's lack of full duplex operation. This is easy enough to confirm, for the purpose of narrowing down this behaviour, however it's purely a suspicion for now.

One option would be to add a random (or warmup) delay to the single VIN query, although the nature of RS485 communication is that there's no greater chance of success in doing this, it just timeshifts the potential collision to what could be a slightly less congested timing.

The alternate, and more robust approach is to maintain a retry timer for the Slave TWC. We don't particularly want to continuously probe for VINs on TWCs without the capability, which could cause further collisions, but we could maintain a retry count of, say, 5 retries and a timestamp of the last attempt. Where a Slave TWC with ```isCharging = 1``` has no ```currentVIN``` and a delta of ```lastVINQuery + (60 seconds?)```, we deduct a retry and query again. It would make sense to shift these retries slightly (either random or adding the retry counter to the interval) to avoid identical timing with any other heartbeat.

MikeBishop, 2020-06-04:
We could also cache in settings.json whether a given slave has ever successfully provided a VIN.  If it hasn't, try 1-2x in case it's gotten a hypothetical firmware upgrade; if it has, keep trying periodically.

ngardiner, 2020-06-23:
That's a good idea @MikeBishop. One thing we do need to consider is that a vehicle may not be a Tesla, so even if the TWC itself supports VIN fetch, the vehicle may not.

I've gone ahead and committed an update that adds some extra resilience to the process. It will query the VIN up to 5 times, at least 1 minute apart, if no VIN is tracked by the slave TWC and a vehicle is charging. I'll need to find a way to replicate the behaviour I saw during testing so I can test it out some more, and I think adding some smarts around whether the TWC supports VIN fetch or not is also a good idea. It's a work in progress.

MikeBishop, 2020-06-23:
I thought that if the vehicle charging wasn't a Tesla, the TWC responded with all zeros?

ngardiner, 2020-06-23:
You're right, good catch - there's no reason that a TWC with the correct firmware would not respond to the VIN query. It will be trivial to add the flag you proposed to the settings file which I'll work on doing - perhaps just to keep it easy there could be two retry values, for example the current 5 for slave TWCs where we are unsure of their firmware support, and an additional number for those which have previously flagged support?

ngardiner, 2020-10-09:
Fixed in db56580

closed 2020-10-09 by ngardiner