ngardiner/TWCManager #112: Charge start/stop notification triggers

MikeBishop, 2020-06-29:
Currently, I have a home automation rule that monitors the Powerwall load, and if the load becomes excessive (7.5kW) without corresponding solar, it switches to grid power instead of battery power to avoid needlessly draining one battery into another.  Right now, it checks every five minutes; that's fine for detecting when to go back to battery power.  However, I'd like to have it switch immediately when charging starts.

We already have actions (check arrival, scheduling of check departure) that fire when charging starts/stops.  It would be nice to be able to provide a web hook, a URL that gets notified when charging begins and when it ends; for my setup, I really only care about these on non-green policies, but one reasonable way to do this would be a property on the policy that gets notified when charging begins.  Notification on charging end would be easy to add in the same way.

The obvious way to build this is simply a property on custom policies and an ability to add them for the built-in policies in config.json.  However, we have so many things like that, I'm almost wondering if there should be a more uniform way to specify property additions to built-in policies.  (Of course, since we already cut a version containing several of those, the more important concern might be not breaking those.)

Any thoughts on how you'd like to see this structured?

ngardiner, 2020-07-01:
I definitely think a standardised way of handling charge start and stop for policy rules is useful, the complex part is the webhook as it would not be a standard interface, unless it was presented as a selection of supported webhook interfaces that users could choose from. It's a tough one! If it's something that is very implementation specific, I couldn't really argue with making it an optional setting per policy rule and just supplying a custom event handler. Unfortunately I don't think there's any real universal approach for something like this.

MikeBishop, 2020-07-08:
Regarding the last part, I was thinking more of a universal approach for augmenting built-in rules, not so much for notifications.  So right now, we have `greenEnergyLatch` and `greenEnergyFlexAmps` for setting the `latch_period` and `allowed_flex` properties on Track Green Energy.  We have `greenEnergyLimit` et al. for setting the `charge_limit` property on all built-in policies.  We have the `restrictions` node to append things to the `match`/`condition`/`value` entries.

Now I'm proposing something else that would probably look a lot like that, a property for custom policies plus config entries for the built-in ones.  Looking at it as "just one more," it's not a big deal.  If it's the last one we add.

But if we keep having things like this slip in....  At some point, it seems like it might just make more sense to be able to write something that looks a lot like a policy object, but gets merged with the built-in policy (append to arrays, override scalar values, add properties which are absent).  Of course, doing that would mean either breaking the things mentioned above or supporting two ways to do them, neither of which is ideal.

It's a meta-point to the actual web notification question, but something we should consider long-term.

PR #139 (2020-07-30): Web hook implementation
Fixes #112, albeit with a little more moved code than I wanted.

Defines four webhooks which can be set on any policy, built-in or custom:
- "enter" when the policy becomes selected
- "start" when any vehicle begins charging while policy is selected (but not fired if vehicle was already charging when policy was selected)
- "stop" when any vehicle stops charging while policy is selected (but not fired if vehicle stops charging after the policy exits)
- "exit" when the policy is no longer selected

Currently, this is always a POST with the same content as /api/getStatus;  [... cut]
closed 2020-07-30 by MikeBishop