ngardiner/TWCManager #113: systemd restart "rule"

beikeland, 2020-06-29:
is currently `Restart=on-abort`

this caused it to not restart for some for now undetermined reason last night. I would propose
```
Restart=always
StartLimitInterval=60s
StartLimitBurst=5
```

to ensure its restarted unless stopped by systemd, but also avoid spamming logs if something goes wonky?

dschuesae, 2020-06-30:
Agree, if it solves the problem.

beikeland, 2020-06-30:
`on-failure` might also work satisfactory. `always` should try to keep it running unless explicitly stopped with `systemctl stop twcmanager.service` or it exceeds say 5 restarts in 60 seconds. `on-failure` would allow a "clean" exit with kill -HUP iirc

The limit values might need some trial and error (based on how long it normally takes to start the process f.ex.) but I won't have access to the system for a while.

ngardiner, 2020-07-04:
Thanks @beikeland. I've updated with the Restart=always configuration you've proposed, however I'm not using systemd myself so it will get limited testing from me. I propose we leave this issue open for a while and get some feedback.

dschuesae, 2020-08-19:
It's working definitely better. No issues for a long time now.

beikeland, 2020-08-19:
I've not poked it much myself, but that I've not gotten a call because it stopped would suggest its at least restarting properly. I'll try to make a note to check logs if has been restarting next time I'm visiting

closed 2021-01-07 by ngardiner