ngardiner/TWCManager #115: Adjust charge rate based on real-time pricing

ngardiner, 2020-07-01:
Based off of Amber: https://www.amberelectric.com.au/pricing

The idea is that instead of increasing/decreasing based on green energy generation, the charge rate could be increased/decreased based on proximity to a target maximum real-time wholesale price. This is more a generic concept rather than an actual implementation as amber do not yet have an exposed API, but a module template to do this would be useful as I am sure there are similar platforms globally.

AndySchroder, 2020-07-03:
How frequently do you expect to adjust the charge rate?

ngardiner, 2020-07-03:
This particular provider changes the rate based on the prevailing wholesale rate for a given region on an hourly basis, although I could certainly foresee cases where the rate would be more real time than that, so it might be a matter of leaving polling interval to the individual interface

MikeBishop, 2020-07-08:
IIRC, I saw a thread on Reddit where the provider gave pricing in 5-minute increments, published an hour in advance.

ngardiner, 2020-07-11:
What I am thinking as a starting point would be to add a module class (Pricing) which then obtains real-time pricing data from an API interface. Generally I would expect the two key values of interest would be Import and Export prices, with a cent per kWh rate for each. Since this involves currencies there is every chance it could be much more complex than I am envisaging, but the idea would be:

   - Track the import and export pricing
   - Make this available to the policy rules

Two examples of where this could be useful:

   - Reduce / stop charging rate as export rate rises (more efficient to export than to consume)
   - Reduce / stop charging rate for non-green energy as import rate rises

In some instances, the import rate has the capability to rise into the dollars per kWh range where grid demand is beyond supply. In such a case there could be a cut-off where we just stop communicating with all slaves to ensure no charging occurs during the excessive pricing period.

mattclar, 2020-08-31:
I use amber and adjust my charging based on pricing... They also helpfully give a green-yellow-red rating on the current price vs upcoming prices. 
I can show you how I query their api

ngardiner, 2020-09-02:
Hi @mattclar - I'd certainly be interested in knowing more about interacting with the Amber API. It might make for a good framework as more providers offer flexible rates and real time APIs

mattclar, 2020-09-02:
@ngardiner honestly it's really just a hack that I've reverse engineered! Hopefully they are working on a real API

mattiasclaesson, 2021-01-08:
This is something that would be nice to get in place!

My provider tibber has this api to query for past, todays, tomorrows prices. Each hour has a dedicated price. 
https://developer.tibber.com/docs/overview
They buy and sell at this same price.

There is a integration for home-assistant also
https://www.home-assistant.io/integrations/tibber/

ngardiner, 2021-01-09:
I'll take a look, @mattiasclaesson. Thanks for the API reference!

ngardiner, 2026-11-02:
**Phase 1 of real-time pricing support has landed on `main`** (commit 7767998f), closing out PR #636 by forward-porting it onto the current architecture.

This first phase makes electricity price a **first-class signal available to the policy engine**, without changing the policy engine itself. A new **Pricing** module type exposes the current import/export price via the existing `modules.<Module>.<attribute>` macro mechanism — the same pattern Storm Watch and Powerwall TOU already use.

Modules available:
- **StaticPricing** — fixed import/export tariff, no network.
- **aWATTarPricing** — aWATTar hourly market pricing (EUR).
- **AmberPricing** — Amber Electric current import/export pricing (AUD), with site auto-discovery.

With this you can already drive charging off price today, by referencing it in a policy condition. For example, an emergency policy that stops charging when the import price spikes:

```json
{
  "name": "Import Price Too High",
  "match": [ "modules.AmberPricing.importPrice" ],
  "condition": [ "gt" ],
  "value": [ "config.maxImportPrice" ],
  "charge_amps": 0,
  "charge_limit": -1
}
```

All pricing modules are **disabled by default** — if none are enabled, charging behaviour is exactly as before.

What this phase deliberately does **not** yet do is the "look ahead and automatically shift charging into the cheapest window" behaviour — forecasting and cheapest-window scheduling aren't implemented yet (the modules don't advertise a Forecasting capability). That's the natural next phase. Keeping this issue open to track that.

One caveat: the Amber and aWATTar modules are unit-tested with mocked responses but haven't been validated against the live provider APIs with a real token yet — if anyone running those providers can give them a shakeout, feedback is very welcome.