ngardiner/TWCManager #125: astimezone() cannot be applied to a naive datetime

GMerg, 2020-07-05:
Hi,

Im running the latest version (dated 2020-07-05) and when trying to debug a problem of "not stopping charge with GreenEnergy policy" I saw this in log...

**TWCSlave   10 astimezone() cannot be applied to a naive datetime**

That same astimezone() is also used in TWCMaster so maybe check that as well.

ngardiner, 2020-07-06:
This is a known issue for python versions < 3.6. In your case, it has been trapped by a try block, and is reported at debug level 10, but it only impacts the ability to use the History API which isn't really a widely used feature currently - mainly due to not much documentation existing around it. So, I wouldn't worry too much, however an upgrade to python >= 3.6 is eventually going to be a requirement due to the number of issues we've been seeing with certain features not existing in python >= 3.3 which is our current minimum version.

Did you manage to find the issue around the Green Energy policy not stopping charge? Do you have your Tesla API key configured?

GMerg, 2020-07-06:
OK so this couldn't be the problem then.
I don't want to hijack this "Issue" reporting if it might be a config thing on my end so before knowing it can't be me, I will shut up and debug some more. Thanks for the support.

closed 2020-07-11 by ngardiner