ngardiner/TWCManager #126: Unable to STOP charge!?

GMerg, 2020-07-06:
Hi,

Do I read the log correctly that the script is trying to send a "STOP CHARGE" to the car (last 5 lines)? 

```
09:30:33 Policy     8 Evaluating Policy match (tm_hour), condition (gte), value (6)
09:30:33 Policy     8 Evaluating Policy match (tm_hour), condition (lt), value (20)
09:30:33 Policy     8 Evaluating Policy match (settings.hourResumeTrackGreenEnergy), condition (lte), value (tm_hour)
09:30:33 Policy     7 All policy conditions have matched. Policy chosen is Track Green Energy
09:30:33 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
09:30:33 TeslaAPI   8 car_api_available returning True
09:30:33 TWCManager 1 Green energy generates 0W, Consumption 0W, Charger Load 1211W
09:30:33 TWCManager 1 Limiting charging to 0.00A - 0.00A = 0.00A.
09:30:33 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
09:30:43 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
09:30:43 TeslaAPI   8 car_api_available returning True
09:30:43 TeslaAPI   8 startOrStop is set to stop
09:30:44 TeslaAPI   8 Car API cmd<Response [200]>
09:30:44 TeslaAPI   4 CARNAME: stop charge response{'response': {'result': True, 'reason': ''}}
```

It doesn't seem to work as the car is charging and not stopping at all!? Is this a bug on my CAR or something? Because my Powerwall is below 90% and should use the solar energy only to charge it till it's 90% and then deliver energy for the car.

ngardiner, 2020-07-06:
Yes, this looks like an issue to me. Might need to page @MikeBishop's expertise for this one, I'll also take a look, but the result being True would suggest the stop command was successful via the API to me so it's an interesting one.

MikeBishop, 2020-07-06:
It's not showing any generation or consumption data, but the car is drawing power. From TWCManager's point of view, it has no green energy to spare, so any power draw gets met with a Stop command. (Which appears to be successful.) So start with checking he Powerwall config.

GMerg, 2020-07-06:
Thanks for the reply.
Yes that was indeed a problem...PW2 EMS module was still disabled (forgot to enable it on a clean install/makeover). However, That STOP Command didn't work...the car was charging all the time.
Anyway, I will try to make a new log later this eve when there is no solar energy generated and hopefully I can reproduce this then.

GMerg, 2020-07-06:
OK here is another log...
```
20:27:52 Policy     7 All policy conditions have matched. Policy chosen is Track Green Energy
20:27:52 TWCManager 1 Green energy generates 172W, Consumption 1369W, Charger Load 1202W
20:27:52 TWCManager 1 Limiting charging to 0.73A - 5.78A = 0.02A.
20:27:52 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
20:27:52 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:27:52 TeslaAPI   8 car_api_available returning True
20:27:52 TWCManager 1 Green energy generates 189W, Consumption 1355W, Charger Load 1202W
20:27:52 TWCManager 1 Limiting charging to 0.80A - 5.72A = 0.15A.
20:27:52 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
20:27:56 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:27:56 TeslaAPI   8 car_api_available returning True
20:27:56 TeslaAPI   8 startOrStop is set to stop
20:27:59 TeslaAPI   8 Car API cmd<Response [200]>
20:27:59 TeslaAPI   4 CARNAME: stop charge response{'response': {'result': False, 'reason': 'could_not_wake_buses'}}

```

I did the following to produce this...
I set the ChargeLimit in th TeslaApp to 90% so that the car requests to get charged...that happens but it should have been rejected really.

MikeBishop, 2020-07-07:
Hm. So the API is telling TWCManager that it stopped the car successfully, except that it's obviously still going. From the lack of further slave heartbeats in the log, there's  no dip to zero and resuming; it just keesp trucking. As a sanity test, does the car actually stop charging if you stop it from the app?

Separately, I wouldn't begin to know how to get our house consumption that low, even in the winter. Kudos!

ngardiner, 2020-07-07:
Could it possibly be related to this? 

https://forums.tesla.com/forum/forums/stop-charging-button-no-longer-works

ngardiner, 2020-07-07:
@MikeBishop I noticed in the threads above one trick people used was to set the maximum charging SoC to the current SoC where stop charging didn't work. Might be something for us to think about, we could send a stop charging message first and perhaps after a grace time set the SoC limit to current SoC?

GMerg, 2020-07-07:
@MikeBishop my house consumption is so low due to the fact that it's Summer where I life. But we haven't much load in Winter either.
---
I have just tested the "**Stop charging**" button in the mobile app. You are right...that one **didn't work at all**.
I stood next to the charging port, pressed the button and I saw car loading and the TWC flashing...no sign of being stopped at all. However, in the app I shortly could read that the car was stopped but a second later I saw the stats again that it it charging and the "STOP Charging" button appeared again.
---
Using the "set charge limit" as an alternative would be a good idea I think. Even though you can only use that on a 50% minimum. If it's really a bug, it might stay there for a lot people as firmware is not outrolled equally. I have a Model-X in EU and here we get the firmware updates a lot later than in the US. Currently Im on 2020.20.1

MikeBishop, 2020-07-07:
> @MikeBishop I noticed in the threads above one trick people used was to set the maximum charging SoC to the current SoC where stop charging didn't work. Might be something for us to think about, we [...]

That assumes we know which car is connected. But we could use the minimum charge at home more generally, or simply set it to the minimum. Regardless, the car would declare itself done. I'd have to go through the logic to make sure the fallback doesn't get overwritten on the next cycle or stored as the new outside value, though.

ngardiner, 2020-07-07:
> That assumes we know which car is connected.

We already assume that in that we send the stop command to the vehicle. This is just a secondary mechanism if the stop command is ultimately unsuccessful.

MikeBishop, 2020-07-07:
Not really; we send the stop command to all vehicles if any of them start charging because we don't know which one to target.  #127 should handle this, I think -- it overrides the limit for the current policy until a new policy is selected or there's a call to start charging.  The limit is set to the lowest of any car believed to be at home.

@GMerg, could you give this a quick test?

    git remote add MikeBishop https://github.com/MikeBishop/TWCManager
    git fetch MikeBishop
    git checkout MikeBishop/limit_unresponsive
    make install

...and see if that fixes your issue?  I've done some quick testing by simulating the failure scenario, and I'm running it on my setup today to ensure it doesn't regress the API-responsive case.

GMerg, 2020-07-07:
@MikeBishop it seems to work; Thank you very much!
However, I one point you should change. It seems you set the charge limit to the exact % it is now. Maybe set it -1% of the one we have now to stop charge immediently? Because I saw the limit getting changed but it was then still charging with 6A for a long time.

```
17:58:38 TWCManager 1 Green energy generates 129W, Consumption 1210W, Charger Load 1191W
17:58:38 TWCManager 1 Limiting charging to 0.55A - 5.15A = 0.47A.
17:58:38 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
17:58:39 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
17:58:39 TeslaAPI   8 car_api_available returning True
17:58:39 TWCManager 1 Green energy generates 130W, Consumption 1219W, Charger Load 1170W
17:58:39 TWCManager 1 Limiting charging to 0.55A - 5.19A = 0.34A.
17:58:39 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
17:58:59 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
17:58:59 TeslaAPI   8 car_api_available returning True
17:58:59 TeslaAPI   8 startOrStop is set to stop
17:59:01 TeslaVehic 8 Car API cmd<Response [200]>
17:59:02 TeslaAPI   8 Car API cmd<Response [200]>
17:59:02 TeslaAPI   4 CARNAME: stop charge response{'response': {'reason': '', 'result': True}}
17:59:07 TWCManager 1 Slave TWC 0441: Delivered 3510 kWh, voltage per phase: (236, 0, 0).
17:59:09 Policy     8 Evaluating Policy match (settings.chargeNowAmps), condition (gt), value (0)
17:59:09 Policy     8 Policy conditions were not matched.
17:59:09 Policy     8 Evaluating Policy match (checkScheduledCharging()), condition (eq), value (1)
17:59:09 Policy     8 Policy conditions were not matched.
17:59:09 Policy     8 Evaluating Policy match (tm_hour), condition (gte), value (6)
17:59:09 Policy     8 Evaluating Policy match (tm_hour), condition (lt), value (21)
17:59:09 Policy     8 Evaluating Policy match (settings.hourResumeTrackGreenEnergy), condition (lte), value (tm_hour)
17:59:09 Policy     7 All policy conditions have matched. Policy chosen is Track Green Energy
17:59:09 TWCManager 1 Green energy generates 130W, Consumption 1219W, Charger Load 1185W
17:59:09 TWCManager 1 Limiting charging to 0.55A - 5.17A = 0.40A.
17:59:09 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
17:59:09 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
17:59:09 TeslaAPI   8 car_api_available returning True
17:59:09 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
17:59:09 TeslaAPI   8 car_api_available returning True
17:59:09 TeslaAPI   2 Attempting to apply limit of 87% to all vehicles at home
17:59:11 TeslaVehic 8 Car API cmd<Response [200]>
17:59:12 TeslaVehic 8 Car API cmd<Response [200]>
17:59:12 TeslaAPI   2 Set CARNAME to charge limit of 87%
```

MikeBishop, 2020-07-07:
Reasonable suggestion.  Anything lower than the current charge should suffice, so another option would just be to always set it to 50%.  Updated, if you do a `git pull`.

GMerg, 2020-07-07:
yes, should work fine now thanks

ngardiner, 2020-07-08:
> Not really; we send the stop command to all vehicles if any of them start charging because we don't know which one to target.

Yes, but I don't understand why sending stop commands to vehicles at home vs setting the SOC to stop them if they didn't respond to the stop command would be significantly different to the point that we'd highlight it over the untargeted stop command default - the reason I ask this is because if there's a genuine concern that I'm missing, we should probably disable the SOC limit fallback by default, however in my perhaps simplified view of the situation, if the behaviour today is to use the API to stop vehicles from charging and we send a stop command to vehicles at home who are currently charging and they fail to stop, the "less bad" option at that point seems to be to adjust the charge limit to or below current SOC, assuming we can revert the charge limit later. If we just let the vehicle keep charging, then it would be a worse user experience, as we're drawing power we don't want to draw per the policy.

Perhaps we just have 3 options in the Web UI. If we send a stop and the vehicle keeps charging we could: Let it charge (default - but should it be default? we send a stop by default), lower charge limit to 50%, or Lower charge limit to 50% only for connected VIN. Option 3 would not be particularly useful for those who have TWCs with firmware incapable of VIN detection, but would be more accurate for those who do.

ngardiner, 2020-07-08:
In fact, there's no setting needed. If we have a populated VIN, we stop just that VIN. If not, we use the current scheme. This is a simple modification to today's approach. The only leftover question I'd have is - do we lower charge limit by default or do we require opt-in for this feature? I admit I haven't looked into the PR enough yet @MikeBishop to understand any potential negative aspects of using this as a fallback, but if we have the restoration of charge limit working upon starting charge or charge session ending, I would think it would be a pretty safe bet to do this by default and remove any need for configuration?

closed 2020-07-08 by ngardiner

GMerg, 2020-07-08:
I think there is still some logic problem here.

When the charging is now stopped by the alternative command to set the limit, it would sit keep that limit the whole time I guess without resetting it to it's original limit once there is again enough power from the solar panels. I think this should also be considered to apply if the charge limite was previously set by the script by this alternative stop-method.

I hope I explained it correctly.

MikeBishop, 2020-07-08:
I believe that should happen with the existing code -- https://github.com/ngardiner/TWCManager/pull/127/commits/6b0aced102c38b97213854db15765dad8c919565 makes it clear the override when TWCManager sends a Start command, which should restore the "outside" limit (or the policy's limit, if you're using one) to affected vehicles.  If it's not, that's something to look at more closely.

GMerg, 2020-07-08:
Im sorry, it seems to work fine. I have tested it the whole evening and it switched the limit on/off according to the solar energy available. Just a bit later as expected sometimes but I guess thats noremal since it first tried the usual start/stop.

Again I have to thank both of you for the great work you do here. Im just not good enough in phthon to be of any help. 
I have already send @ngardiner my SolarEdge EMS module in shell-script the other day just to give back something to this project...if I can be of any help, let me know.

PR #127 (2020-07-08): Set charge limit if car ignores Stop API
This starts a ten-second timer when asked to stop charging.  (For my car, there appears to be about a two second gap between the first non-zero SHB message and the stop charge response, so I think ten seconds is enough buffer.  Easy to increase if not.)  If asked to stop again after the ten seconds are up but before all cars have stopped charging, the charge limit for the current policy is overridden so that the cars will consider themselves fully charged (modulo the minimum charge limit of 50%).  This lasts until the next policy change.

Some minor cle [... cut]