ngardiner/TWCManager #128: Feature Request: Stop only connected VIN if it is populated

ngardiner, 2020-07-08:
It would be good to check ```currentVIN``` when requesting vehicles to stop charging via API.

It's probably more complex than I am giving it credit for, though. For example, two Slave TWCs, one with VIN capability and one without. We drop below minAmpsPerTWC - what do we do? Perhaps what we need is a quorum where if all Slave TWCs are marked as supporting VINs, we only send stop messages to those VINs - otherwise we stop all vehicles per today's behaviour. Or is this just an overly complex rabbit hole to go down?

MikeBishop, 2020-07-08:
I think another way you could scope it, since we've already taken the step of keeping home/not-home status for each car, is to send the stop command to all cars at home, instead of all cars.

MikeBishop, 2020-07-08:
Unless you're envisioning a mode where one charger is allowed to proceed and the other isn't (which there is a use for -- two chargers, with available power between minAmpsPerTwc and 2*minAmpsPerTwc), I don't see a real use for limiting the stop command to the TWC's connected vehicle.

Actually, I take that back -- a garage with both a TWC and an outlet, and TWCManager shouldn't be stopping the car on the outlet.  That's fairly niche, though; I'd think if you're investing in TWCs and TWCManager, you have one per car you intend to charge at once.

ngardiner, 2020-07-11:
>  Default on, or just remove the setting?

I'm thinking default on to start with - that way we could pick up any corner cases with a workaround of switching the behaviour off if it's impacting anyone. 

This is a setting that I'd think was better in settings.json, default to on and toggle via Web Interface. What do you think of this approach, @MikeBishop?

- Check onlyChargeMultiCarsAtHome from the settings dict instead of config
- Add a toggle in the Web Interface
- On init, onlyChargeMultiCarsAtHome is either set to the config value or a default of on
- Remove onlyChargeMultiCarsAtHome from the config template in git

This way, anyone who currently has it configured retains their config, anyone who doesn't have it configured gets the default, and it can be toggled for everyone via the Web GUI?

ngardiner, 2020-07-11:
Few extra things to consider about this however:

- The above setting is currently on by default in the config template
- It helps in a situation where you have 2 vehicles in your API account and one is not home, but it is not evaluated for a single vehicle account
- How about if a Tesla which is not the TWC owner's is charging? That is where VIN detection could avoid stopping a single vehicle account which is charging elsewhere from being stopped when a different Tesla is using the Slave TWC. This may be a rare occurrence for some sites, and a common occurrence for others.

MikeBishop, 2020-07-13:
I honestly question whether that setting provides value, given that we use location for charge limits without a gate.  We should either have a "use location" toggle that's respected globally (though what should the behavior of charge_limits be if it's turned off?) or simply say that it's a part of the API like any other and doesn't need its own toggle.

I think the flow would look like:
- If TWC supports VIN:
    - If VIN is on account:
        - Send stop command to that car only
    - Else:
        - Send hard-stop command to TWC
- Else:
    - If any cars at are home, send stop command to at-home cars
    - If charging doesn't stop after a reasonable timeout, send hard-stop command to TWC

StefanSchuerchSBB, 2020-11-25:
I would highly appreciate if this feature would be realized soon!

MikeBishop, 2026-10-26:
Start/stop commands now target the VIN the TWC reports as connected, rather than every car on the account (1f92c041, v1.3.3; tightened further in #697, v1.4.0). Closing as done.

closed 2026-10-26 by MikeBishop