ngardiner/TWCManager #135: NameError: name 'time' is not defined

Gabtoof, 2020-07-26:
The following occurs quite often (at first it was a bit infrequent, at the moment it seems to be constant), and I'm open to any ideas of what I may be doing wrong. It seems to halt the app and the car won't charge.

18:01:05 TWCManager 1 Slave TWC 4864: Delivered 1321 kWh, voltage per phase: (251, 0, 0).
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/TWCManager/TWCManager.py", line 183, in background_tasks_thread
    carapi.car_api_charge(task["charge"])
  File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 624, in car_api_charge
    time.sleep(5)
NameError: name 'time' is not defined

MikeBishop, 2020-08-10:
@Gabtoof, have you had a chance to try this?  You'd want to do something like the following:

    git fetch origin
    git checkout 118_fix_time

...and then restart TWCManager.  If the errors still happen, I'll need to look more closely.  If they're gone, I'll merge that into the v1.1.8 branch.

ngardiner, 2021-05-03:
Closing this off as it is stale

closed 2021-05-03 by ngardiner