ngardiner/TWCManager #138: Unexpected behaviour + Powerwall 2 Issue

jakew89, 2020-07-27:
Thanks to all the development work that has gone into TWCManager - amazing effort.

I've have had TWCManager setup for the past week with a single TWC using a Powerwall 2 for generation/consumption readings (+ 4.5 kw solar system). While everything appears to be working based on the ouput, I'm encountering some unexpected behaviours:

1.) I have set TWCManager to "do not charge" for unscheduled charging. I assumed this meant that whenever I return home and plug in, the vehicle will not begin charging immediately. Unfortunately, however, everytime I do plug in, the vehicle starts to charge at the minimum 6A, regardless of whether there is sufficient green generation or not. I then have to tell the car through the Tesla app to stop charging, but I'm concerned this is then overriding the subsequent settings.

2.) I have set TWCManager to scheduled charge between 01.00 and 02.00 weekdays, and turned off all scheduled charging in the vehicle itself. Unfortunately, however, the scheduled charge is not actioned by TWCManager.

3.) I have changed config.json to start diverting power to the TWC above 80% except this has no effect. Instead, I have found I need to change MinSOE in TeslaPowerwall2.py to 80 in order for available generation to appear after 80% SOE. This normally works, but occasionally even after this SOE/SOC has been reached, 0W is still displayed for green generation. Does green generation only show when there is sufficient power to charge the vehicle and the SOE criteria is met?

4.) The vehicle will start charging at 32 amps if I click 1-day charge through the basic TWCManager portal, but when I click cancel it doesn't stop charging, it just reverts to 6 amps. I can only fully stop charging through the Tesla App.

Could someone help me to understand if I have the correct settings for the behaviour I am aiming to achieve, and if there is anything I have missed? Thanks!

Desired outcome:
- Vehicle to charge using a minimum of 6A when excess solar is available after the Powerwall reaches 80% SOE
- Vehicle to charge weeknights, for 1 hour, between 1 AM and 2 AM at 32 amps

15:04:17 HTTPCtrl   1 Serving at port: 8080
15:04:17 TWCManager 1 ImportError: No module named 'lib.TWCManager.EMS.Enphase', when importing module EMS.Enphase, not using EMS.Enphase
15:04:17 TWCManager 1 TWC Manager starting as fake Master with id 7777 and sign 77
15:04:19 TWCManager 1 32.00 amp slave TWC 2446 is ready to link.  Sign: A5
15:04:19 TWCManager 1 Set slave TWC 2446 protocolVersion to 2, minAmpsTWCSupports to 6.
15:04:20 Policy     1 New policy selected; changing to Track Green Energy
15:04:20 TWCSlave   1 SHB 2446: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
15:04:20 TWCManager 1 Slave TWC 2446: Delivered 1474 kWh, voltage per phase: (245, 0, 0).
15:04:22 TWCSlave   1 SHB 2446: 09 00.00/06.00A 0000 0000  M: 09 00.00/04.13A 0000 0000
15:04:23 TWCManager 1 Green energy generates 1373W, Consumption 380W, Charger Load 0W
15:04:23 TWCManager 1 Limiting charging to 5.61A - 1.47A = 4.14A.
15:04:23 TWCManager 1 Charge when above 6A (minAmpsPerTWC).

[33m15:06:02 [0m[32mHTTPCtrl  [0m[36m 1 [0mServing at port: 8080
[33m15:06:02 [0m[32mTWCManager[0m[36m 1 [0m[31mImportError[0m: No module named 'lib.TWCManager.EMS.Enphase', when importing module [31mEMS.Enphase[0m, not using [31mEMS.Enphase[0m
[33m15:06:02 [0m[32mTWCManager[0m[36m 1 [0mTWC Manager starting as fake Master with id 7777 and sign 77
[33m15:06:03 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:03 [0m[32mTWCManager[0m[36m 1 [0mSet slave TWC 2446 protocolVersion to 2, minAmpsTWCSupports to 6.
[33m15:06:03 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:03 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:04 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:05 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:15 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:15 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:15 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:17 [0m[32mTWCManager[0m[36m 1 [0m32.00 amp slave TWC 2446 is ready to link.  Sign: A5
[33m15:06:25 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'setNonScheduledAmps=0'', id 0, time 1595826385, type 2
[33m15:06:25 [0m[32mPolicy    [0m[36m 1 [0mNew policy selected; changing to [31mTrack Green Energy[0m
[33m15:06:25 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 08 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
[33m15:06:25 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'setScheduledAmps=32\nstartTime=01:00\nendTime=02:00\ndays=31'', id 0, time 1595826385, type 2
[33m15:06:25 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'setResumeTrackGreenEnergyTime=05:00'', id 0, time 1595826385, type 2
[33m15:06:25 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'chargeNow'', id 0, time 1595826385, type 2
[33m15:06:25 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'getStatus'', id 55950, time 1595826385, type 2
[33m15:06:25 [0m[32mTWCManager[0m[36m 1 [0mSlave TWC 2446: Delivered 1474 kWh, voltage per phase: (243, 0, 0).
[33m15:06:27 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 08 00.00/06.00A 0000 0000  M: 09 00.00/03.16A 0000 0000
[33m15:06:29 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 09 00.00/06.00A 0000 0000  M: 09 00.00/03.16A 0000 0000
[33m15:06:35 [0m[32mTWCManager[0m[36m 1 [0mGreen energy generates [35m1092W[0m, Consumption [35m324W[0m, Charger Load [35m0W[0m
[33m15:06:35 [0m[32mTWCManager[0m[36m 1 [0mLimiting charging to [35m4.49A[0m - [35m1.33A[0m = [35m3.16A[0m.
[33m15:06:35 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).
[33m15:06:55 [0m[32mPolicy    [0m[36m 1 [0mNew policy selected; changing to [31mCharge Now[0m
[33m15:06:55 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 09 00.00/06.00A 0000 0000  M: 09 00.00/32.00A 0000 0000[33m15:06:55 [0m[32mTWCManager[0m[36m 1 [0mLimiting charging to [35m32.00A[0m.
[33m15:06:55 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).

[33m15:06:57 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 09 00.00/32.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
[33m15:07:10 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 00 00.00/32.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
[33m15:07:16 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 01 00.00/32.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
[33m15:07:26 [0m[32mTWCManager[0m[36m 1 [0mLimiting charging to [35m32.00A[0m.
[33m15:07:26 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).
[33m15:07:26 [0m[32mTWCManager[0m[36m 1 [0mSlave TWC 2446: Delivered 1474 kWh, voltage per phase: (241, 0, 0).
[33m15:07:30 [0m[32mTWCManager[0m[36m 1 [0mCharge Session Started for Slave TWC 2424
[33m15:07:30 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 01 16.51/32.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
[33m15:07:34 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'getStatus'', id 57316, time 1595826454, type 2
[... 20 lines cut ...]
[33m15:08:59 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).
[33m15:09:27 [0m[32mTWCManager[0m[36m 1 [0mSlave TWC 2446: Delivered 1474 kWh, voltage per phase: (243, 0, 0).
[33m15:09:29 [0m[32mTWCManager[0m[36m 1 [0mGreen energy generates [35m922W[0m, Consumption [35m1914W[0m, Charger Load [35m1446W[0m
[33m15:09:29 [0m[32mTWCManager[0m[36m 1 [0mLimiting charging to [35m3.80A[0m - [35m3.80A[0m = [35m0.00A[0m.
[33m15:09:29 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).
[33m15:10:01 [0m[32mTWCManager[0m[36m 1 [0mGreen energy generates [35m950W[0m, Consumption [35m1894W[0m, Charger Load [35m1502W[0m
[33m15:10:01 [0m[32mTWCManager[0m[36m 1 [0mLimiting charging to [35m3.91A[0m - [35m3.91A[0m = [35m0.00A[0m.
[33m15:10:01 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).
[33m15:10:11 [0m[32mWebIPCCtrl[0m[36m 1 [0mWeb query: 'b'setNonScheduledAmps=0'', id 0, time 1595826611, type 2
[... cut ...]
[33m15:10:30 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
[33m15:10:32 [0m[32mTWCManager[0m[36m 1 [0mGreen energy generates [35m905W[0m, Consumption [35m444W[0m, Charger Load [35m0W[0m
[33m15:10:32 [0m[32mTWCManager[0m[36m 1 [0mLimiting charging to [35m3.76A[0m - [35m1.84A[0m = [35m1.92A[0m.
[33m15:10:32 [0m[32mTWCManager[0m[36m 1 [0mCharge when above [35m6A[0m (minAmpsPerTWC).
[33m15:10:32 [0m[32mTWCSlave  [0m[36m 1 [0mSHB 2446: 09 00.00/06.00A 0000 0000  M: 09 00.00/01.91A 0000 0000

MikeBishop, 2020-07-27:
Do you have the Tesla API set up in your config file (username and password)?  The things you're seeing that aren't expected would all be explained by an inability to access the Tesla API.

> 1.) I have set TWCManager to "do not charge" for unscheduled charging. I assumed this meant that whenever I return home and plug in, the vehicle will not begin charging immediately. Unfortunately, [...]

Because there is no effective "stop charging" command that the TWC can send to the car, stopping actually happens in two steps:
  - Turning the TWC to its minimum supported value (6A in many regions)
  - Calling the Tesla API to tell the car to stop

> 2.) I have set TWCManager to scheduled charge between 01.00 and 02.00 weekdays, and turned off all scheduled charging in the vehicle itself. Unfortunately, however, the scheduled charge is not actioned by TWCManager.

Similarly, a scheduled charge means telling the TWC to offer current, and then calling the Tesla API to tell the car to start charging.

> 3.) I have changed config.json to start diverting power to the TWC above 80% except this has no effect. Instead, I have found I need to change MinSOE in TeslaPowerwall2.py to 80 in order for [...]

Line 44 of TeslaPowerwall2.py should be overwriting the default value at the top of the file with whatever is in config.json; if it's not, you may need to double-check your config file.

The behavior is that (with some hysteresis), it will suppress the reports of generation until the Powerwall has enough charge; that is, it will behave (and say) that there is no solar power.  If the Powerwall goes below 76% (95% of target level), it stays suppressed until 84% (105% of target level) in order to avoid immediately dropping below the target level if there's sudden power usage.

> 4.) The vehicle will start charging at 32 amps if I click 1-day charge through the basic TWCManager portal, but when I click cancel it doesn't stop charging, it just reverts to 6 amps. I can only fully stop charging through the Tesla App.

See above; sounds like the Tesla credentials aren't configured / aren't working.

jakew89, 2020-07-28:
Thanks for the quick response @MikeBishop.

I originally configured the API token by entering my Tesla credentials through the IPC interface as I didn't see a way to enter it through the HTTP interface.

I've gone back over the config.json file in /etc/twcmanager/ but I cannot find the line where I should enter my tesla user and password. I'm in v1.2.0. Can you point me to what line I should be looking at?

Definitely appears to be some API interaction, as the car's maximum charging limit changes when I click start charging now through either IPC or HTTP interface, and reverts back when I try to stop the charge; but it doesn't actually start/stop the charge.

MikeBishop, 2020-07-28:
Sorry, I worded that the wrong way. It would be through the web interface - check the page on port 8080 and see if you can log in there.

Changing the max current is done via the charging connector, and is actually about the only thing that can be controlled directly.

jakew89, 2020-07-28:
Thanks again. 

So I checked the HTTP interface page (port 8080) and there is no window/box there to enter API details. I might have to start from the beginning as I've tried removing lighttpd and reinstalling TWCManager, but it seems to keep the old API settings I previously entered.

Also, what I meant was happening is that the maximum allowed SoC for the vehicle is being affected through the HTTP interface. For example, when I click charge now, the vehicle doesn't start charging but I can see the max allowable charge increases to 100%. Then I manually start charging the vehicle with the app, click stop through the HTTP interface and it doesnt stop, but the maximum charge defaults back to 80/90%. So there is some API connection to vary the allowable SoC, but failing to actually stop/start the charge. Maybe it is a new update on the API for Model 3's?

As mentioned, I'll start from the beginning again and see if I can get the API details to show up under the HTTP interface. Last time I couldn't find them there, so entered them through IPC (without port 8080), and I'm not sure where it has saved that token, but no matter how many times I delete TWCManager it still has them saved and doesn't allow me to enter the details/detail the old token.

Should the API user and password be entered through HTTP or IPC or it shouldn't matter?

jakew89, 2020-07-28:
Short update. I flashed raspi with buster to start again from scratch. Install went fine. Edited config.json in nano. All settings appear to have come across, including correct readings from powerwall2.

Loaded TWCManager at port 8080 through HTTP interface. Tesla API box appeared at the top; entered account details successfully.

Plugged in Model 3, which then registered for the first time, and TWCManager set long/lat for home.

Vehicle started charging straight away at 6A, despite no excess green energy being available. Waited to see if TWCManager would eventually stop the charge but continued after 10 mins so had to stop again using the Tesla App.

Clicked 'Start Charge Now' at 32A through HTTP interface, and battery SoC limit changed on vehicle, but would not start charging. Manually clicked start charging through Tesla App and it then started at 32A.

Then tried to cancel charge through HTTP interface, and charge rate was reduced to 6A, vehicle battery SoC limit changed, but vehicle continued to charge, so had to again force stop on the Tesla App.

No errors were thrown during install; nor shown during running script. 

Any ideas on why battery SOC limit can be varied (over API), but vehicle won't start/stop charging? Are there any API settings I should be changing?

MikeBishop, 2020-07-29:
Yeah, if it's picking up location and charge level, but not starting/stopping, I think you hit the bug I found last night. Check now that #141 is merged, and i bet you'll be up and running.

jakew89, 2020-07-29:
Spot on! Just did a reinstall and appears to be working. Should know overnight for sure if the scheduled charge works, and reasonably good solar in QLD this week so hopefully get to see it in action tomorrow!

Thanks so much @MikeBishop! Champion!

MikeBishop, 2020-07-30:
Glad to find it! And honestly, if I hadn't accidentally hit this myself, I'm not sure how I would have found it. It's a lot more likely to hit on a fresh install, so pure luck I triggered it on my own setup and then tried to figure out why my car didn't charge overnight. 😁

closed 2020-07-30 by MikeBishop