ngardiner/TWCManager #143: minAmpsPerTWC ignored, TWC told to charge at as low as 0.58 amps

JinbaIttai, 2020-08-05:
I'm really glad I found this project! 
I recently started modifying the original "draconventions" version to add smartmeter support etc, and then came across this project - which does most of what I wanted and much more.

I have come across some odd behaviour, if that's due to misconfiguration at my end I apologise, but they appear to be bugs in the code.  I'm also new to programming in Python so please excuse me if I'm way off base.

I have TWCManager tracking my Fronius inverter/smartmeter and configured to charge the car with any surplus energy instead of sending it to the grid.  

I have `"subtractChargerLoad": true` because my smartmeter measures everything to/from the grid, and `"minAmpsPerTWC": 6`
But for some reason the `minAmpsPerTWC` setting was being ignored so with less than 6A available the car starts to charge, ramps up to 6A and then is told to stop again as it's drawing too much - repeating every minute or so.

Here's log output from my car being told to charge at 0.58A this morning.
I changed some of the debug levels to get the relevant output without getting spammed by unrelated stuff:

```
07:52:36 TWCManager 1 Green energy generates 890W, Consumption 750W, Charger Load 0W
07:52:36 TWCManager 1 Limiting charging to 3.71A - 3.12A = 0.58A.
07:52:36 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
07:52:37 TWCSlave   1 desiredAmpsOffered (0.58) < minAmpsToOffer:6
07:52:37 TWCSlave   1 set_last_amps_offered(TWCID=01 B6, desiredAmpsOffered=0.58)
07:52:37 TWCSlave   1 SHB 01B6: 08 00.00/01.06A 0000  M: 05 00.00/00.57A 0000
07:52:37 TWCSlave   1 desiredAmpsOffered (0.58) < minAmpsToOffer:6
07:52:37 TWCSlave   1 set_last_amps_offered(TWCID=01 B6, desiredAmpsOffered=0.58)
07:52:39 TWCSlave   1 desiredAmpsOffered (0.58) < minAmpsToOffer:6
07:52:39 TWCSlave   1 set_last_amps_offered(TWCID=01 B6, desiredAmpsOffered=0.58)
07:52:39 TWCSlave   1 SHB 01B6: 01 00.00/00.57A 0000  M: 05 00.00/00.57A 0000

```

I think I managed to track this behaviour down to the relevant section of TWCSlave.py where it goes:
  `if desiredAmpsOffered < minAmpsToOffer:`

but then wraps most of the sanity checking code inside `if numCarsCharging > 0:`
If no cars are charging then it seems the `desiredAmpsOffered` is passed straight through to the TWC.
To test this theory I added an `else:` statement which sets `desiredAmpsOffered = 0` and so far that seems to work for me.

```diff -r -u TWCManager-1.2.0/lib/TWCManager/TWCSlave.py TWCManager/lib/TWCManager/TWCSlave.py
--- TWCManager-1.2.0/lib/TWCManager/TWCSlave.py 2020-08-05 13:25:15.989067466 +0800
+++ TWCManager/lib/TWCManager/TWCSlave.py       2020-08-05 13:31:56.827181614 +0800
@@ -769,6 +769,15 @@     
                         desiredAmpsOffered = 0
                     else:
                         desiredAmpsOffered = minAmpsToOffer
+            else:
+                # no cars are charging, and desiredAmpsOffered < minAmpsToOffer
+                # so we need to set desiredAmpsOffered to 0
+                self.master.debugLog(
+                    1,
+                    "TWCSlave  ",
+                    "no cars charging, setting desiredAmpsOffered to 0"
+                )
+                desiredAmpsOffered = 0
         else:
             # We can tell the TWC how much power to use in 0.01A increments, but
             # the car will only alter its power in larger increments (somewhere
```

MikeBishop, 2020-08-11:
That's a good improvement, but I'm surprised it works.  The TWC is not capable of offering less than its minimum, so the car will still see the minimum offer.  The expected behavior in that case is that the car will start charging, then stop -- and stay stopped.  The thing that's worth investigating is why it's restarting.

JinbaIttai, 2020-08-14:
> The TWC is not capable of offering less than its minimum

But what is the minimum my TWC supports?  
I have a feeling that because it's a three-phase TWC it can handle fewer amps (per phase) than the 5A assumed by TWCManager.

As a test I just hard-set `minAmpsTWCSupports` to `3` and my car is happily charging at 3A - the Tesla Mobile app shows 3/3 A.
Not that useful to me since I only have a single phase supply connected, but I can see the benefit for people with three-phase connections that want to track their solar exports.

MikeBishop, 2020-08-14:
That's interesting....  The code currently assumes that V2 TWCs support 6A and V1 TWCs support 5A.  If I send my TWC anything under 6A, it responds by saying it's set to 6A.  I wouldn't really recommend charging that slowly, but we also shouldn't be artificially inflating the minimum value.

closed 2020-08-14 by MikeBishop

JinbaIttai, 2020-08-14:
>  I wouldn't really recommend charging as slowly as 3A

Not on single phase, as that's only ~700W and the car likely consumes more than that to be "awake" and charging
But on three phase 3A = over 2000W (and equivalent to 9A single phase) 
And the current assumed minimum of 6A is over 4kW on three-phase, which is way too high unless you have a substantial solar installation - so I know people that have wired their TWCs single phase just so they can charge from solar at a lower charge rate.

ngardiner, 2020-08-16:
> But what is the minimum my TWC supports?
> I have a feeling that because it's a three-phase TWC it can handle fewer amps (per phase) than the 5A assumed by TWCManager.
>
> As a test I just hard-set [...]

Hi @JinbaIttai, is this a V2 installation? Going off the documentation, the traditional (ie non-load sharing protocol) amperage settings available for the DIP switches are between 6A-32A, which is the same as a Single Phase install:

https://www.tesla.com/sites/default/files/pdfs/wall-connector-eu/v2/tesla-32a-wall-connector-installation-manual-en-AU-v2.pdf

The test you did to set your TWC to charge at 3A, this was on a 3 phase TWC with 1 phase connected? Are you able to see how low of a rate you can specify to charge at?

JinbaIttai, 2020-08-16:
>  is this a V2 installation?

No, it's V1,  one of the very first batch of three-phase TWCs that became available in August 2016.

```

08:58:25 TWCManager 1 32.00 amp slave TWC 01B6 is ready to link.  Sign: A7
08:58:25 TWCManager 1 Set slave TWC 01B6 protocolVersion to 1, minAmpsTWCSupports to 5.

```

> the traditional (ie non-load sharing protocol) amperage settings available for the DIP switches are between 6A-32A, which is the same as a Single Phase install

That's right, the lowest available via DIP switches is 6A.

> The test you did to set your TWC to charge at 3A, this was on a 3 phase TWC with 1 phase connected?

Yes I only have single phase available here.

> Are you able to see how low of a rate you can specify to charge at?

I will do some more testing today.
I had the car charging at 1A earlier, but then it dropped to 0.70A and I discovered TWCManager treats <1.0 A as "stopped"

```
09:02:13 TWCManager 1 Charge Session Stopped for Slave TWC 0101
09:02:13 TWCSlave   1 SHB 01B6: 01 00.70/01.00A 0000  M: 05 00.00/00.00A 0000
```

I've just changed this to 0.3 and will try again.

`if slaveTWC.reportedAmpsActual >= 1.0:`

It would be useful if someone with three-phase can test this, as the car might be happier to charge at lower amp rates so long as it's getting enough kW to cover its operating loads.

JinbaIttai, 2020-08-16:
It's now been stable at 1A for the last 10 minutes

TWC reports it's drawing between 0.61A and 0.70A

```
09:54:54 TWCSlave   1 SHB 01B6: 00 00.00/00.00A 0000  M: 05 00.00/05.00A 0000
09:54:56 TWCSlave   1 SHB 01B6: 00 00.00/05.00A 0000  M: 00 00.00/00.00A 0000
09:54:56 TWCSlave   1 SHB 01B6: 05 00.00/05.00A 0000  M: 00 00.00/00.00A 0000
09:54:57 TWCManager 1 Solar 866W, Consumption 576W, Charger 0W
09:54:57 TWCManager 1 Limiting to 3.61A - 2.40A  + 0.00A = 1.21A.
09:54:58 TWCSlave   1 SHB 01B6: 01 00.00/05.00A 0000  M: 00 00.00/00.00A 0000
09:55:00 TWCSlave   1 SHB 01B6: 01 00.00/05.00A 0000  M: 05 00.00/01.00A 0000
09:55:02 TWCSlave   1 SHB 01B6: 01 00.00/01.00A 0000  M: 00 00.00/00.00A 0000
09:55:13 TWCManager 1 Solar 861W, Consumption 550W, Charger 146W
09:55:13 TWCManager 1 Limiting to 3.59A - 2.29A  + 0.61A = 1.91A.
09:55:30 TWCManager 1 Solar 852W, Consumption 552W, Charger 158W
09:55:30 TWCManager 1 Limiting to 3.55A - 2.30A  + 0.66A = 1.91A.
09:55:47 TWCManager 1 Solar 839W, Consumption 548W, Charger 168W
09:55:47 TWCManager 1 Limiting to 3.50A - 2.28A  + 0.70A = 1.91A.
09:56:03 TWCManager 1 Solar 827W, Consumption 560W, Charger 146W
09:56:03 TWCManager 1 Limiting to 3.45A - 2.33A  + 0.61A = 1.72A.
09:56:18 TWCManager 1 Solar 812W, Consumption 551W, Charger 158W
09:56:18 TWCManager 1 Limiting to 3.38A - 2.30A  + 0.66A = 1.75A.
09:56:35 TWCManager 1 Solar 800W, Consumption 556W, Charger 168W
09:56:35 TWCManager 1 Limiting to 3.33A - 2.32A  + 0.70A = 1.72A.
```

[The Tesla Mobile app shows the car is drawing 1A/1A but still says "Starting to charge"](https://i.imgur.com/8B243S3.png) 

Likely because it's not getting enough kW.  It would be good to try that on three phase.

PR #148 (2020-08-14): Always Honour minAmpsPerTWC
resolves #143