ngardiner/TWCManager #157: 2FA Authentication Changes

ngardiner, 2020-10-09:
Tesla are rolling out 2FA Authentication changes to the Tesla API authentication mechanism. These will shortly impact our Tesla API module. It looks like we need to provide a 2FA code parameter alongside the username and password.

MikeBishop, 2020-10-09:
Just to confirm, the 2FA code is required to get tokens with a username/password, but not required to refresh or use tokens?

ngardiner, 2020-10-10:
It's a little vague at the moment. From what I make of the following discussion:

https://github.com/timdorr/tesla-api/issues/215

- Newer versions of the Tesla app will not auth directly from the app to the API, but rather use an OAuth2 Authentication Server on a new URL to then authenticate them on behalf of the API. This leads to the token being issued by the OAuth2 server, not the API itself. OAuth2 uses an authorization and refresh token the way that the API does today, which suggests our behaviour won't have to change drastically as long as we can in fact leverage OAuth2 for this.

This would be the easiest implementation for us (and this would actually make the web interface even simpler) - to redirect the browser to the OAuth2 URL with TWCManager's URL as the callback URL, meaning we drop the part where we submit the auth credentials at all, redirect the user to Tesla's auth page and then wait for the tokens to be delivered via callback. 

The problem is that having tested this approach, others have found that the OAuth2 server throws a HTTP 400 error for any callback URL other than the Tesla API, which suggests that Tesla currently have it locked down to only provide auth tokens to the "offical" Tesla API app. This may change in future.

- One user found during their testing with the API that after enabling 2FA, they were able to authenticate to the API using the mfaCode parameter. This seems to have had mixed results for different people. Some see the same, some have no requirement to supply the mfaCode on their side. One user saw the behaviour after changing their paasword and now cannot authenticate unless they omit the mfaCode. This was all reported since I last caught up on the MFA code changes which at the time seemed to have been solved with the mfaCode parameter, but now I'm not so confident.

So it's going to be an interesting journey, I think :)

ngardiner, 2020-10-10:
> The 2FA code is required to get tokens with a username/password, but not required to refresh or use tokens?

Missing from my reply above but any way that we cut it (adding the mfaCode parameter or via OAuth2) you are correct, and in addition, Tesla would have to invalidate existing tokens for this to impact those who had already authed and recieved tokens, so the expectation at this stage is no major impact to current installations.

dschuesae, 2020-10-27:
My problem at moment is that car does not wake up anymore. Even not with the app. Could this be related? I switched to 2FA

ngardiner, 2020-10-31:
It could be - I see lots of reports of issues from those who have enabled 2FA on other projects such as TeslaMate. Not sure anyone knows definitively what is impacted by 2FA being enabled yet.

dschuesae, 2020-10-31:
Tried a lot yesterday. Reset whole installation. Removed 2FA. It won't wake up anymore. I have another idea that I will try today.

dschuesae, 2020-10-31:
> Tried a lot yesterday. Reset whole installation. Removed 2FA. It won't wake up anymore. I have another idea that I will try today.

As it's not related to 2FA - I will open a new issue.

MikeBishop, 2020-11-02:
FWIW, about a day after I enabled 2FA, TWCManager stopped being able to control anything through the API.  When I restarted TWCManager, it showed the credentials prompt in the WebUI as if I had never entered them; after entering my credentials again, it's back to working normally.  Teslamate seems unaffected as well, so far.

MikeBishop, 2021-01-29:
This is now critical; the v2 authentication endpoint is deprecated and removed.  Existing tokens continue to work (though they're also apparently rolling out IP blocking for people who generate too many requests; I think my Teslamate instance is moving to a VPN) but refreshes and new authentications with the password will fail.

Per https://github.com/timdorr/tesla-api/issues/215, people have figured out how to issue new tokens, but refresh is still unsolved.  In the meantime, the workaround for folks will be to manually generate tokens using http://registration.teslatasks.com/generateTokens and add them to `/etc/settings.json`.

ngardiner, 2021-01-30:
Thanks for the heads up, @MikeBishop 

It's interesting that teslatasks seem to have fixed the issue of oauth2 flow through some client-side javascript. I was quietly hoping that one of the Tesla python modules might implement the authentication flow so we could just leverage off of that (and avoid this same situation in the future) but taking a look at their github issues, all have this exact issue open right now, because the oauth2 flow for the new authentication workflow becomes a client side problem due to the current implementation:

- https://github.com/mlowijs/tesla_api
- https://github.com/tdorssers/TeslaPy

That said, the recent updates to the issue you linked to above do suggest that refresh of a token that has been fetched (currently requiring manual user intervention) is possible without user intervention:

```
    self.networkingAuth.post("/oauth2/v3/token", parameterType: .formURLEncoded, parameters:
        [   "grant_type" : "refresh_token",
            "scope" : "openid email offline_access",
            "client_id" : "ownerapi",
            "client_secret" : kTeslaSecret,
            "refresh_token" : "\(refreshToken)"]
```

So if we are able to implement this as the refresh logic, we may at this point only need users to hit the teslatasks URL once for initial fetch of the new token, and then manage refresh in the backend.

ngardiner, 2021-01-30:
Well having thought this over for a little while, I do have a plan. How about I add a text entry box in the Web UI settings to allow entering of an API token, along with a link (for now) to the teslatasks generateToken URL, allowing users to override the token in the Web UI rather than having to edit the settings file each time.

This could be an interim workaround, until a better option is available.

ngardiner, 2021-01-30:
5a7c7f2 implements the web UI change to allow overriding the bearer and refresh tokens

ngardiner, 2021-01-30:
There's proof of concept python code for handling the new login flow here: https://github.com/enode-engineering/tesla-oauth2 however it lacks MFA support currently.

MikeBishop, 2021-02-01:
Looks like TeslaMate has it working nicely, albeit presumably not in Python.

ngardiner, 2021-02-02:
@MikeBishop, could I please ask you to please give my last commit (d9170a3) a try with your Tesla authentication details, to see if it works for you? It is an implementation of the new auth flow based on the POC above, but without MFA support for now, however if MFA is not turned on for your account username/password should work.

It's working fine for my login, and I have an idea of what I need to add in order to get MFA working, just need to think about the web interface flow before implementing it.

MikeBishop, 2021-02-02:
Confirmed; when I disabled MFA, it's able to fetch new tokens via signin and interact with the API.  I left a few comments on the commit as well.

ngardiner, 2021-02-02:
Thanks for the confirmation @MikeBishop!

ngardiner, 2021-02-03:
3ee4dfb introduces a few new improvements to the login process:

- Better error handling: We now detect a few known errors/issues at several points during the auth flow (Phases 1 & 2, aligned with the steps 1 & 2 in timdorr's documented flow at https://tesla-api.timdorr.com/api-basics/authentication)

- Included in this is a catch for an error that is thrown for a Tesla account which is locked, which previously caused an uncaught exception. This is something that anyone testing the auth mechanism over the last few days will have run into a few times.

- A redirect mechanism for handling MFA, when support is added (soon). What will happen now is that if we detect an MFA flow during Phase 2 of the authentication process, we redirect the browser to a dedicated URL which will provide a list of supported MFA types and an input for the code. At the moment, the URL displays an error saying MFA is not supported

- From @one4many's feedback (thank you!) and testing, all UA headers are removed as they aren't needed for the auth flow.

ngardiner, 2021-05-08:
Good news, MFA login is now supported through #276, which will be merged shortly and will be part of v1.2.2. Testing is very much welcomed, it will be nice to have the new auth flow squared off.

jherby2k, 2021-05-21:
So, i'm on v1.2.1 (haven't tested 1.2.2) and i've noticed that, at least recently, i've had to re-authenticate to Tesla after every restart of TWCManager. Is this expected?

ngardiner, 2021-05-21:
> So, i'm on v1.2.1 (haven't tested 1.2.2) and i've noticed that, at least recently, i've had to re-authenticate to Tesla after every restart of TWCManager. Is this expected?

This would be because of a permissions issue, 1.2.2 will show an error in the WebUI if it can't write the ```/etc/twcmanager/settings.json``` file but 1.2.1 doesn't.  If you're running TWCManager as the ```pi``` user, the quick fix is:

```
chown -R pi:pi /etc/twcmanager
```

jherby2k, 2021-05-21:
Yup that fixed it, thanks! was owned by root:pi for whatever reason.

ngardiner, 2021-06-09:
Closing this issue out, as the latest release of TWCManager (v1.2.2) will support all of the Tesla auth changes that this issue was created to discuss - both the new oauth flow and MFA login.

closed 2021-06-09 by ngardiner