ngardiner/TWCManager #161: Feature: Ability to Count pure Powerwall2 as GreenEnergy to a certain %

GMerg, 2020-10-11:
Mayby this is already possible but I can't see how to easily do this.
The idea is to charge the car from the PW2 in case it's loaded to like 90% down to like 40%. 

Thats what I do at the evening now manually as there too low solar power to serve it with 6A+

I think the powerwall2 EMS module might be ideal to configure it for such task.

MikeBishop, 2020-10-12:
I have something like this configured on my personal setup, albeit for different reasons.  You could do it by adding a few additional things to your config file.  First, you want to create a policy that permits charging based on the Powerwall battery level:

    "policy":{
      "extend":{
        "after":[
          { "name": "Charge from Powerwall",
            "match": [ "modules.TeslaPowerwall2.batteryLevel" ],
            "condition": [ "gte" ],
            "value": [ 40 ],
            "charge_amps": 6
	      }
      }
    }

The effect of this would be that when Green Energy says it's done, this matches and continues charging at a certain rate as long as the Powerwall has some power.

You might also want to put a minimum output restriction on Track Green Energy to avoid it matching late into the evening but offering no power, like so:

    "policy":{
        "restrictions":{
          "Track Green Energy":{
            "match": [ "modules.TeslaPowerwall2.generatedW" ],
            "condition": [ "gte" ],
            "value": [ 1500 ]
          }
        }
    }

If you want to smooth over the gap between them (i.e. avoid two separate charge sessions), you could consider putting this as well:

        "greenEnergyFlexAmps": 6,

...which means that as long as the Track Green Energy policy remains active, it won't actually cut off the charge once started.

Now, if you're looking for something that says "If Powerwall is over 90%, go until Powerwall reaches 40%," that's not directly supported -- we don't have the concept of different entry/exit conditions for a policy.  The closest is latching, which says that a policy is treated as continuing to match for X minutes after it last actually matched.  So you could approximate it with something like....

    "policy":{
      "extend":{
        "before":[
          { "name": "Charge from Powerwall",
            "match": [ "modules.TeslaPowerwall2.batteryLevel" ],
            "condition": [ "gte" ],
            "value": [ 90 ],
            "charge_amps": 32,
            "latch_period": 120
	      }
      }
    }

...which says that if the Powerwall is over 90%, give the car two hours of 32A charging, which is probably enough to run you back down.  Or you could insert another policy earlier than this which matches on a Powerwall lower than 40% and says not to charge.

Once you get into the rabbit hole of custom policies, there's a lot of flexibility to build whatever behavior you want.

GMerg, 2020-10-12:
wow thats indeed all I need to look for. The different extra variables on the "match" part are all I need to script myself something.
Again @MikeBishop thanks for your details and excelent support. I really feel the need to donate something now...any BTC address or something?

MikeBishop, 2020-10-13:
Not at this time.  There are enough people who've contributed pieces of this, I'd be hesitant to accept an individual donation.  If you want to dive further into customization, see [the docs](https://github.com/ngardiner/TWCManager/blob/v1.2.1/docs/PolicyCustomization.md)

closed 2020-10-13 by MikeBishop