ngardiner/TWCManager #165: Scheduled Charging Flex start looks at current limit, not policy

MikeBishop, 2020-10-29:
It's slightly more complicated than this, because the car's _current_ charge limit might not be the charge limit applied by the Scheduled Charging policy.  You should check whether the Scheduled Charging policy has a limit set, and if not, use the car's current limit.

Even that's not totally correct, since a custom policy could check whether it's within the Scheduled window as one of its criteria; I haven't done that, but one could, especially if they're doing a fully custom policy set.  You'd almost want either the limit or the policy object to be an argument to this function, so it could always reference the right thing, but then you need to figure out which policy you're talking about in places where it's called.

_Originally posted by @MikeBishop in https://github.com/ngardiner/TWCManager/pull/159#r513474811_

MikeBishop, 2020-10-29:
I leave my car's limit set to 100% when it's away from home, and rely on TWCManager to set it to what the policy is going to use (e.g. Scheduled Charging to 50%). By looking at the car's current limit rather than the Scheduled Charging policy's limit, I suspect the code is going to get into this loop:

- Car's limit is 100%; calculate the start time needed to reach 100%
- Scheduled Charging begins and sets the limit to 50%
- The time needed to reach 50% means not starting yet, so Scheduled Charging is no longer the selected policy.
- Non-Scheduled Charging begins and removes the limit override; the limit returns to 100%
- Scheduled Charging begins, because we're past the start time needed to reach 100%
- ...

ngardiner, 2026-11-03:
Fixed in b8380fee (on main).

`getScheduledAmpsTimeFlex` sized the flex-start window from the car's current charge limit. As you noted, that isn't necessarily the limit the Scheduled Charging policy will apply — if the policy sets a `charge_limit` (e.g. `config.scheduledLimit = 50`) the car stops there regardless of its own setting. Sizing the window off the car's limit then starts charging far too early, which produces exactly the oscillation loop you described (car at 100% → start time computed for 100% → policy caps at 50% → no longer within the needed window → policy deselected → limit override removed → car back to 100% → …).

The fix adds `getFlexChargeLimit(vehicle)`, which prefers the Scheduled Charging policy's `charge_limit` when it resolves to a usable value (50–100, matching the band `enforcePolicy` actually applies) and otherwise falls back to the car's current limit. With no Policy module, or an unset / out-of-range policy limit, behaviour is unchanged.

Your broader point still stands for fully custom policy sets — a hand-rolled policy that gates on the scheduled window via its own conditions won't necessarily be the one named "Scheduled Charging", so passing the matched policy (or its limit) into the function would be more general. This fix handles the default policy set, which is where the reported loop occurs; the general case is left for a follow-up.

ngardiner, 2026-11-03:
Closing: the default-policy case that produces the reported oscillation loop is fixed in b8380fee (merged to main). The more general case — a fully custom policy set that gates on the scheduled window via its own conditions — is noted above as a possible follow-up and can be filed separately if someone wants it; it isn't needed for the behaviour reported here.

closed 2026-11-03 by ngardiner