ngardiner/TWCManager #182: Proposed GUI changes

hopfi2k, 2020-12-06:
I'd like to propose the following GUI changes for further discussion:

[image]

Looking forward for your comments!

hopfi2k, 2020-12-07:
Oh, and it's responsive too (for using on your phone/tablet etc.)
[image]

hopfi2k, 2020-12-07:
And some more ideas:

[image]

I consider all this as a starting point for discussions. Looking forward for your feedback!

hopfi2k, 2020-12-07:
Tooltips, maybe?
[image]

P.S. Added a small arrow before "Consumption" that shows the direction of electricity flow (red = from the grid, green = feeding surplus to the grid with arrow in opposite direction)

P.P.S. This is no mockup, but fully working code (except for the policy buttons...)

P.P.P.S. Need ideas for the "Chargepoint(s)" (or shall it be "Loadpoints"?) Anyone?

integlikewhoa, 2020-12-07:
I'm liking the fresh look

ngardiner, 2020-12-08:
Looks great @hopfi2k!

I'd personally like to take this opportunity to see some modularity with the web interface so that if it turns out other use cases (such as battery users) find that this doesn't suit their use case, users could switch between interfaces. It's natural imho, given everything else is modular.

How would you see this achieved? Perhaps it's just a matter of stylesheets or templates - for example we could use jinja2 templates to template the HTML output and allow switching between interfaces to allow for different use cases? perhaps we just create a templates directory and have templates as external files, and all the HTTPControl module would do is pass a dict to the template engine and call the specified template file.

Given you are mid development on the interface, if I was to pick up the templating piece for the front page as a start point and provide a switch between interfaces. would you be happy to refactor your work into a template file and we could then merge it even while it's WIP and get people using it?

hopfi2k, 2020-12-08:
Thanks very much!

Using [jinja2](https://jinja.palletsprojects.com/en/2.11.x/) would be the right step imo. Refactoring my code accordingly would be a piece of cake. 

Would you consider creating a new branch for developments concerning a new GUI?

Maybe introducing a simple HTTPControl2.py file that handles the new GUI stuff would be an easy way to "switch" between old and new GUI (perhaps with a variable in the config file) instead of doing it in the frontend (for now)?

P.S. If anyone can please suggest how the Chargepoint part could look like, I'd be more than happy for your ideas and inspirations (even hand drawn sketches would help)

MikeBishop, 2020-12-08:
I like the look, but what does the "Charge Policy" component show?  If it's indicating which policy is currently active, how would this respond to custom policies?  Do they need to add description elements?

ngardiner, 2020-12-08:
> Would you consider creating a new branch for developments concerning a new GUI?

We could absolutely do that, yes.  In fact that is underway with the charge scheduling GUI changes which could be merged into a new branch along with the GUI changes and delivered as one unit, as it really is just GUI logic. I'll get something set up and then we'll have the flexibility to get GUI improvements tested and merged in the near future, it's a high priority for me to improve the GUI experience

hopfi2k, 2020-12-09:
@MikeBishop thanks for your comment. Concerning the "Charge Policy" component allow me to point out that this a mockup and nothing final. More a starting point for discussion. My idea behind the "Charge Policy" was: a) give the user a visual indication of which policy is used and b) allow the user to quickly change the policy by clicking on it. Instead of having users editing configuration files, my approach was to offer "pre-defined" policies that suit most use cases. Whereas "Automatic" translates to "Track green energy", "Scheduled" would use a configurable schedule plan (maybe to be configured in "Settings"?), "Connected" would charge the car immediately when its plugged in and "Never" wouldn't charge at all. Again, this is all open to be discussed, I just wanted to through in something that we can start discussing.

@ngardiner Fantastic! Please let me know, once you have prepared the new GUI branch (or merged it with the schedule branch) and I'll immediately start working on it. Meanwhile, any input or suggestions for a new GUI is more than appreciated (my work was just to bring fresh ideas onto the table)

integlikewhoa, 2020-12-09:
Here is my thoughts or use cases for GUI and probably about features we don't have. 

1) 90% of the time I just want to charge when there is solar "tracking green energy". So that's like my default. 

2) But some days there is not enough solar and I have a trip tomorrow so I want to charge what I can when the sun is up today then I want to wait until after 10pm (Time Of Use plan) and finish charging when electricity is cheap. I think this is Scheduling in TWCManager which we currently don't have working. But I'm hoping I can set the OFF PEAK hours once and then have a GUI button to just turn on after if grabbed what it could from the sun the days I know I need it charged by the next morning. This probably gets tricky in my case with more then one car and more then one charger but I only need one charged at night from the grid.

3) Then very rarely I want to use the Charge Now function to just charge now because I have a trip today and I'm low on charge and I don't care about the cost of electricity. 

Since we polling tesla API we should have each cars current battery or amount of range in the GUI while its charging so we can see its charging at X amps and it's 80% charged or what ever. There is probably more info from the vehicleAPI we could pull and display to get us more info in one spot. 

For me the other things don't get used much. Different "charge now" rates are good but for the most part if I'm throttling its due to low sun but then I would just use "track green energy" and not charge now.  If I'm using the charge now function I probably only want max charging to get me charged in this situation.

Being able to Label the Chargers (I have more then one) would be a nice feature. 

Using VIN's to label the cars is probably going to be needed in future and seems like the current GUI has some options for "Vehicles connected" but it's not currently in use. For me like in situation #2 above I would need that info.

hopfi2k, 2020-12-09:
Thanks very much for your comments. Much appreciated. Good to know that others are basically having the same use cases as I do ;-) I do also most times use the "Track green energy" feature as it basically charges my car for free. Polling battery level from connected cars is easy and I do really want it in the new GUI.
Very interesting to know, that you are having more than one charger. May I ask how many cars do charge parallel at max?

integlikewhoa, 2020-12-09:
I have Two HPWC and two Tesla's at home. 

Both HPWC share a 100amp breaker. 

The car I drive, I recently have been charging at work now when I can. So I'm trying to limit at home charging except maybe weekends on this car. The other car always charges at home. 

I recently ran into some problems with the TWCManager due to multi cars and multiple charging locations since when I started running the python script, one car wasn't home but charging at work and it incorrectly set my home location to work when polling the location. So managing vehicles and chargers might not be for everyone but it would be useful for me.

integlikewhoa, 2020-12-09:
I don't think that answered the question. 

I only recently got solar but had the cars and chargers well before. So I used to charge only at night after 10mp both cars staggered (one at 10pm one at 1am) to be easy on the load.  but I have a 48 amp on board charger on one car and a 32 amp on the other. So I could charge both at max at the same time although I never did it much.  

Now I have solar 14.8kW (40 370w panels) and I moved over to track green energy and charge during the day. Even more recently I started charging one car at work when I can. So my charging is going down at home as I'm more cautious of being net zero at home.

ngardiner, 2020-12-13:
Hi @hopfi2k,

I've just committed a rather large change which:

- Takes all of the GUI presentation logic and pushes it out to jinja2 templates, which I have placed in the Default template directory, ie Default is today's GUI
- Every page/tab _except_ policy can currently be overridden with a theme. I'll get policy across ASAP, it's just written in a non theme friendly way so I'll need to adjust it a little to get it migrated.
- You can then go ahead and add your GUI as a theme, and users can freely switch between themes on the Settings page. As long as Settings works, people can switch to/from your theme, so there's no difficulty with merging your proposed GUI changes even if it's a WIP, and people can then switch to/from it as they like

One great thing about the way that jinja2 works is that it allows multiple paths in the template search order, and Default is always the second order, so if you don't provide a template for any given page it will automatically fall back to the Default theme's page.

Look forward to merging your theme soon! It will be nice to have a modern GUI option.

hopfi2k, 2020-12-13:
Hi @ngardiner,

This are fantastic news. Thanks a lot! I'll checkout your commit and start working on the GUI part. Will come back with a commit shortly.

I'm excited to help lifting TWCManager to the next level!

hopfi2k, 2020-12-17:
Hi @ngardiner,

just a quick update to report, that I'm making good progress on finishing the first version of the (incomplete) new theme with ninja2.
Should hopefully commit after the weekend

ngardiner, 2020-12-19:
Sounds great @hopfi2k, will be great to see it in action!

hopfi2k, 2020-12-30:
I'm late, I'm late, I'm late - sorry folks, but still busy with some other "last minute end-of-year things" I have to finish. Hope I can do a commit the next 2-3 days. Stay tuned.

hopfi2k, 2021-01-16:
I have somewhat complete the first iteration of the new 'Modern GUI'. What I really wanted to do is being able to start/stop individual TWC's (if there are multiples). From looking through the source code, this would require some changes at least in HTTPControl.py. Before screwing up everything, I'd like to know if some work has been done already to allow start/stopping of individual TWC's based on their ID? Nevertheless I'm going to commit my current work so work can continue based on this.

[image]

MikeBishop, 2021-01-16:
Yeah, at the moment the system is all-on or all-off with a certain maximum current for the system.  We don't currently have a concept of supplying a certain amount of current to a single TWC, unless the maximum for the system is greater than what that TWC is capable of delivering (80A maximum shared between two 48A TWCs, for example).

M-VdM, 2021-01-16:
On the _interwebs_, I found a couple of interesting "**Power Wheel**" components that might be useful:
- [power-wheel-card](https://github.com/gurbyz/power-wheel-card): An intuitive way to represent the power and energy that your home is consuming or producing. _(A custom card for the Lovelace UI of Home Assistant.)_ [(Blogpost)](https://community.home-assistant.io/t/lovelace-power-wheel-card/82374)
- [animated-consumption-card](https://github.com/bessarabov/animated-consumption-card) : Card with animated home consumption for Home Assistant Lovelace UI. _[(some other power wheel examples)](https://github.com/gurbyz/power-wheel-card/issues/23)_

BTW: this other [twc-controller](https://github.com/shreddedbacon/twc-controller) project also has a nice GUI ...

ngardiner, 2021-01-16:
Looking great, @hopfi2k! 

In terms of charge start/stop, there's a command that allows stopping individual TWCs, but the reason we haven't prominently implemented it is because of the way it works. It opens the contact, causing the charging to stop abruptly and without CAN bus communication to the vehicle, causing the vehicle to throw up an error and not start charging again if it is re-engaged. Unfortunately it's a known issue with the way the TWC command set is implemented (there's no appropriate command available unless we were to add hardware and modify the TWC wiring to support CAN bus communication), and is the reason we have the Tesla API mechanism to stop charging on the vehicle side via API command.

(There is also the ability to stop communicating with a TWC, which does stop it from charging, but that also has some side effects)

The downside of using the vehicle API is that it is hard to map vehicle to TWC. It's potentially easier given we now capture VIN (and could map VIN to TWC) but we don't yet do that, it's on the to do list of course, but it's a bit of a tenuous link and could result in the wrong TWC being stopped (by virtue of us trying to determine which vehicle is connected to it) if the TWC isnt reporting VIN or if it's not a Tesla connected to the TWC.

That said I may be missing the use case, if you wanted to for example disable a given TWC on the bus from offering charging, we don't have a built in function today but it wouldn't be hard for us to add a routine to ignore a given slave when doing the usual TWC routines, so if there's a particular use case you can let us know and we'll see how feasible it might be to add it in.

hopfi2k, 2021-01-17:
> Yeah, at the moment the system is all-on or all-off with a certain maximum current for the system. We don't currently have a concept of supplying a certain amount of current to a single TWC, unless [...]

Understood. Thanks very mich for the explanation!

hopfi2k, 2021-01-17:
@M-VdM Thanks for pointing to those site. Interesting stuff. Will have a look at it!

hopfi2k, 2021-01-17:
@ngardiner Thanks very much! Concerning starting/stopping of individual chargers, I think there were some changes in the Tesla Firmware a few months ago that allowed to stop charging if charge power was set to 0A...but i'm not 100% sure. Maybe worth a try. Other than that, shouldn't Tesla API meanwhile allow proper starting/stopping charging? https://owner-api.teslamotors.com/api/1/vehicles/:id/command/charge_start
The whole idea behind starting/stopping charging individual TWCs is, that TWCManager supports multiple TWCs and my GUI would even should them. To simplify usage, I was hoping to be able to set individual charge duration/amps for each TWC's and allow manual start/stop charging while still having a "global" start/stop charging button for all TWC's. I don't if we really need this, but wouldn't it be nice to have ;-) 
Anyhow, I may kindly ask you to approve my pull request (https://github.com/ngardiner/TWCManager/pull/197) so I can create a new one for the GUI theme (although this is still WIP, as it isn't fully responsive yet and has some other minor issues...). Thanks!

M-VdM, 2021-01-18:
> @M-VdM Thanks for pointing to those site. Interesting stuff. Will have a look at it!

I found a couple more ;-)

- @dslik > [svg-solar](https://github.com/dslik/svg-solar): Power flow visualization for solar power systems [(crazy online DEMO)](https://dslik.github.io/svg-solar/solar_demo.html)

- by [Maxem.io](https://maxem.io/):  _sleeeeeekk_
[image]

hopfi2k, 2021-01-19:
Great find @M-VdM ! Thank you very much. I do like @dslik version a lot, although I don't know if things like this are a bit of a overkill...?

hopfi2k, 2021-02-04:
@ngardiner I'd like to commit my current work on a modern UI (it still is far from perfect, but somehow does it's job) as WIP as I think there need a lot of things to be discussed and for sure to be changed/optimized. Would you like me to put that into a different branch or just commit it as WIP?

ngardiner, 2021-02-04:
Totally up to you how you'd prefer to do it @hopfi2k, but we do have a switch within the Web UI for which theme a user would like to use and it currently defaults to the old UI, so you could commit it directly as it is now if you'd like, which would allow people to voluntarily change their UI to the Modern theme and test it, and later on once you're happy with it, we could then flip the default over. 

If you'd like to do it that way, feel free to raise a PR and we can just merge it into the current v1.2.1 branch. All we'd need to do is make sure the template files go into a directory that is not ```lib/TWCManager/Control/themes/Default```, which is where the old UI and current default will be.

integlikewhoa, 2021-02-04:
Looking forward to testing it. I say keep it in this branch and get it out so we can test it.

hopfi2k, 2021-02-04:
Sure, I've already made the pull request, so you should be able to give it a try, once this has been accepted.
Don't expect too much, but I promise to keep working on it and incorporating the changes that will be suggested - I'm sure there are many areas that need to be discussed and improved.

mikey4321, 2021-02-06:
Hi @hopfi2k , @integlikewhoa, @ngardiner 

Just to revisit the charging scenarios, I think it would be good to expand it slightly for scenario #2, as depending on what tomorrow's driving looks like I want to be able to 
a. set the time it starts/stops (i.e. the scheduler and what is already in the works), 
b. also want to be able to change the charge % limit.  (the feature request here)

Maybe its a separate discussion, as I see there is already some functionality to save the a charge limits into the settings.json, but it would be really good if we could adjust the limits for each of the four charging modes via the GUI, instead of having to edit a config file.  

The scenario I have encountered today is it is the middle of summer so I've previously set the overnight charging low (50%) and greenEnergy limit high (90%) - eg have enough to do most things from the overnight coal power, then plenty of headroom to absorb all the solar if its a sunny day and the car is home in the garage.  But today is different - I've done lots of driving, and its raining, and tomorrow i want more charge as I need to travel.  Apart from manually starting it at peakhour, the only way to say "charge at night, and give me a bit more than usual" is to edit a config file or use the tesla app.  It would be good if this could all be done together via the GUI.  

 > Here is my thoughts or use cases for GUI and probably about features we don't have.
> 1)  90% of the time I just want to charge when there is solar "tracking green energy". So that's like my default.
>
>    2) But some days there is not enough solar and I have a trip tomorrow so I want [...]