ngardiner/TWCManager #184: Tracking Green Energy

addy199, 2020-12-10:
Hi ngardiner,
I am new for python programming.
I am using a 3-phase power for my TWC with a 16A breaker. I want get the green energy data from my SMA inverter. But I do not see any configurations for config.json for SMA inverters. What is the best way to get this working?

Thanks in advance.

hopfi2k, 2020-12-10:
May I step in...could you kindly let me know the model of your SMA inverter? Is there a Sunny Home Manager in the play?
There currently is no support for SMA inverters, but if your's is a recent model that supports the ModBus protocol, I could write a module to support it - if you are willing to help testing it ;-)

addy199, 2020-12-10:
Sure definitely. 
Model-SMA25000TL
Yes, it has a sunny portal which logs the generation and consumption data.

hopfi2k, 2020-12-12:
Thanks very much - cooking something, although it will take some days, due to time constraints (Christmas preparations)...

addy199, 2020-12-13:
No Problem @hopfi2k.

Saftwerk, 2021-03-10:
Hope anyone can help me on "Tracking Green Energy".

With this command, I can read the unused green energy power available, which would go to grid:
http://192.168.178.20:8080/api/data.txt?uuid%5B%5D=00000006-0002-1570-ffff-0123456789ab
The answer is text (without quotes): "4199.9 W"
Before sending that command again I must wait 2 seconds or more.

How do I put this into the TWCManager config?
Thank You!

Saftwerk, 2021-03-26:
Please, could anybody help?

ngardiner, 2021-04-18:
Hi @Saftwerk,

I have added support for this and am looking for your help to test it. Please make sure you've got the latest v1.2.2 sources:

```
git checkout main
git pull
```

And please configure the following in your ```/etc/twcmanager/config.json``` file under the sources section:

```
        "Volkszahler": {
          "enabled": true,
          "serverIP": "192.168.178.20",
          "serverPort": "8080",
          "uuid": "00000006-0002-1570-ffff-0123456789ab"
        }
```

And then hopefully you should see values. If not, could you please check for any errors on the console, for example;

```
23:30:04 Volkszahle 17 Error connecting to Volkszahler to fetch sensor values
```

And paste those errors here so I can see the output.

Saftwerk, 2021-04-18:
Hi @ngardiner,

here the results from today's tryout .
1. I do not see any errors.
2. I do not see if Volkszahler is used and connected.
3. I do not see see how much Green Energy is going to grid.

sudo systemctl stop twcmanager.service
cd TWCManager
python3 -m TWCManager
[image]

Here is my config.
[config.json (Saftwerk).txt](https://github.com/ngardiner/TWCManager/files/6331516/config.json.Saftwerk.txt)

I think the UUID is better to use for TWCManager to track energy: 00000004-0001-1670-ffff-0123456789ab as "Overall Total Grid Power"
It gives positive numbers "2109.7 W" when I draw power from the grid.
It gives negative numbers "-51 W" when I push power to the grid. (sun is almost down)

Question:
Is the setting **"subtractChargerLoad": true,** correct for this case?

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look. (For SMA support, see #222.)

closed 2026-10-26 by MikeBishop