ngardiner/TWCManager #188: Green Energy JSON / curl

Dustovsky, 2020-12-16:
Hey

First, I have to admit I'm very new in any programming and my knowledge is limited.
Since more than a year, I have the TWCManager from dracoventions running and I was able to set it up including green energy data from a "smart-me" smartmeter.

To get the data, I use the following curl-request in the python script

greenEnergyData = run_process('curl -u user:password -s -m 600 "https://smart-me.com:443/api/DeviceBySerial?serial=XYZ"')

Now, honestly, I have no clue how to implement this request in the new Fork of the TWCManager.
Maybe someone finds the answer in here: https://smart-me.com/swagger/ui/index

Sorry for being the naive newbie here :-) but I really like your work and solution for smart charging!

Cheers

Dusty

MikeBishop, 2020-12-16:
Seems like it would be fairly straightforward -- one could take one of the simpler EMS modules (maybe Fronius?), change the URL it's querying, and add it as a new module.  

I don't see that the sample response indicates both generation and consumption, though.  Do you have separate meters for grid energy and solar production, or does a different device type expose both?

Dustovsky, 2020-12-16:
well, the answer of the curl looks like the following one:

{
  "Id": "XYZ",
  "Name": "Solar-meter",
  "Serial": ZXY,
  "DeviceEnergyType": 1,
  "FamilyType": 1,
  "ActivePower": 0.00016,
  "ActivePowerUnit": "kW",
  "CounterReading": -1.921,
  "CounterReadingUnit": "kWh",
  "CounterReadingT1": -1.921,
  "CounterReadingImport": -1.921,
  "SwitchOn": true,
  "Voltage": 230.9,
  "VoltageL1": 230.9,
  "Current": 0.085,
  "PowerFactor": 0.01,
  "PowerFactorL1": 0.01,
  "Temperature": 8.15,
  "ValueDate": "2020-12-16T19:40:01.1507821Z"
}

whereas I use "ActivePower" as the parameter to determine the usable power. It's a simple smartmeter put close to the electric meter, so it only looks is there feed in or import. It works with the "old" TWC :-)

most probably I'm really to stupid, but in TWCManager/etc/twcmanager/config.json I do not see any URL I could change or adapt into a curl request...? Am I missing something?

MikeBishop, 2020-12-17:
Ah, so it's looking to see whether there's import or export, and trying to drive the export to zero.

No, we don't have anything that just takes a URL, though it's possible something might be generic enough across devices to do that.

Dustovsky, 2020-12-19:
The electric setup is even simpler:
The Wallbox is connected BETWEEN the electric meter and the smart-meter.
So it just takes the number from the smart-me and everything what is negative is export is green energy. So charging has no influence on this number...

so I have to find a way to implement this curl-request into one of these modules (or actually create a new one...) 
Looks like a challenging task ;-)

ngardiner, 2021-01-07:
Hi Dusty,

I've been a bit busy lately so I wasn't able to give this a try, but I'll work on a test implementation of this now, and let's see how it goes in your environment. I'll let you know when it is ready.

ngardiner, 2021-01-07:
Hi @Dustovsky, I've added a new module for smart-me, but will need you to do some testing for me as I don't have an account. If you could check out the doco here: https://github.com/ngardiner/TWCManager/blob/v1.2.1/docs/modules/EMS_SmartMe.md and let me know how it goes for you, that would be great.

Dustovsky, 2021-01-10:
@ngardiner Many thanks for your work!! I wasn't able to check so far, but will do as soon as possible and keep you adviced :-)

Dustovsky, 2021-01-18:
@ngardiner somehow I couldn't get it to work :-/
Maybe you can do a try with your setup?

That's what I put in the config.json:

 },
        "SmartMe": {
          "enabled": true,
          "username": "mail@chregu.com",
          "password": "test@TWC",
          "serialNumber": "06000839"
        },

You may use my credentials, it gets you to the home smartmeter...

Script is running, but I get 0 0 0 output:
09:09:51 TWCManager 1 Green energy generates 0W, Consumption 0W, Charger Load 0W
09:09:51 TWCManager 1 Limiting charging to 0.00A - 0.00A = 0.00A.
09:09:51 TWCManager 1 Charge when above 5A (minAmpsPerTWC).

ngardiner, 2021-01-21:
Hi @Dustovsky,

I've just pushed an update which should help. What I've done, to help make it clearer that a value is being read from the sensor, is to either interpret the value as a Generation value if the value is negative, or as a Consumption value if the value is positive. So if you're exporting power you should see generation value, and if you're importing power you should see a Consumption value, but never both

[image]

Could you please let me know how it's looking to you - does it represent generation vs consumption correctly based on your observations?

Dustovsky, 2021-02-11:
@ngardiner Hey, eventually I got time to test your module :-)
sweet, looks like it's working now! Many thanks for your help!!

ngardiner, 2021-02-11:
Great stuff, thanks for confirming!

closed 2021-02-11 by ngardiner