ngardiner/TWCManager #189: Error from unescape_msg function: 'bytes' object does not support item assignment

gescarra, 2020-12-20:
Every few days, and generally when a car is charging, the wall connectors start showing blinking red lights and I see the following timestamped every 5 seconds (two instances for the same error, one for each TWC I assume).

19:49:10 TWCManager 1 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 667, in <module>
    msg = unescape_msg(msg, msgLen)
  File "/home/pi/TWCManager/TWCManager.py", line 166, in unescape_msg
    msg[i : i + 2] = [0xC0]
TypeError: 'bytes' object does not support item assignment

Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 667, in <module>
    msg = unescape_msg(msg, msgLen)
  File "/home/pi/TWCManager/TWCManager.py", line 166, in unescape_msg
    msg[i : i + 2] = [0xC0]
TypeError: 'bytes' object does not support item assignment

Have yet to find a pattern or a way to reproduce it, will increase logging and log to a file to find out more.

ngardiner, 2020-12-21:
Thanks for reporting this! It looks like at some point, the list msg is being overwritten by a byte string instead. I'll need to do some investigation, would be great to hear back on your findings

ngardiner, 2020-12-21:
Hi @gescarra, I've just pushed a potential fix in 751ce27 for this - if you could try again after a git pull and let me know if it happens again, that would be great

gescarra, 2020-12-21:
Awesome, will pull this tonight and report on success or issues. Usually it would happen every 2-3 days. Thanks!

gescarra, 2020-12-28:
No issues at all since the fix, I think we're good, thanks!

closed 2020-12-28 by gescarra