ngardiner/TWCManager #201: Dummy broken?

tjikkun, 2021-01-16:
I am correct that the Dummy interface doesn't work anymore? If so, is this something to add back?

ngardiner, 2021-01-16:
It's a good question, the dummy interface was intended to give us testing capability and I don't recall the current state of it, but I'll take a look and get back to you.

tjikkun, 2021-01-18:
I am working on this in https://github.com/tjikkun/TWCManager/tree/dummy

awegmann, 2021-01-20:
Hi Nathan. I'm also interested in the dummy interface. I'm currently trying to implement a KNX command interface for TWCManager. And for testing my first intention was to disable the RS485 interface. But on startup I get immediately a failure because TWCManager tries to open the /dev/ttyUSB0 interface - even when this module is disabled. 
I can try to fix this, but I wonder if this should be the normal operation if I disable RS485 and enable dummy interface and if you just didn't use the dummy interface in the near past.

awegmann, 2021-01-20:
@tjikkun What's the current state of your "dummy" branch? Can you provide a pull request and @ngardiner Nathan, can you accept the pull request? This is exactly what I was looking for to make the dummy interface usable.

tjikkun, 2021-01-21:
Although I would like to do some more work to make dummy module more  useful, it can be merged as is, since it solves the immediate issue.

PR #208 (2021-01-21): Fix dummy module
With this change, the Dummy module can be loaded again, without having a serial device present
Fixes #201
closed 2021-01-21 by ngardiner