ngardiner/TWCManager #203: Old Interface not working anymore

Dustovsky, 2021-01-18:
Hello guys

I did a re-installation recently, but since then the old interface (IP/index.php) seems not working anymore.

Script is basically running in the background, I can use the new interface @ 8080 but not the old one.

Either I get an error like "Couldn't send ''. Error code ." 

or the "known" one

"Timed out waiting for response from TWCManager script.If the script is running, make sure the $twcScriptDir parameter in the source of this web page points to the directory containing the TWCManager script."

What I tried so far is to adapt the mentioned parameter in index.php

 // Point $twcScriptDir to the directory containing the TWCManager.py script.
    // Interprocess Communication with TWCManager.py will not work if this
    // parameter is incorrect.
    $twcScriptDir = "/home/pi/TWCManager/";
    
Neither "/etc/twcmanager" nor the above one from another TWCManager I'm using at home works....

Anybody an Idea what's the matter or what I'm doing wrong?

Cheers from Switzerland

ngardiner, 2021-01-19:
Hi @Dustovsky,

This interface was recently set to disable by default because it interferes with the ability to run as a service (due to the IPC channel it opens) and is also being actively replaced by the new interface, although it does still exist for backwards compatibility.

The instructions to enable it are here:

https://github.com/ngardiner/TWCManager/blob/v1.2.1/docs/modules/Control_WebIPC.md

The web interface was updated to show a message pointing to that page, however if you still have a previous version of the web php interface in /var/www/html rather than the updated version, you wouldn't see the notice and wouldn't know it was changed.

Dustovsky, 2021-01-19:
hmm... that's what I did...

config.json looks like this:

 "control": {
      "HTTP": {
        "enabled": true,
        "listenPort": 8080
      },

      "IPC": {
           "enabled": true
      },

      "MQTT": {
        "enabled": false,
        "brokerIP": "192.168.1.2",
        "topicPrefix": "TWC",
        "username": "mqttuser",
        "password": "mqttpass"
      }

not working for me :-/

ngardiner, 2021-01-24:
Unfortunately there's some bad advice in that script. Basically the script is untouched from the old fork and it did create the IPC pipe in the program directory, but with this fork we just standardised on ```/etc/twcmanager``` because that's common to all installs regardless of where you place the scripts.

So the two things that should make the IPC interface work for you are:

```
    "IPC": {
         "enabled": true
       },
```

in config.json, and 

``` 
$twcScriptDir = "/etc/twcmanager/";
```

in index.php. I'll remove the incorrect comment from the file so it doesn't catch anyone else out. 

Also, can you do REST API calls from your automation platform? We have API calls under the new web interface on 8080 which can do what you need, they're documented here:

https://github.com/ngardiner/TWCManager/blob/v1.2.1/docs/modules/Control_HTTP.md

Dustovsky, 2021-02-06:
thanks @ngardiner !!
works... or actually worked...
Then my RS485-Adapter broke litterally, prob too cheap.... that's why my smart-me testing is further delayed, sorry!

closed 2021-04-18 by ngardiner