ngardiner/TWCManager #205: Not working for me, messages start with F8 or F0

tjikkun, 2021-01-19:
In reading the code and forum etc, it must be that messages start with C0.
In my case however that is not the case. They start with F8 or F0 and seem to be 23 bytes long
This is a new TWC. Can it be that the protocol has changed, or is this a problem with hardware maybe? Like interference in the wire. Any pointers would be appreciated.

MikeBishop, 2021-01-19:
Assuming this is a Gen3 TWC, since it's new.  Everything we have is for Gen2; Gen3 hasn't even had load sharing enabled, last I'd heard, and I'm unaware of any work reverse-engineering what it sends over RS232.  (It's actually kind of surprising it sends _anything_, given that feature hasn't rolled out yet.)

tjikkun, 2021-01-20:
Ah yes, I should have been more specific. I am from The Netherlands. In Europe it still seems to be Gen2 that is being sold.
Now this TWC is actually from my neighbour, doing it as a favour (and a fun project). The TWX is connected via a +/- 15m long UTP cable to the USB-RS485-WE-1800-BT adapter in an RPI. I just found out the electrician has run the UTP-cable right next to the power cable, so maybe that is the issue. Gonna try testing with a cable in another location.

tjikkun, 2021-01-20:
If I take out the UTP cable, it works. So the issue is with the cable. I tried another one, which didn't go next to power cables, but I had the same issue. I don't know much about the physical side of things, but I would have thought using CAT6 cables would be OK for RS485.

ngardiner, 2021-01-20:
It would likely be a combination of the twisted pair cabling and length

https://e2e.ti.com/blogs_/b/analogwire/archive/2016/07/28/rs-485-basics-when-termination-is-necessary-and-how-to-do-it-properly

From that article:

```
The 120Ω value for termination resistors stems from what is known as the differential-mode characteristic impedance of the twisted-pair bus wires. Simply put, the wire gauge, insulation type and thickness, and number of twists per unit length all contribute to an impedance that high-speed data signals “see.” This impedance is denoted in ohms and typically ranges from 100Ω to 150Ω for twisted-pair cables. The writers of the RS-485 standard choose 120Ω as the nominal characteristic impedance; therefore, to match this impedance the termination resistors also have a default value of 120Ω.
```

So the advice would be a 120 ohm resistor between + and - on the Pi side, and if possible, ground the ground pin on the USB adaptor if it has one (as Pi doesn't have a proper ground). That should hopefully clean up the signal. Check this thread for more discussion:

https://github.com/dracoventions/TWCManager/issues/9

tjikkun, 2021-01-20:
Thanks, got it working now! Didn't make changes yet, but redid the connections. Must have been a bad connection somewhere. If I see bad messages in the future, I will look into adding a resistor.

closed 2021-01-20 by tjikkun