ngardiner/TWCManager #209: Non-scheduled charging and charge now keeps vehicles always awake.

StefanSchuerchSBB, 2021-01-21:
I have a TWC only for two teslas which are not mine. Therefore the vehicles are mainly not at home. If TWCManager has "non-scheduled charging" or "charge now" policy selected. Both cars are always awake. See the extraction of the logfiles:

2021-01-21 07:48:41,190 - Policy     01 New policy selected; changing to Charge Now
2021-01-21 07:48:41,212 - TWCManager 01 Limiting charging to 32.00A.
2021-01-21 07:48:41,232 - TWCManager 01 Charge when above 6A (minAmpsPerTWC).
2021-01-21 07:48:41,258 - TWCSlave   01 SHB 7132: 00 00.00/00.00A 0000 0000  M: 09 00.00/32.00A 0000 0000
2021-01-21 07:48:44,610 - TeslaAPI   01 Car API wake car failed.  State remains: 'asleep'.  Will try again in 30 seconds.
2021-01-21 07:48:47,058 - TeslaAPI   01 Car API wake car failed.  State remains: 'asleep'.  Will try again in 30 seconds.
2021-01-21 07:48:53,322 - TWCManager 01 Slave TWC 7132: Delivered 503 kWh, voltage per phase: (237, 234, 235).
2021-01-21 07:48:53,324 - TWCManager 00 Slave TWC 7132: Delivered 503 kWh, voltage per phase: (237, 234, 235).
2021-01-21 07:49:03,707 - TeslaAPI   01 Weitsox gÃ¶ppel is not at home.  Do not start charge.
2021-01-21 07:49:13,997 - TeslaVehic 01 Car API returned 'vehicle unavailable: {:error=>"{\"error\": \"timeout\"}"}' when trying to get status.  Try again in 1 minute.
2021-01-21 07:50:17,552 - TeslaAPI   01 White Knight is not at home.  Do not start charge.
2021-01-21 07:50:17,572 - TWCManager 01 Limiting charging to 32.00A.
2021-01-21 07:50:17,582 - TWCManager 01 Charge when above 6A (minAmpsPerTWC).
2021-01-21 07:50:18,812 - TWCManager 01 Slave TWC 7132: Delivered 503 kWh, voltage per phase: (237, 233, 236).
2021-01-21 07:50:18,814 - TWCManager 00 Slave TWC 7132: Delivered 503 kWh, voltage per phase: (237, 233, 236).
2021-01-21 07:50:43,506 - TWCManager 01 Limiting charging to 32.00A.
2021-01-21 07:50:43,528 - TWCManager 01 Charge when above 6A (minAmpsPerTWC).
2021-01-21 07:51:14,271 - TWCManager 01 Limiting charging to 32.00A.
2021-01-21 07:51:14,286 - TWCManager 01 Charge when above 6A (minAmpsPerTWC).
2021-01-21 07:51:22,521 - TeslaAPI   01 Weitsox gÃ¶ppel is not at home.  Do not start charge.
2021-01-21 07:51:26,070 - TeslaAPI   01 White Knight is not at home.  Do not start charge.

And more than an hour later:
2021-01-21 08:54:27,059- TeslaAPI   01 Weitsox gÃ¶ppel is not at home.  Do not start charge.
2021-01-21 08:54:30,250- TeslaAPI   01 White Knight is not at home.  Do not start charge.

As checked with the Tesla app, both cars are always awake during "non-scheduled" or "charge now" policy.

MikeBishop, 2021-02-02:
Can you try the [`location_frequency`](https://github.com/ngardiner/TWCManager/tree/location_frequency) branch and see if that helps?  It looks like it pings the car's location with a cache time of 1 minute, and checking the location could keep the car awake.  The change in 251c7e uses a one hour cache time, which should be enough time for the cars to go to sleep.  Once the cars are asleep, it shouldn't wake them up.

StefanSchuerchSBB, 2021-06-14:
I was not able to test it properly, sorry. It seemed to not make a difference or even not for one of the two cars. In the meanwhile with the new version 1.2.1, I am not able to log in with my tesla account (without MFA).
The proper solution for this issue will be to check the connected VIN to start/stop only this vehicle and to get rid of the location checking. This solves my problem waking up the vehicles forever and allows a proper solar charging even if both vehicles are at home. Will this feature be available soon in the v1.2.2?

MikeBishop, 2021-06-21:
Unfortunately, that's a bit of a chicken and egg problem.  From the TWC's perspective, there's no difference between not being plugged in and being plugged into a sleeping car.  We get the VIN when the car starts charging.  We could potentially wake the car that was last connected to each charger, but there's no guarantee that car hasn't left the premises.

There's a corollary, though -- we could reasonably say that any car which is plugged into the TWC *must* be at home, even if we didn't know that before.

StefanSchuerchSBB, 2021-06-25:
In my case there are two cars in the Tesla account but only one charger at home. I would appreciate to disable the periodic location checking e.g. with configuration setting, as it is not properly working for more than one car in the Tesla account.

Sending the stop command to the car that is currently charging will work, since the VIN information is available. Waking up the car that was last connectet to the charger should also be fine. If the car is not connected anymore, the start charging command can be eliminated.

MikeBishop, 2026-10-26:
Fixed in v1.4.0 (8f77b134): TWCManager no longer wakes vehicles it knows are away from home, and charge commands go only to the VIN a TWC is serving (#697). This was listed as closing #209 in the v1.4.0 changelog but the issue didn't auto-close. Closing now; please re-open if you still see cars kept awake on 1.4.0.

closed 2026-10-26 by MikeBishop