ngardiner/TWCManager #215: Scheduled charging not working - Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.

db9carbon, 2021-02-01:
v1.2.1. EV Car is a Polestar 2.   

On schedule charging, the charge does not start. 
Log dump (debug set to 10). In this example, schedule is 1400-1500, at 10amps. 

> Feb 01 13:59:57 TWCManager python3[965]: 13:59:57 TWCManager 1 Green energy generates 0W, Consumption 0W, Charger Load 0W
> Feb 01 13:59:57 TWCManager python3[965]: 13:59:57 TWCManager 1 Limiting charging to 0.00A - 0.00A = 0.00A.
> Feb 01 13:59:57 TWCManager python3[965]: 13:59:57 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
> Feb 01 14:00:27 TWCManager python3[965]: 14:00:27 Policy     1 New policy selected; changing to Scheduled Charging
> Feb 01 14:00:27 TWCManager python3[965]: 14:00:27 TWCSlave   1 SHB 6680: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 000014:00:27 TWCManager 1 Limiting charging to 10.00A.
> Feb 01 14:00:27 TWCManager python3[965]: 14:00:27 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
> Feb 01 14:00:27 TWCManager python3[965]: 14:00:27 TWCManager 1 Slave TWC 6680: Delivered 64 kWh, voltage per phase: (254, 0, 0).
> Feb 01 14:00:37 TWCManager python3[965]: 14:00:37 TWCSlave   1 Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.
> Feb 01 14:00:37 TWCManager python3[965]: 14:00:37 TWCSlave   1 SHB 6680: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
> Feb 01 14:00:43 TWCManager python3[965]: 14:00:43 TWCSlave   1 SHB 6680: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
> Feb 01 14:00:54 TWCManager python3[965]: 14:00:54 TWCSlave   1 Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.
> Feb 01 14:00:54 TWCManager python3[965]: 14:00:54 TWCSlave   1 SHB 6680: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
> Feb 01 14:00:58 TWCManager python3[965]: 14:00:58 TWCManager 1 Limiting charging to 10.00A.

Now, I can successfully start a charge from the new HTTP interface by pressing 'Start All charging'

> Feb 01 14:07:49 TWCManager python3[965]: 14:07:49 TWCSlave   1 SHB 6680: 09 00.00/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:07:57 TWCManager python3[965]: 14:07:57 TWCManager 1 Charge Session Started for Slave TWC 6666
> Feb 01 14:07:57 TWCManager python3[965]: 14:07:57 TWCSlave   1 SHB 6680: 09 01.25/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:08:00 TWCManager python3[965]: 14:08:00 TWCSlave   1 SHB 6680: 01 01.61/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:08:02 TWCManager python3[965]: 14:08:02 TWCSlave   1 SHB 6680: 01 10.09/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:08:06 TWCManager python3[965]: 14:08:06 TWCManager 1 Limiting charging to 10.00A.14:08:06 TWCSlave   1 SHB 6680: 01 09.01/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:08:06 TWCManager python3[965]: 14:08:06 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
> Feb 01 14:08:08 TWCManager python3[965]: 14:08:08 TWCSlave   1 SHB 6680: 01 10.09/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:08:32 TWCManager python3[965]: 14:08:32 TWCSlave   1 SHB 6680: 01 08.97/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
> Feb 01 14:08:34 TWCManager python3[965]: 14:08:34 TWCSlave   1 SHB 6680: 01 10.05/10.00A 0000 0000  M: 00 00.00/00.00A 0000 0000

It also seems, pressing 'Start Charge now' does not work either, gives same error as the schedule. 

Great work on this fork by the way, as I am looking for half an hour charging scheduling from 0030-0430, so also keen to help test that when you deprecate the original web interface!

MikeBishop, 2021-02-01:
Starting/stopping charging is done through the Tesla API, which obviously doesn't do anything for a non-Tesla EV.  The TWC is not capable of offering no energy if it's in communication with TWCManager.  You may want to change stopping mode to "Stop Responding," which will cause the TWC to offer absolutely nothing.

db9carbon, 2021-02-01:
> Starting/stopping charging is done through the Tesla API, which obviously doesn't do anything for a non-Tesla EV. The TWC is not capable of offering no energy if it's in communication with [...]

Although pressing ''Start All charging' on the new interface, does start the charge, i.e. not via the Tesla API

Saftwerk, 2021-06-04:
### I get lots of this Log entries when **no car is plugged**.
`TWCSlave   20 Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.`

To be more precise: The entries are created twice after this log:
TWCManager 20 Limiting charging to XX.XXA - YY.YYA = **ZZ.ZZA.**
and **ZZ.ZZ** is in between 10 .. 16 A.

I'm not sure if there is something wrong, because but the documentation says:
"_it's still drawing at least # 2A less than spikeAmpsToCancel6ALimit_."
and  **no car is plugged**.

### Source of Log `TWCSlave.py` (lines 898-908)
```
          # We've been offering the car spikeAmpsToCancel6ALimit
                        # for over 10 seconds but it's still drawing at least
                        # 2A less than spikeAmpsToCancel6ALimit.  I saw this
                        # happen once when an error stopped the car from
                        # charging and when the error cleared, it was offered
                        # spikeAmpsToCancel6ALimit as the first value it saw.
                        # The car limited itself to 6A indefinitely. In this
                        # case, the fix is to offer it lower amps.
                        logger.info(
                            "Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less."
                        )
```

### Log: (no car is connected)
2021-06-04 13:21:00,554 - TWCManager 20 Limiting charging to 13.99A - 1.16A = **12.84A**.
2021-06-04 13:21:00,558 - TWCManager 20 Charge when above 10A (minAmpsPerTWC).
2021-06-04 13:21:02,416 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:21:02,419 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:21:02,420 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:08,512 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:21:08,514 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:18,654 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:21:18,656 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:21:18,657 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:24,712 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:21:24,714 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:29,085 - TWCManager 20 Slave TWC 2388: Delivered 15 kWh, voltage per phase: (239, 232, 234).
2021-06-04 13:21:31,090 - TWCManager 20 Green energy Generates 3272W, Consumption 237W (Charger Load 0W)
2021-06-04 13:21:31,092 - TWCManager 20 Limiting charging to 13.92A - 1.01A = **12.91A.**
2021-06-04 13:21:31,093 - TWCManager 20 Charge when above 10A (minAmpsPerTWC).
2021-06-04 13:21:35,041 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:21:35,043 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:21:35,044 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:41,114 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:21:41,116 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:51,257 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:21:51,260 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:21:51,261 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:21:57,317 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:21:57,319 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:22:01,487 - TWCManager 20 Green energy Generates 3827W, Consumption 241W (Charger Load 0W)
2021-06-04 13:22:01,489 - TWCManager 20 Limiting charging to 16.28A - 1.02A = **15.26A.**
2021-06-04 13:22:01,491 - TWCManager 20 Charge when above 10A (minAmpsPerTWC).
2021-06-04 13:22:07,461 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:22:07,464 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:22:07,465 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:22:13,536 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:22:13,538 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:22:23,681 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:22:23,683 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:22:23,684 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:22:29,771 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:22:29,774 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:22:30,120 - TWCManager 20 Slave TWC 2388: Delivered 15 kWh, voltage per phase: (239, 228, 240).
2021-06-04 13:22:32,092 - TWCManager 20 Green energy Generates 4093W, Consumption 299W (Charger Load 0W)
2021-06-04 13:22:32,094 - TWCManager 20 Limiting charging to 17.37A - 1.27A = **16.10A.**
2021-06-04 13:22:32,096 - TWCManager 20 Charge when above 10A (minAmpsPerTWC).
[... 8 lines cut ...]
[... cut ...]
2021-06-04 13:25:14,922 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:25:20,827 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2021-06-04 13:25:20,830 - TWCSlave   20 Slave power for TWCID 2388, status: 0
2021-06-04 13:25:30,954 - TWCSlave   20 **Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.**
2021-06-04 13:25:30,957 - TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
2021-06-04 13:25:30,958 - TWCSlave   20 Slave power for TWCID 2388, status: 0

Just for info.

MikeBishop, 2026-10-27:
The 6 A spike workaround this log shows (`spikeAmpsToCancel6ALimit`) was removed in 1.4.1 (#710); `spikeAmpsProactively` and `spikeAmpsReactively` are now ignored. If scheduled charging still doesn't start on 1.4.1, please open a new issue with a debug log.

closed 2026-10-27 by MikeBishop