ngardiner/TWCManager #216: Is there no limit to the current taken from the grid?

juanjoqg, 2021-02-02:
I can't see any way to limit the current taken from the grid.

I've been using the old TWCManager branch and I made some changes adding a back_ground_task checking the actual consumption from the grid, through the Fronius inverter interface, and adapting the  wiringMaxAmpsAllTWCs in order to prevent taking more power from the grid than I can/want.

I'm trying to figure out how to implement it in this new brach, but I'm just wondering if this is allready there but I didn't find it.

Thanks.

MikeBishop, 2021-02-02:
So essentially, just as there is currently a max power draw per TWC and a max for all TWCs collectively, you'd like to set a maximum net consumption for the house as a whole.  Alternatively, as you said, you'd like to vary `wiringMaxAmpsAllTWCs` based on consumption instead a fixed wiring value.

I think a complete solution would involve:
- Retrieving the consumption/generation data at all times, which we currently don't do due to API limits; whether this is a concern would depend on what API a person is using.
- Similar logic to what's used for keeping charger power in sync with green energy would also need to be used to keep in sync with "available" grid energy.  Rather than mimicking that logic, it might be worth considering whether it could be abstracted so that one piece of code could handle both alignment decisions.  Need to think about this.

Note that this doesn't actually keep you from exceeding the configured value.  If you're sitting at the limit, then something in the house starts consuming extra power, you will spike above the limit and then adjust back down when TWCManager notices.  So if your goal is not to trip a circuit breaker, you'll have up to 30 seconds exceeding the configured limit and you'll trip the breaker anyway.  (Also the spike amps could cause briefer, 5s variations, but that's rare if you're not in Track Green Energy.)

juanjoqg, 2021-02-02:
Using wiringMaxAmpsAllTWCs  was just an easy way to introduce this "feature" to the old TWCManager branch but  I'm sure not a clean one .
I'm just trying to get familiar with this TWCManager branch code, but so far I see it as follows:
- New config parameter maxPowerAllowedFromGrid
- Adding a new policy as maxPowerFromGrid, it will trigger a new background task checkMaxPowerFromGrid
-  checkMaxPowerFromGrid should use some way to get the power we are taking from the grid, it could be the same as greenEnergy or another device, set a new attribute for TWCMaster maxAmpsToDivideAmongSlavesFromGrid=maxPowerAllowedFromGrid+generationW - consumption+currentOffer
- setMaxAmpsToDivideAmongSlaves will check that maxAmpsToDivideAmongSlaves<=maxAmpsToDivideAmongSlavesFromGrid

tjikkun, 2021-02-03:
> Note that this doesn't actually keep you from exceeding the configured value. If you're sitting at the limit, then something in the house starts consuming extra power, you will spike above the limit [...]

Well, actually (in The Netherlands anyway, but I suspect in other countries as well) you can exceed the amperage for some time, the higher you go over it, the faster the fuse will blow. See for example this (random) [datasheet](http://www.littelfuse.com/data/en/Data_Sheets/FLM.pdf): 
If you have a 25A fuse and you load it with 33.75A it can take up to an hour before it blows. With 50A it still takes at least 12 seconds.
So in any case this drastically reduces the chance of it happening.

When implementing this feature, I would suggest also supporting checking per-phase, not just a total.

MikeBishop, 2021-02-03:
Just like for anything on GitHub, you can fork it, make changes to your fork, and then create a pull request.  The only wrinkle is that PRs will naturally try to go to the top of the tree, which is not this repo; you'll have to manually target the pull request to ngardiner/TWCManager branch v1.2.1.  

(@ngardiner, it might be worth breaking this repo off so that PRs target properly by default....  It appears that only GitHub support can do this with an existing repo, however.)

juanjoqg, 2021-02-03:
Thanks, I think I get it.

juanjoqg, 2021-02-03:
Pull request sent https://github.com/ngardiner/TWCManager/pull/220

closed 2021-02-03 by juanjoqg