ngardiner/TWCManager #217: Error in "/lib/TWCManager/Control/HTTPControl.py" line 450 + possible fix

GMerg, 2021-02-02:
When having a policy with nested conditions, you get an error when showing policy in web interface.
The file has to be changed from:

**`page += "<td>" + condition + "</td>"`**

to:

**`page += "<td>" + str(condition) + "</td>"`**

A sample policy in **/etc/twcmanager/config.json** to debug this:
```
...
"extend":{
...
"after":[
          { "name": "Charge on FR for SA",
            "match": [ "modules.TeslaAPI.minBatteryLevelAtHome", "tm_hour", "tm_wday", [ "tm_mon", "tm_mon" ] ],
            "condition": [ "lt", "gte", "eq", ["gte", "lte"] ],
            "value": [ 80, 15, 4, [ 10, 2] ],
            "charge_amps": 12
          },
....
```

MikeBishop, 2021-02-02:
Ooh, good catch.  I recently rewrote my policies so I don't have anything nested right now, so I didn't encounter this in my testing.  Yes, that change keeps it from crashing, but doesn't actually give you the same view into why the policies are / aren't matching.

GMerg, 2021-02-02:
OK, glad I could help in some way. :)

ngardiner, 2021-04-27:
Thanks for this @GMerg - I'm doing some work on the UI now as part of an upcoming feature, so I'm going to merge this in with that code now and it will be in v1.2.2 shortly. Thanks again :)

closed 2021-04-27 by ngardiner