ngardiner/TWCManager #225: Repo rebasing

ngardiner, 2021-02-06:
As mentioned by @MikeBishop in #216, the branching of the codebase that I originally used to hedge against large scale changes when pulling apart the monolith TWCManager into modules has become the achilles heel of release strategy as PRs have to target a certain release.

I'm planning to reach out to github support to ask to rebase 1.2.1 as master and all current PRs to be retargeted, that way we can do versioning independent of PRs without needing to change things on the submitter's side.

That said, I want to make sure I'm not missing anything obvious in terms of impact to users/contributors. I am hoping tagging the current commit with v1.2.1 would allow people already on 1.2.1 to continue to pull but perhaps the branch/tag URLs are different - the other option is to just do a release after the rebasing - we're about due for one anyway (and 1.2.0 is now effectively useless for API users given the auth changes), and get started on the next development version.... feedback is welcome on this idea

ngardiner, 2021-02-06:
As an alternative, should we take this as an opportunity to move this project to a more appropriate place - a centralised account and if so, I might need to look into the mechanics of the best approach to both move the repo and its context and to rebase/retarget at the same time.

hopfi2k, 2021-02-06:
Having a "master" (shouldn't it be "main" or something like this as the word "master" might be linked to slavery...🤷🏻‍♂️) would be fantastic as it would ease submitting pull requests independently of the most current branch (while we still could use different branches for trying out new features etc.).

ngardiner, 2021-02-06:
This is a good point @hopfi2k - although "master" is convention we could quite easily switch branch to a branch named devel as the working development branch (and set it as default in GitHub) and then tag releases from there.

hopfi2k, 2021-02-06:
That would be really fantastic and ease things a lot @ngardiner !

MikeBishop, 2021-02-08:
GitHub is actually encouraging projects to switch to "main" by default instead; see https://github.com/github/renaming.  If we're actually switching up the branch structure entirely, though, I would encourage switching to something like this:

```
main
 \--- develop
         \--- (PRs go here)
```

...where `main` is the latest stable release, `develop` is mostly-stable but actively changing, and past releases are tags on the `main` branch.  That means casual users check out `main` and can always stay current but stable by doing a `git pull`.

MikeBishop, 2021-02-08:
I think it's fine on a personal account; I don't see a need to move it to an organization so long as you're content to remain the person in charge.  😁  And yes, we should cut a release ASAP as soon as we're confident in the authentication changes.

tjikkun, 2021-03-17:
Personally I am more in favor of:
```
main
 \--- (PRs go here)
 \--- v1.2
      \--- v.1.2.0 (tag)
      \--- v.1.2.1 (tag)
 \--- v1.3
      \--- v.1.3.0 (tag)
      \--- v.1.3.1 (tag)
      \--- v.1.3.2 (tag)
 \--- feature_branch_a
 \--- feature_branch_b
 \--- feature_branch_c
```

where `main` is mostly-stable but actively changing, `v*.*` are stable versions, past releases are tags in their version branch. For users this means it is probably best to use either a specific version (tag), or a version branch, to get updates but not breakage. If you want, you can also use `main`, there shouldn't be any breakage there either because development is usually done in a feature branch before it is merged to `main`.
The downside of `main -> develop` as @MikeBishop suggested is, in my opinion, that eventually you might not get away with breaking some backwards compatibility to move the project forward. In that case, for a user things will break at an arbitrary time.

Just my 2 cents.

ngardiner, 2021-04-22:
Huge thanks @MikeBishop and @tjikkun for the feedback. I know I've been silent but I have slowly and carefully been looking to unwind some of the mess of the previous structure which was of my own causing, and also to improve the experience of submitting PRs overall.

The first and simplest step was to rebase the default branch to main, which is the branch that all users see when accessing the project, and also the default branch to which PRs are submitted. This has been done now for a while, it broke a few PRs but thankfully most were ours and can be easily fixed.

The second step which is not technically related to this but which is complimentary is that I've added an automated CI/CD pipeline for running automated tests of TWCManager after a merge. It doesn't do pre-merge testing yet (for reasons related to how GitHub Actions work currently) but they will once everything is well established and tested.

Step 3 will be to create the appropriate development branch, and both proposals above take this approach. I'll ensure we don't repeat the previous error where branch and tag names conflicted which caused all sorts of issues, but we can implement appropriate development branch(es) going forward.

What I think we could do to satisfy the above approaches (and what I'd like your feedback on) is:

   - Leave main as the default repo branch. This has the disadvantage of defaulting the PRs to the release/stable branch, but has the advantage of not changing the branch name if we went with a major version development branch similar to the above.
   - Add a GitHub Actions automation to trigger, on submission of a PR to the main branch, a PR to the appropriate development branch. By doing things that way, we could swing the development branch to a new major version per @tjikkun's proposal above if we're trying to close off a development branch and do stability testing for a major version.
   - If there's an issue with the PR, we could then just roll back the merge and add comments to the main branch PR, and while I'll have to work out the automation in GitHub Actions, essentially we'd re-send the PR to the Dev branch via GitHub Actions on subsequent push to the PR branch.
   - We could then merge the PR to the dev branch and test from the dev branch, and once it's confirmed within the dev branch as being stable, merge the original PR into main
   - All release tagging will be done from the main branch
   - Feature branches would remain as is today, as shown above in @tjikkun's diagram

How does this sound as an approach?

ngardiner, 2021-04-27:
BTW - there's a few issues with the current github build action, just because I haven't ironed out one or two docker related issues, it causes some builds to fail not because there was an issue with the build but because of a lack of automated pruning and starting of the build workers, something I will get fixed tonight hopefully - for now if a commit shows as failed due to the buld env I just restart it - it will cause an email to be sent saying the build failed which is the key issue, but this shouldn't be an issue going forward once I fix things up on my end.

ngardiner, 2021-04-27:
Just a follow up that this is fixed now - the builds are completely automated now

tjikkun, 2021-05-11:
> How does this sound as an approach?

Sorry for the late reply, but the general plan look great. I don't understand every detail of what you wrote, for instance "If there's an issue with the PR, we could then just roll back the merge and add comments to the main branch PR" but I don't think it is very important that I do. What is more important is how it works in practice, and I am confident you have made a nice setup. Especially nice is that you added GitHub actions!
If you need any help with unstucking the runner, feel free to ask, but my impression is you can handle it just fine :)

closed 2022-02-15 by ngardiner