ngardiner/TWCManager #229: Charger goes to error state (red led flash 4 times)

Nachtkapp, 2021-02-11:
I have a strange problem with the TWCManager and the charger.

From time to time the charger stops working in conjunction with the TCWManager. In this state the top light of the charger is green and the lower light flashed red 4 times, then pauses and flahsed 4 times red again and so on. In this state I cannot charge, the car led goes red immediately. The manual says this indicates that there is more than one master. 

In this time, the TWCManager log shows mostly this message every 10 seconds:
TWCManager 01 32.00 amp slave TWC 4704 is ready to link.  Sign: 93

From time to time there is a message like this:
TWCManager 01 ERROR: Ignoring message of unexpected length 15: FD E2 47 04 93 0C 26 88 11 71 F7 ED F5 3F FF

I restart/disconnect from power of the charger and the TWCManager does not help. I tried this in overy order and even let the charger powered of so 2 hours. Nothing works.

The only way to get it working again is:
- Disconnect charger from power
- Disconnect cable to TWCManager
- Change the charger to run standalone
- Power up the charger
- Charger works standalone
- Disconnect charger from power
- Connect cable to TWCManager
- Change the charger to run as slave
- Power up TWCManager
- Power up the charger
- Charger works with TWCManager again

This works for about 2 months and then the problem come up again. 

Has anyone an idea what could cause this?

ngardiner, 2021-02-12:
Do you have any resistor between the RS485 ```+``` and ```-``` wires? How far is the distance between the pi and the TWC?

Nachtkapp, 2021-02-12:
Hi Nathan. Thanks for the quick reply. 
No, I have no resistor between the RS485 module and the wires to the charger. Is that needed? The cable has a lenght of about 2 meters. 
Regards
Sven

MikeBishop, 2021-02-12:
The resistors aren't always needed, but they reduce noise on the connection.  I see where @ngardiner's coming from; those symptoms do sound like messages are getting corrupted in both directions.

I wound up cutting my cable much shorter so it would fit inside the case of the TWC; I installed without bias resistors and haven't had an issue.  I don't know if a shorter cable would fix it for you, or if you need the resistors either way, or if there's a totally different problem.

ngardiner, 2021-02-14:
@MikeBishop has summarized it well, I have 2 likely theories at this point Sven, one is that it could be some sort of interference, and the other is that it could be related to your RS485 module. If it were just straight interference you'd probably not see it degrade over a period of time as you are, but it's possible that a combination of the cable length and attenuation/interference plus perhaps some sort of condition with the RS485 module or localised noise might cause it to essentially fall over the threshold for being too noisy and cause the behavior you're seeing at the time that you see it.

A 120 ohm resistor across the + and - pins is most likely to resolve this problem, it will reduce the effects of that noise, and places less stress on the RS485 adapter as well. It isn't strictly necessary for installation, but when you're seeing the sort of behavior you are it is advisable.

Out of interest, have you tried leaving the TWCManager/RS485 off for those two hours rather than the TWC? I can't explain the exact behavior you see but I wouldn't be shocked if having the RS485 powered down for a while may fix things - there are some reportedly less than perfect chipsets out there, I did spring the extra money for an FTDI chipset for that reason, others have seen their units stop working after a period of continuous use.

Nachtkapp, 2021-02-14:
Hi Nathan. Thank you for the details. When I powered down the charger I also powed down the Pi with the RS485 USB adapter. 
So far I replaced the calbe by a now 30 cm twisted pair cable. I also replaced the RS485 adapter as I had 2 second one of the same type. But the problem stays the same. I see the link ready messages all the time but the charger does not seems to react to the master anymore. 
I will get an 120 Ohm resister and try it with it. I will keep you updated. 
Regards
Sven

juanjoqg, 2021-02-18:
Hi @Nachtkapp, another similar issue I have seen was due to the pi power supply, changing it solved the issue, it could cause the signal from the pi to the TWC is not strong enough

Nachtkapp, 2021-02-20:
Hello Together. I was just able to implement the 120 Ohm resistor and it's working again :-)
I just hope the resistor did the trick and not unplugging the Pi from the charger for several days. Time will tell. 
Thanks to all of you for your hints and tipps. That helped me allot. From my side, the issue can be closed. 
Regards
Sven

closed 2021-02-20 by Nachtkapp