ngardiner/TWCManager #232: Charge Limit changes automatically, unexpectedly.

PJMikols, 2021-02-16:
I have my charge limits set: 
config.json: 
        "chargeNowLimit": 75,
        "greenEnergyLimit": 90,
        "scheduledLimit": 75,
        #"nonScheduledLimit": -1, 

With the recent cold weather, if i manually raise the limits in my App, within minutes it repeatedly is set back to 75%. I assumed the 75% was set only at the beginning of the "charge now" cycle, but maybe it's doing it multiple times? ( see 20:24:52 , 20:30:09 ) 

20:24:16 Policy     7 All policy conditions have matched. Policy chosen is Charge Now
20:24:16 TWCManager 1 Limiting charging to 20.00A.
20:24:16 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
20:24:16 TWCSlave   1 SHB 9555: 01 49.26/48.00A 0000 0000  M: 09 00.00/20.00A 0000 0000
20:24:16 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:24:16 TeslaAPI   8 car_api_available returning True
20:24:18 TWCSlave   1 SHB 9555: 09 49.35/20.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
20:24:20 TWCSlave   1 SHB 9555: 09 20.70/20.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
20:24:30 TWCSlave   1 SHB 9555: 01 20.70/20.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
20:24:47 Policy     8 Evaluating Policy match (settings.chargeNowAmps [20]), condition (gt), value (0)
20:24:47 Policy     8 Evaluating Policy match (settings.chargeNowTimeEnd [1613445848.0266306]), condition (gt), value (0)
20:24:47 Policy     8 Evaluating Policy match (settings.chargeNowTimeEnd [1613445848.0266306]), condition (gt), value (1613442287.1356814)
20:24:47 Policy     7 All policy conditions have matched. Policy chosen is Charge Now
20:24:47 TWCManager 1 Limiting charging to 20.00A.
20:24:47 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
20:24:47 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:24:47 TeslaAPI   8 car_api_available returning True
20:24:48 TeslaVehic 8 Car API cmd https://owner-api.teslamot0rs.com/api/1/vehicles/68291277884 <Response [200]>
20:24:49 TeslaVehic 8 Car API cmd https://owner-api.teslamot0rs.com/api/1/vehicles/68291277884/data_request/drive_state <Response [200]>
20:24:50 TeslaVehic 8 Car API cmd https://owner-api.teslamot0rs.com/api/1/vehicles/68291277884/data_request/charge_state <Response [200]>
20:24:52 TeslaVehic 8 Car API cmd set_charge_limit <Response [200]>
20:24:52 TeslaAPI   2 Set 386365 **to charge limit of 75%**
20:24:58 TWCManager 1 Slave TWC 9555: Delivered 6837 kWh, voltage per phase: (242, 0, 0).
20:25:02 TWCManager 1 Charge Session Stopped for Slave TWC 9595
20:25:02 TWCSlave   1 SHB 9555: 03 00.00/06.00A 0000 0000  M: 09 00.00/21.00A 0000 0000
20:25:03 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:25:03 TeslaAPI   8 car_api_available returning True
20:25:03 TeslaAPI   8 startOrStop is set to start
20:25:03 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:25:03 TeslaAPI   8 car_api_available returning True
20:25:04 TWCSlave   1 SHB 9555: 09 00.00/21.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
20:25:05 TeslaAPI   8 Car API cmd charge_start <Response [200]>
20:25:05 TeslaAPI   4 386365: start charge response{'response': {'reason': 'complete', 'result': False}}
20:25:05 TeslaAPI   1 386365 is done charging or already trying to charge.  Stop asking to start charging.
20:25:08 TWCSlave   1 SHB 9555: 09 00.00/21.00A 0000 0000  M: 09 00.00/20.00A 0000 0000
20:25:10 TWCSlave   1 SHB 9555: 09 00.00/20.00A 0000 0000  M: 00 00.00/00.00A 0000 0000

20:29:54 Policy     7 All policy conditions have matched. Policy chosen is Charge Now
20:29:54 TWCManager 1 Limiting charging to 20.00A.
20:29:54 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
20:29:54 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:29:54 TeslaAPI   8 car_api_available returning True
20:30:00 TWCManager 1 Slave TWC 9555: Delivered 6837 kWh, voltage per phase: (246, 0, 0).
20:30:02 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:30:02 TeslaAPI   8 car_api_available returning True
20:30:03 TeslaVehic 8 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/68291277884 <Response [200]>
20:30:04 TeslaVehic 8 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/68291277884 <Response [200]>
20:30:06 TeslaVehic 8 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/68291277884/data_request/drive_state <Response [200]>
20:30:07 TeslaVehic 8 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/168291277884/data_request/charge_state <Response [200]>
20:30:09 TeslaVehic 8 Car API cmd set_charge_limit <Response [200]>
20:30:09 TeslaAPI   2 Set 386365 to charge limit of 75%
20:30:17 TWCManager 1 Charge Session Stopped for Slave TWC 9595
20:30:17 TWCSlave   1 SHB 9555: 03 00.00/06.00A 0000 0000  M: 09 00.00/21.00A 0000 0000
20:30:17 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:30:17 TeslaAPI   8 car_api_available returning True
20:30:17 TeslaAPI   8 startOrStop is set to start
20:30:17 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:30:17 TeslaAPI   8 car_api_available returning True
20:30:19 TWCSlave   1 SHB 9555: 09 00.00/21.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
20:30:19 TeslaAPI   8 Car API cmd charge_start <Response [200]>
20:30:19 TeslaAPI   4 386365: start charge response{'response': {'reason': 'complete', 'result': False}}
20:30:19 TeslaAPI   1 386365 is done charging or already trying to charge.  Stop asking to start charging.
20:30:23 TWCSlave   1 SHB 9555: 09 00.00/21.00A 0000 0000  M: 09 00.00/20.00A 0000 0000
20:30:25 Policy     8 Evaluating Policy match (settings.chargeNowAmps [20]), condition (gt), value (0)
20:30:25 Policy     8 Evaluating Policy match (settings.chargeNowTimeEnd [1613445848.0266306]), condition (gt), value (0)
20:30:25 Policy     8 Evaluating Policy match (settings.chargeNowTimeEnd [1613445848.0266306]), condition (gt), value (1613442625.1302254)
20:30:25 Policy     7 All policy conditions have matched. Policy chosen is Charge Now
20:30:25 TWCManager 1 Limiting charging to 20.00A.
20:30:25 TWCSlave   1 SHB 9555: 09 00.00/20.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
20:30:25 TWCManager 1 Charge when above 6A (minAmpsPerTWC).
20:30:25 TeslaAPI   8 Entering car_api_available - next step is to query Tesla API
20:30:25 TeslaAPI   8 car_api_available returning True

PJMikols, 2021-02-17:
I think i may have found it, but will watch for a few more days. My settings file was Read only so the checks were failing and the "Charge Now' state was being set every 30 sec causing the limit to be set repeatedly.

MikeBishop, 2021-02-22:
A lot of our code relies on the pattern of being invoked frequently and then deciding whether it needs to do any work.  The logic (IIRC, not double-checking the code) works like this:

- Every time the policy is selected, regardless of whether it's a change, it checks whether it needs to apply the policy's limit
- Before applying the limit, TWCManager caches what the "normal" limit is, so it can restore it later
- After applying the policy, TWCManager will remember that it applied the limit and not re-apply it later (but if the `settings.json` file is read-only, it might be unable to remember this)
- When the car leaves the geofence / the next policy doesn't impose a limit, the saved normal limit is restored **if the car is still set to the limit TWCManager applied**.  If the limit has been changed manually, the car assumes that's your new desired "normal" limit and treats the limit as having been restored to normal without changing it.

If you're seeing the limit be applied repeatedly after making `setting.json` writeable, I'll try to reproduce it.

closed 2022-02-15 by ngardiner