ngardiner/TWCManager #234: Control Charge current from OpenWB - Switch off Green Policy

Flitzer42, 2021-02-17:
Thanks for this great software and all the efforts you invest.
I tried to use TWCManager just as an interface to the TWC and use OpenWB to manage the energy flow.
I am still using 1.2.0 as I did not manage to get 1.2.1 up and running (another detailed issue report will follow).
Controlling the charge power from OpenWB works well at night (no green policy) but at daytime it got stuck.
At daytime the old WebUI shows “Power available for all TWCs: None”, the new UI shows “Is a Green Policy? Yes” and “Amps offered 0.00”, but the car is still charging a 6 Amps (min setting) which is fine. OpenWB reports the current charge Amps, but changing the power has no effect to TWCManager.
At night the old WebUI shows “Power available for all TWCs: 7.00A”, the new UI shows “Is a Green Policy? No” and “Amps Offered 7.00” when OpenWB set to 7 Amps. Chaning the charge power from OpenWB is accepted bei TWCManager and the charge curetn of the car follows as well.
TWC Manager seems to not react on the inputs from OpenWB when green policy is active. Is there any chance to disable the green policy permanently and just remotely control the charge power? I did not find a corresponding parameter in config.json.
Any help is appreciated

Flitzer42, 2021-02-19:
Managed to update to V1.2.1
Now it works fine. The improved settings section is great.
Strange observation: Need to configure "Resume tracking green energy at:" to "23:45" to be able to control power from OpenWB.
Would appreciate a switch to disable tracking green energy completely.

ngardiner, 2021-02-19:
This is due to the way the openWB integration works. It's a little less than optimal, but openWB's interface to TWCManager works by setting the non-scheduled charge rate via the legacy web interface. This means that if you have any other process or policy running (such as Track Green Energy or Charge Now) it will not control the charge rate.

If you have the Non-Scheduled Power Action set to Charge at Non-Scheduled Charge Rate, then resuming green energy at any time after 8pm, green energy tracking is entirely disabled. I'm hoping in the future that openWB will switch to using the chargeNow policy which overrides all of the policy selection and just charges at the specified rate, this would not require any changes on the TWCManager side to work effectively.

Flitzer42, 2021-02-19:
Thanks for the explenation. Took me just a while to figure out the tweak with the "resume green enery after 8 pm". I fully support you in not doing everything to compensate others using TWC Manager not in the intended way.
Your proposal is great. I was curious regarding track green energy flipping to yes while no green energy source confgiured (and thus it always reported 0 :-)). The order of evaluation makes absolutely sense.
So lets close this issue.

MikeBishop, 2021-02-19:
Another way to disable the Track Green Energy policy would be to add this to your policy\extend\restrictions section in the config.json file:

          "Track Green Energy":{
            "match": [ "" ],
            "condition": [ "false" ],
            "value": [ "" ]
          }

ngardiner, 2021-02-20:
The check for any loaded EMS module before going into Track Green Energy policy is a no-brainer for me, this is really a good reminder that we should add a check for if 1 or more instance of a given module is loaded before activating the policy. The openWB case was a bit of an odd one in that until now, it was almost (not entirely, but almost) certain that someone using the project is tracking green energy in some form.

I suspect that is changing, though. For example, once we add power pricing support, we'll have a whole new area of functionality that doesn't require green energy tracking to provide efficient charging for people with variable power plans that have API interfaces. So I think the time is right to drop the assumption about green energy tracking in the policy.

Let's keep this issue open for a little while longer @Flitzer42 while we add an additional policy check for any EMS modules loaded, and that should resolve the issue of openWB not working during daylight hours out of the box. There's still the other caveats around policy priority but I think people would expect that either activating charge now or scheduled charging would override openWB anyway. so it's really just the tracking green energy that should be a surprise for someone who doesn't have an EMS module loaded.

Appreciate you raising this @Flitzer42!

Flitzer42, 2021-02-20:
Works perfekt. This was the setting I was looking for. Thanks !!!!!!!!!!!!!!!

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop