ngardiner/TWCManager #241: Scheduled Completion

MikeBishop, 2021-02-24:
I don't believe we ever went back and fixed the Flex Schedule code to ensure it was calculating based on the limit from the Scheduled Charging policy and not the limit currently applied to the car.  We should do that at some point, or it could start charging catastrophically late.

jherby2k, 2021-03-26:
Please, yes! I have my car set to 90% but the scheduled charging policy set to 50%. idea being it tops me off before my morning commute to at lease 50%, regardless of how successful solar generation was the day before. 

Right now it seems to be calculating the change minutes needed to hit 90%, not 50%. So it'll be done charging to 50% at like 2AM, and i'll have a cold battery when I leave in the morning.

MikeBishop, 2021-03-30:
@jherby2k, you can work around this in the meantime by setting the charge limit for Unscheduled Charging to 50% as well.

ngardiner, 2026-10-31:
Closing as a duplicate of #165, which tracks the same problem: the flex start time is worked out from the car's current charge limit (`getScheduledAmpsTimeFlex`), not the Scheduled Charging policy's limit.

closed 2026-10-31 by ngardiner