ngardiner/TWCManager #243: [Feature Suggestion] Weather API integration to e.g. delay charging on next day being sunny

GMerg, 2021-03-05:
It would be great if we could somehow integrate a weather forecast into the scripting.
We would be able to use them in the policy scripting and e.g. define to only "charge now" if next day isn't sunny, else wait and charge tomorrow. That would safe a lot money if you are producing your own energy.

MikeBishop, 2021-03-06:
A straightforward way to add this might be a module that isn't actually leveraged from the core code, but could be used in policy restrictions.  For example, if there were an OpenWeatherMap module, you could then add a restriction to Scheduled Charging that caused it to only activate when cloud cover was forecast for the following day.

GMerg, 2021-03-06:
Yes, that souds perfect. I really would like helping with that but my phyton knowlage is still limited. I can do some research if thats needed...like what API to call and how to get the data wanted.

MikeBishop, 2021-03-07:
I know from previous projects that it's hard to find a free/cheap-for-individual API that's sufficiently local to make these decisions on. Dark Sky was nice, but it's being EOL'd. OpenWeatherMap is probably the best choice for new work, IIRC.

EricE, 2021-03-07:
Another possible option though very new:  https://www.reddit.com/r/homeassistant/comments/lwkidx/for_anyone_using_the_soon_to_shut_down_dark_sky/

GMerg, 2021-04-05:
We could also use e.g. **OpenWeatherMap** as a _PseudoInverter_ that gives back the kW value based on the current weather condition. That way it will work with ANY solar system (not as accurate as with a real inverter interface, but better than nothing).

_Needed to configure_ :

- API key that you cab get for free when signuing up on https://home.openweathermap.org/users/sign_up
- Lattitude/Longitude (maybe get from TeslaAPI itself)
- Peek kW that your solar system can generate (maybe split per month to be more accurate)

_Coding work_:

- Call this API: https://openweathermap.org/api/one-call-api (cache results to stay below 60 calls/minute and 1,000,000 calls/month)
- Compare results of *.weather.id with ids on https://openweathermap.org/weather-conditions#Weather-Condition-Codes-2 and calculate possible kW (intersting ids 800 (best) - 803 (worst), maybe light rain...) + also compare sunset/raise with current time)

Having this as an extra module is of course better for policy scripting, but for people without being able to access there Inverter easily, an alternative.

GMerg, 2021-04-19:
I have now written a module with my limited python knowlage: https://github.com/ngardiner/TWCManager/issues/262

MikeBishop, 2026-10-26:
An OpenWeatherMap EMS module was contributed (c4f1417f, v1.2.2; `docs/modules/EMS_OpenWeatherMap.md`). It estimates generation from the weather rather than limiting scheduled charging on a forecast. Could you check whether that covers what you wanted? If the forecast-based restriction is still wanted, say so here.