ngardiner/TWCManager #244: Modern Web:

MikeBishop, 2021-03-07:
[image]

The new UI says that my current policy is Do Not Charge and that I have zero solar surplus, yet I am on Track Green Energy policy and delivering my 27A of surplus to the car.

Something is clearly different in my environment than your test environment. I haven't yet attempted to debug, but I'm guessing it's the presence of custom policies in the list. Have you taken a dependency on the length of the array?

ngardiner, 2021-06-25:
Looks like that selection is actually made based on the non-scheduled charging action, rather than current policy.

I'll have to think about how to hook this into the UI element that's there today, or how to adapt the UI element to be more universal

MikeBishop, 2021-06-25:
If you're able to call (or expose) the policyIsGreen() function, the three fields might be as simple as:

- If policyIsGreen()
- Else If charge_amps > 0
- Else...

VIDGuide, 2021-06-26:
Looks like that is coming back in the getStatus set.

```
 "generationWatts": "6800.00",
    "isGreenPolicy": "Yes",
    "scheduledChargingStartHour": 22,
    "scheduledChargingFlexStart": 22,
    "scheduledChargingEndHour": 7,
```

Presently the selected button is actually set at render time, not via the ajax lookup.
`
 {% set active = master.settings.get("nonScheduledAction", 1) %}
`

But should be able to be manipulated beyond that. As Nathan mentions, it's reflecting the non-scheduled charge action, rather than policy.

MikeBishop, 2026-10-26:
The Modern theme now derives the active policy from `/api/getActivePolicyAction` rather than the `nonScheduledAction` setting (7f7aeb49 and follow-ups, v1.3.0). Closing as done.

closed 2026-10-26 by MikeBishop