ngardiner/TWCManager #245: Modern Web: Charge Now in Portrait

MikeBishop, 2021-03-07:
[image]
I'd suggest a minimum width on these controls, so that if the screen is narrow, each selector fills the width and the controls arrange vertically. The current layout makes it difficult to select a value accurately.

hopfi2k, 2021-03-09:
I've spent a fair amount of work, making the modern web design more responsive and will commit these changes shortly.

VIDGuide, 2021-06-27:
I've got a fix for this one in my current pull request #317 

Responsive mode will stack like this:
[image]

MikeBishop, 2021-06-29:
Closing since #317 is merged.

closed 2021-06-29 by MikeBishop