ngardiner/TWCManager #246: [Code Architecture]  Consider using AsyncIO

MikeBishop, 2021-03-09:
Rather than having our current multiple threads with dedicated tasks and worrying about when one activity might block another, we should consider transitioning to Python's async/await pattern.  I was looking at moving the TeslaAPI module to Tesla's websocket API to reduce polling / get instant updates, but Python's websocket library has a hard dependency on async.

ngardiner, 2021-04-29:
What would be really interesting would be if we could note some of the things that we think would benefit from async events, for example:

- EMS values: Do we think these will always be polled? The actual inverter fetch has to be a routine but it's a feasible idea that it could be a background thread of its own with an async callback if the value changes. Then again, these are highly variable values and managing individual bg threads per EMS is probably not our plan, so these might not be the best

- Things like charge / getLifetimekWh / getVehicleVIN are actually a bit of a hack if I do say so myself :) The primary reason for making many of these background tasks is because it provides us with "clean airways" on the RS485 bus as these all involve speaking to slaves, so we effectively turn the TX bus into a half-duplex channel (which RS485 is)

- Async processing of RS485 RX is probably similar, it's a big monolithic loop now but what would happen in future if it was async - is the loop today protecting us from collisions. I'm not entirely sure given the lack of exploration on my side

It would be interesting to brainstorm if any cases do seem like clearcut async optimizations, and we could do some testing. Anything which unwinds the (being honest here) monolithic mess inside ```TWCManager.py``` is a benefit in my eyes - one way in which I've started this is to move the RS485 protocol out to a separate module, initially to allow us to do testing using the Dummy module but ultimately to unwind the procedural code that makes up protocol parsing into something more flexible, so I can see some potential to then couple that with async IO in future - if we can do it without introducing RS485 bus collisions (or more than we might have today, anyway).

MikeBishop, 2021-04-29:
I'll note that my attempt to use the websocket API in MMM-Powerwall ran hard into the restriction that there can be only one websocket client at a time, and for me that's already taken by Teslamate.  I pivoted to an MQTT integration that can consume Teslamate events.  So I'm no longer looking at websocket usage in TWCManager for now.

The need to sequence things in the RS485 bus is definitely valid, but could be addressed by running a similar thread-safe queue to that currently used for background tasks -- "sending" actually inserts in the queue, and you can have an async task which just loops on get an item / wait for a new item in the queue, then send that item on the RS485 bus.  It's also worth exploring to what extent the RS485 calls themselves provide thread-safety, since IIRC we're not particularly bound to the particular order that messages get sent.

closed 2022-02-15 by ngardiner