ngardiner/TWCManager #247: Problems installing on a new Pi

jherby2k, 2021-03-12:
Hello, just trying this out for the first time and i'm encountering an issue at "make install". This is on a fresh install of Raspberry Pi OS (Lite) 5.4. I did an apt-get upgrade prior to following the installation steps at https://github.com/ngardiner/TWCManager/blob/v1.2.1/docs/README.md verbatim.

```bash
Running sysv_ipc-1.1.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-9bxmwsmw/sysv_ipc-1.1.0/egg-dist-tmp-at4340em
******************************************************************************
* Setup can't determine the value of PAGE_SIZE on your system, so it will
* default to 4096 which may not be correct.
*
* Please report this message and your operating system info to the package
* maintainer listed in the README file.
******************************************************************************
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.o' found anywhere in distribution
sysv_ipc_module.c:32:10: fatal error: Python.h: No such file or directory
 #include "Python.h"
          ^~~~~~~~~~
compilation terminated.
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
make: *** [Makefile:16: install] Error 1
```

tjikkun, 2021-03-12:
`sudo apt-get install python-dev` will probably fix it for you

jherby2k, 2021-03-12:
python3-dev, yep. I filed PR #248 to update the docs for Raspberry Pi OS

jherby2k, 2021-04-09:
Closing as #248 was merged to 1.2.1

closed 2021-04-09 by jherby2k